espressif/idf-eclipse-plugin #129: Not able to build with "Building Active Configuration" encountering a problem

EvanS124, 2020-09-21:
Currently, I am on the Compiling a Project step on https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md.
Whenever I click on build, I receive a dialogue box that says:
"Building Active Configuration" has encountered a problem. 
Errors occurred during the build.

Then on details its says:
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'test'.
org/eclipse/cdt/utils/Platform

[image]

**Expected behavior**
I expect my project to build properly.

bseishen, 2020-09-21:
Banging my head against the wall on this one too. I think this is the same issue. https://github.com/espressif/idf-eclipse-plugin/issues/127

kifantidis, 2020-09-21:
Hello. Which eclipse version do you have ??

kolipakakondal, 2020-09-21:
Hi All, 

We will provide the fix for this soon to support the Eclispe 2020-09 version. Please continue to use Eclipse 2020-06 until then.

kolipakakondal, 2020-09-21:
Duplicate of #127

closed 2020-09-21 by kolipakakondal

EvanS124, 2020-09-21:
> Hi All,
> 
> We will provide the fix for this soon to support the Eclispe 2020-09 version. Please continue to use Eclipse 2020-06 until then.

Downloading the 2020-06 version solved this problem. Thanks