espressif/idf-eclipse-plugin #131: Build issue with eclipse 20-09

alkosa, 2020-09-21:
**Describe the bug**
Hi,
I have tired to build a hello world project with eclipse 20-09, but I have got a message:
```bash
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'hello-world'.
org/eclipse/cdt/utils/Platform
```
I have switched back to eclipse 20-06. It is working perfectly.

I do not know what is the problem.

Thank you very much for your help.

kolipakakondal, 2020-09-21:
Duplicate of https://github.com/espressif/idf-eclipse-plugin/issues/127

ylj135cool, 2020-09-25:
Right clicked the project of your workspace,Selectd Properties->Builders,remove the selection for CDT Core Builder,and rebuild it

bheestand, 2020-09-30:
Having same issue

kolipakakondal, 2020-10-09:
Hi All,

ESP-IDF Eclipse Plugin 1.2.3 is released to support Eclipse 2020-09
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.3

Please update to the latest

closed 2020-10-09 by kolipakakondal