espressif/idf-eclipse-plugin #133: Eclipse 2020-9安装ESP-IDF eclipse plugin后可以正常编译下载，但是找不到头文件，无法转到头文件

ylj135cool, 2020-09-25:
**Describe the bug**
A clear and concise description of what the bug is.

安装完所有插件以及相关软件后，以Blink为模板新建一个项目（TestBlink)，根据说明操作，中途出现Build错误，去掉CDT的勾后正常，可以正确编译程序并下载到开发板，但是在程序开头的头文件包含中，程序提示找不到相关的头文件，也无法跳转到对应的头文件，头文件如下:

`
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"

`
以下两个头文件可以找到：
#include <stdio.h>
#include "sdkconfig.h"

尝试过手动添加包含文件目录，但是以IDF插件模板新建的项目，找不到包含路径相关的设置选项。
以上问题，请问是哪里设置有误吗？

**Screenshots**
[image]

**IDF Eclipse Plugin Version:*1.2.2* 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
  Espressif IDF Plugins for Eclipse	1.2.2.202009041040	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon
ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD	ESP-IDF Eclipse Plugin	1.2.2.202009041040	com.espressif.idf.feature

kolipakakondal, 2020-10-09:
Hi All,

ESP-IDF Eclipse Plugin 1.2.3 is released to support Eclipse 2020-09
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.3

Please update to the latest

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent release

closed 2021-02-16 by kolipakakondal