espressif/idf-eclipse-plugin #135: Eclipse is not starting after installing the idf-eclipse-plugin

markus-fischbacher, 2020-09-28:
Hi there,
I tried to set up the Eclipse with idf-eclipse-plugin environment as described in the documentation of this repository.
After installing the plugin, Eclipse does not start anymore.
I have no clue how to fix that.

Any help is appreciated.

Regards,
Markus

----------------------------------------------------------
```
!SESSION 2020-09-28 13:03:38.506 -----------------------------------------------
eclipse.buildId=4.16.0.I20200604-0540
java.version=1.8.0_222-2-ojdkbuild
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=de_DE
Framework arguments:  -product org.eclipse.epp.package.cpp.product -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product -data file:/C:/DEV/eclipse-workspace/ -product org.eclipse.epp.package.cpp.product

!ENTRY org.eclipse.cdt.arduino.core 4 0 2020-09-28 13:03:40.571
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: org.eclipse.cdt.arduino.core [90]
  Unresolved requirement: Require-Bundle: org.eclipse.tools.templates.freemarker; bundle-version="[1.1.100,2.0.0)"; visibility:="reexport"
    -> Bundle-SymbolicName: org.eclipse.tools.templates.freemarker; bundle-version="1.2.0.202008311307"
       org.eclipse.tools.templates.freemarker [622]
         Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=11))"
         Unresolved requirement: Require-Bundle: org.eclipse.tools.templates.core; bundle-version="1.1.2"; visibility:="reexport"
           -> Bundle-SymbolicName: org.eclipse.tools.templates.core; bundle-version="1.2.0.202008310315"
              org.eclipse.tools.templates.core [621]
                Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=11))"

	at org.eclipse.osgi.container.Module.start(Module.java:463)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel$2.run(ModuleContainer.java:1845)
	at org.eclipse.osgi.internal.framework.EquinoxContainerAdaptor$1$1.execute(EquinoxContainerAdaptor.java:136)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1838)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1779)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1743)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1665)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:234)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:345)

!ENTRY org.eclipse.cdt.arduino.ui 4 0 2020-09-28 13:03:40.580
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: org.eclipse.cdt.arduino.ui [91]
  Unresolved requirement: Require-Bundle: org.eclipse.tools.templates.ui; bundle-version="[1.1.100,2.0.0)"
    -> Bundle-SymbolicName: org.eclipse.tools.templates.ui; bundle-version="1.2.0.202008312024"; singleton:="true"
       org.eclipse.tools.templates.ui [623]
         Unresolved requirement: Require-Bundle: org.eclipse.tools.templates.core; bundle-version="1.0.0"; visibility:="reexport"
           -> Bundle-SymbolicName: org.eclipse.tools.templates.core; bundle-version="1.2.0.202008310315"
              org.eclipse.tools.templates.core [621]
                Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=11))"
  Unresolved requirement: Require-Bundle: org.eclipse.cdt.arduino.core; bundle-version="[2.1.300,3.0.0)"
    -> Bundle-SymbolicName: org.eclipse.cdt.arduino.core; bundle-version="2.1.301.202005042001"; singleton:="true"
       org.eclipse.cdt.arduino.core [90]
         Unresolved requirement: Require-Bundle: org.eclipse.tools.templates.freemarker; bundle-version="[1.1.100,2.0.0)"; visibility:="reexport"
           -> Bundle-SymbolicName: org.eclipse.tools.templates.freemarker; bundle-version="1.2.0.202008311307"
              org.eclipse.tools.templates.freemarker [622]
                Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=11))"
                Unresolved requirement: Require-Bundle: org.eclipse.tools.templates.core; bundle-version="1.1.2"; visibility:="reexport"
                  -> Bundle-SymbolicName: org.eclipse.tools.templates.core; bundle-version="1.2.0.202008310315"
  Unresolved requirement: Require-Bundle: org.eclipse.tm.terminal.view.core; bundle-version="[4.6.0,5.0.0)"
    -> Bundle-SymbolicName: org.eclipse.tm.terminal.view.core; bundle-version="4.7.0.202008310315"; singleton:="true"
       org.eclipse.tm.terminal.view.core [619]
         Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=11))"
  Unresolved requirement: Require-Bundle: org.eclipse.tm.terminal.view.ui; bundle-version="[4.6.0,5.0.0)"
    -> Bundle-SymbolicName: org.eclipse.tm.terminal.view.ui; bundle-version="4.7.0.202008310315"; singleton:="true"
[... 86 lines cut ...]
!ENTRY org.eclipse.osgi 4 0 2020-09-28 13:03:45.477
!MESSAGE Application error
!STACK 1
java.lang.NullPointerException
	at org.eclipse.e4.ui.internal.workbench.ResourceHandler.loadMostRecentModel(ResourceHandler.java:182)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.loadApplicationModel(E4Application.java:384)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createE4Workbench(E4Application.java:260)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:579)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:557)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:154)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:657)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:594)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1447)
```

kolipakakondal, 2020-10-01:
Hi @markus-fischbacher 

IDF Eclipse plugin requires Java 11 and above, and Eclipse 2020-06 CDT package. 
https://github.com/espressif/idf-eclipse-plugin#Prerequisites

From the log, I could see you're using Java 1.8.0_222-2-ojdkbuild which won't work the plugin. Could you please install Java 11 and above and see if that helps.

Thanks

markus-fischbacher, 2020-10-01:
Hi @kolipakakondal,
thanks for your hint. Changing Java to 11 did the trick. Just overseen that I used the wrong path in JAVA_HOME.

closed 2020-10-01 by markus-fischbacher