espressif/idf-eclipse-plugin #136: No Includes directory in Project Explorer and No C/C++ Build Opention is project properties.

Rajssss, 2020-10-01:
I have just installed Eclipse CTD v2020.6 and ESP-IDF v4.1 and configured it in Windows 10. I can build and flash without any problem. But I don't know why I can't see the Includes directory in Project explorer. Is it normal, if so is there any way I can add it there? Along with that, I have noticed, there is no C/C++ build section in Project Properties. :?: I am attaching the screenshots for reference.
[image]
![Project Properties(https://user-images.githubusercontent.com/24260856/94763149-1f186600-03c7-11eb-9274-dfe23f179c4e.jpg)

kolipakakondal, 2020-10-05:
Hi @Rajssss That's not applicable for Eclipse CDT CMake projects. Let me know what you're trying to achieve with those options so that we can take a look at it

Rajssss, 2020-10-05:
Yeah, I get it now. But is it same for the C/C++ build options in Project properties too?

matheusmbgeng, 2020-12-14:
@kolipakakondal Hello my friend, i have an project that i used to develop using GNU Make. I would like to develop it using CMake on eclipse right now. Any advices on that? I already setup the eclipse plugin and its working fine on new "idf-projects". But when i try to import this old makefile project as "idf-project", the project doesn't appear on "launch configuration". It looks like eclipse see this project still as GNU Make even if i import it as "Existing ESP-IDF PROJECT". I noticed that this project has this "Include" directory and i cant delete it. Any hints or advices of something i could do?

kolipakakondal, 2020-12-15:
Hi @matheusmbgeng - Could you provide a sample project for this?

matheusmbgeng, 2020-12-15:
@kolipakakondal Here's a project sample that happens to have the same behaviour than the project i mentioned on the other post.
[TestProject.zip](https://github.com/espressif/idf-eclipse-plugin/files/5696292/TestProject.zip)

matheusmbgeng, 2020-12-15:
I tried to change the "Project Nature" on Properties window. I configured project nature exactly as the examples from esp-idf. Now my project appear on launch configuration, but when i press build button, nothing happens. I also saw that after changing project nature, now on "Builders" tab i have no builder configured, while on esp-idf examples there is a "CDT Core Builder" selected.

sigmaaa, 2020-12-22:
Hi @matheusmbgeng. Please try to build the project via the run button after selecting a project in the launch configuration. It should build a project first. In your project looks like the build configuration is not using the IDF CMake command and the right toolchain and that's why the build button doesn't work, but it looks like the run button works ok.

kolipakakondal, 2022-02-14:
Hello,

We have implemented source code navigation functionality(pressing F3) and fixed unresolved header issues with the plugin as part of the v2.4.0 release.

Please get the latest changes using the update site. Please check more release page for more details.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

We also have the v2.4.1 patch release, please make sure you get the v2.4.1 and update

Thank you

closed 2022-02-14 by kolipakakondal