espressif/idf-eclipse-plugin #138: Mac OS - Install plugin for Eclipse - Python 3

mcmega, 2020-10-02:
Hi everyone!
Mac OS system.

Install plugin for Eclipse. Can't install.
In my system Python 3.8.6 only.

Console Eclipse:

Executing /usr/bin/python /Users/poe/esp/esp-idf/tools/idf_tools.py install-python-env
ImportError: No module named python2.python2
Creating a new Python environment in /Users/poe/.espressif/python_env/idf4.2_py2.7_env
Traceback (most recent call last):
  File "/Users/poe/esp/esp-idf/tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "/Users/poe/esp/esp-idf/tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "/Users/poe/esp/esp-idf/tools/idf_tools.py", line 1208, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python', '-m', 'virtualenv', '/Users/poe/.espressif/python_env/idf4.2_py2.7_env']' returned non-zero exit status 1

Executing /usr/bin/python /Users/poe/esp/esp-idf/tools/idf_tools.py export --format=key-value
PATH=/Users/poe/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin:/Users/poe/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin:/Users/poe/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/poe/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/poe/.espressif/tools/cmake/3.16.4/CMake.app/Contents/bin:/Users/poe/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/bin:/Users/poe/.espressif/tools/ninja/1.10.0/:/Users/poe/esp/esp-idf/tools:$PATH
OPENOCD_SCRIPTS=/Users/poe/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420/openocd-esp32/share/openocd/scripts
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.

I do not have Python2.7 in system.

kolipakakondal, 2020-10-05:
Hi @mcmega 

Could you try the steps mentioned here and see if that helps https://www.esp32.com/viewtopic.php?t=12481#p49793

It seems to be some issue in installing and configuring the virtual environment

mcmega, 2020-10-05:
Don't work

poe@POE-MacBook-Pro ~ % python --version
Python 3.8.6
poe@POE-MacBook-Pro ~ % python3 -m pip install virtualenv
Collecting virtualenv
  Downloading virtualenv-20.0.33-py2.py3-none-any.whl (4.9 MB)
     |████████████████████████████████| 4.9 MB 1.6 MB/s 
Collecting filelock<4,>=3.0.0
  Using cached filelock-3.0.12-py3-none-any.whl (7.6 kB)
Collecting distlib<1,>=0.3.1
  Using cached distlib-0.3.1-py2.py3-none-any.whl (335 kB)
Collecting appdirs<2,>=1.4.3
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting six<2,>=1.9.0
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Installing collected packages: filelock, distlib, appdirs, six, virtualenv
  WARNING: The script virtualenv is installed in '/Library/Frameworks/Python.framework/Versions/3.8/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed appdirs-1.4.4 distlib-0.3.1 filelock-3.0.12 six-1.15.0 virtualenv-20.0.33
WARNING: You are using pip version 20.2.1; however, version 20.2.3 is available.
You should consider upgrading via the '/usr/local/bin/python3 -m pip install --upgrade pip' command.

**In Eclipse**
Installing tools...
This can take a while. Please be patient.
Executing /usr/bin/python /Users/poe/esp-idf/tools/idf_tools.py install all
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20191114 (already installed)
Skipping ninja@1.9.0 (already installed)

Executing /usr/bin/python /Users/poe/esp-idf/tools/idf_tools.py install-python-env
/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python: No module named pip
Creating a new Python environment in /Users/poe/.espressif/python_env/idf4.1_py2.7_env
Installing virtualenv
Traceback (most recent call last):
  File "/Users/poe/esp-idf/tools/idf_tools.py", line 1387, in <module>
    main(sys.argv[1:])
  File "/Users/poe/esp-idf/tools/idf_tools.py", line 1383, in main
    action_func(args)
  File "/Users/poe/esp-idf/tools/idf_tools.py", line 1186, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1

Executing /usr/bin/python /Users/poe/esp-idf/tools/idf_tools.py export --format=key-value
PATH=/Users/poe/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/Users/poe/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/Users/poe/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/poe/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/poe/.espressif/tools/cmake/3.13.4/CMake.app/Contents/bin:/Users/poe/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/bin:/Users/poe/.espressif/tools/ninja/1.9.0/:/Users/poe/esp-idf/tools:$PATH
OPENOCD_SCRIPTS=/Users/poe/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/share/openocd/scripts
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.

mcmega, 2020-10-05:
I changed the path to Python, but terminal:
```
Building in: /Volumes/Temp/eclipse-workspace/Test/build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=/Users/poe/esp-idf/tools/cmake/toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DIDF_TARGET=esp32 /Volumes/Temp/eclipse-workspace/Test
```
And pop-up window:
```
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'Test'.
org/eclipse/cdt/utils/Platform
```

kolipakakondal, 2020-10-06:
That's a problem with Eclipse 2020-09. Please switch back to Eclipse 2020-06 CDT until we provide for the latest eclipse.https://github.com/espressif/idf-eclipse-plugin/issues/127

mcmega, 2020-10-06:
I'm install Eclipse 2020-06 CDT - It Ok!

closed 2020-10-06 by mcmega