espressif/idf-eclipse-plugin #139: Something prepends /usr/bin to PATH

szekelyisz, 2020-10-02:
When I try to run the build I get this error:

```
-- Checking Python dependencies...
The following Python requirements are not satisfied:
setuptools>=21
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui>=0.13.2.0
pygdbmi<=0.9.0.2
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
To install the missing packages, please run "/home/szabi/esp/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/szabi/.espressif/python_env/idf4.3_py3.8_env
    Python interpreter used: /usr/bin/python
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /usr/bin:/home/szabi/.espressif/python_env/idf4.3_py3.8_env/bin:/home/szabi/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin:/home/szabi/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin:/home/szabi/.espressif/tools/riscv-none-embed-gcc/riscv-none-embed-gcc-8.2.0/xPacks/riscv-none-embed-gcc/8.2.0-3.1/bin:/home/szabi/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/szabi/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/szabi/.espressif/tools/cmake/3.16.4/bin:/home/szabi/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin:/home/szabi/.espressif/python_env/idf4.3_py3.8_env/bin:/home/szabi/esp/esp-idf/tools:/home/szabi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```

As you can see, the problem is that something prepends `/usr/bin` to PATH. This is not coming from my project environment settings, it starts with the second entry of the above.

Anybody seen this before? I spent hours looking through project settings and eclipse preferences, haven't found anything. A `grep -r` in the project folder didn't reveal any file containing `/usr/bin` apart from the project environment settings file.

kolipakakondal, 2020-10-05:
Hi @szekelyisz 

Could you check the CDT build environment PATH variable?

`Preferences > C/C++ Build > Environment`

igrr, 2020-10-05:
I think I had seen behavior with `/usr/bin` being prepended on macOS some time ago, mentioned in the comment here: https://github.com/espressif/idf-eclipse-plugin/issues/40#issuecomment-599966025.

szekelyisz, 2020-10-06:
@kolipakakondal It doesn't start with `/usr/bin`, but with the second entry from the problematic PATH. Now actually I ran into #35 so I can't reproduce the problem anymore. If I click the Build button on the Launch Bar, I just get `No esp launch target found. Please create/select the correct 'Launch Target'` on the build console. But I can't select or create one because the drop-down menu is gone. (And, interestingly enough, if I click the Debug button on the Launch Bar, I get the same error, but the debug session starts. Maybe the status reported by the build process is wrong or ignored?)

kolipakakondal, 2020-10-06:
@szekelyisz It's showing the wrong error message in the debug mode. This seems to happen when the launch target is not available in the debug mode. Please ignore this for a while. We will fix this.

szekelyisz, 2020-10-06:
OK so I [got past](https://github.com/espressif/idf-eclipse-plugin/issues/35#issuecomment-704042505) that issue. Building now just seems to work.

But if I click the Run button on the Launch Bar after a successful build, here's what I see:

[image]

Build is actually successful but something still makes Eclipse think otherwise. But that probably belongs to its own issue.

I found what causes the problem. I played around with Project Natures before and added "CDT Builder Project". Which caused a lot of problems (null pointer exceptions all over the place), so I removed it. This left a setting behind that prepends `/usr/bin` to my PATH every time.

I wanted to play with Project Natures to see if I can make this project work with the [cmake4eclipse](https://github.com/15knots/cmake4eclipse) plugin. I managed to put this together earlier somehow, but then I switched computers and what worked on a mac doesn't work on Linux. So I tried to build it from scratch. cmake4eclipse requires some CDT natures ("CDT Builder Project" or "CDT Make Nature" or both, I couldn't figure out yet). The cmake4eclipse project has very nice features like providers for automatic include directory discovery based on the `CMakeLists.txt` files and compiler built-in options. The latter will tell Eclipse about system header files that are not seen by the idf-eclipse-plugin for now. You can achieve a source code tree that's 98% error-free according to Eclipse (only the ESP_LOG* macros had errors), autocompletion works with almost anything, and all headers are resolved correctly. Would be a good idea to make it work and integrate it into this project.

igrr, 2020-10-06:
> The cmake4eclipse project has very nice features like providers for automatic include directory discovery based on the CMakeLists.txt files and compiler built-in options. The latter will tell Eclipse about system header files that are not seen by the idf-eclipse-plugin for now.

@szekelyisz theoretically, same should be the case with this plugin, as it obtains information about include paths and preprocessor flags from compile_commands.json, generated by CMake. If for some reason indexing is not working for you, we can try to investigate.

szekelyisz, 2020-10-06:
Thanks @igrr that's great. I do have issues with Eclipse not seeing system include paths (and hence all my standard headers remaining unresolved), but if that should work, I'll open a separate issue for that.

It looks like my original problem came from modifying project natures which is admittedly not a supported operation (Eclipse vigorously warns about it). Feel free to close this issue or I'd be happy to assist in further investigation if you want to dig deeper into this issue regardless.

kolipakakondal, 2020-10-07:
HI @szekelyisz  For unresolved errors, I would request you to create a new issue with the following logs.

Before you capture the logs, please create a new project from the templates and build the project.

1. Error log: https://github.com/espressif/idf-eclipse-plugin#error-log
2. Parser log: Project > C/C++ Index > Create Parser Log
3. Eclipse Environment info: https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-provide-eclipse-envrionment-and-plugins-information

closed 2021-02-16 by kolipakakondal