espressif/idf-eclipse-plugin #140: eclipse 2020-06 doesnt start after installing ESP-IDF plugin v4.1

Hullesha, 2020-10-03:
followed steps for installing ESP-IDF for eclipse ide from below link
https://github.com/espressif/idf-eclipse-plugin#InstallTools

1. installed Eclipse 2020-06 CDT.
2. Installed IDF Plugin using update site URL 
  
    Go to Help -> Install New Software
    Click Add…
    Enter Location of the repository https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
    Enter Name as Espressif IDF Plugin for Eclipse
    Click Ok
    Select Espressif IDF from the list and proceed with the installation
above installed esp-idf successfully.
3. but once i tried opening the eclipse ide, its not opening.
it throw error saying, error occurred see the log file in C://.../.metadata/.log
after looking into log

!ENTRY org.eclipse.e4.ui.workbench 4 0 2020-10-03 10:35:10.851
!MESSAGE Unable to load resource file:/C:/Users/guh6kor/202006_eclipse_workspace/.metadata/.plugins/org.eclipse.e4.workbench/workbench.xmi
!STACK 0
java.lang.NullPointerException
	at org.eclipse.e4.ui.internal.workbench.ResourceHandler.getResource(ResourceHandler.java:287)
	at org.eclipse.e4.ui.internal.workbench.ResourceHandler.loadResource(ResourceHandler.java:263)
	at org.eclipse.e4.ui.internal.workbench.ResourceHandler.loadMostRecentModel(ResourceHandler.java:169)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.loadApplicationModel(E4Application.java:384)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createE4Workbench(E4Application.java:260)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:579)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:557)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:154)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:657)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:594)
	at [org.eclipse.equinox.launcher.Main.run(Main.java:1447)]

pranaysharma, 2020-10-04:
+1 same issue on MacOs

kolipakakondal, 2020-10-05:
Hi @Hullesha Could you start Eclipse with a new workspace and see if that helps. It seems to be current workspace is corrupted.

Could you also share C://.../.metadata/.log which is generated to look further on this

martinloren, 2020-10-07:
Confirm the issue on Windows. Actually is not possible to install ESP-IDF plugin any more.

kolipakakondal, 2020-10-07:
Hi @martinloren  Could you share the error log to check further on this

Hullesha, 2020-10-09:
Hi Kondal, i am able start eclipse 2020-06 with esp-idf v4.1 now.
i was using java 1.8.0 but in prerequisites its mentioned java 11 and above, after updating java to 11 version, eclipse starts without any error.

thank you.

closed 2020-10-09 by Hullesha

SpiRaiL, 2020-11-17:
I can confirm that its Java as well. I had the right java installed, but this only worked after I uninstalled all the old java versions. Hopefully I didn't need them for something else on my system.