espressif/idf-eclipse-plugin #147: Unresolved inclusion and Symbol "XXX" cound not be resolved in latest idf-eclipse-plugin in Debug Launch mode

ThibaudBrandt, 2020-10-16:
Hi,
I use Windows10 with Eclipse Version: 2020-06 (4.16.0) / Build id: 20200615-1200 and Espressif IDF Plugins for Eclipse Espressif IDF 1.0.0.7f8U7QcLblnh87Z6Yc_2Qq8-B (latest found today on https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/)
My probleme is very similare to #60 but I dont have any solution to fix my problem (switching to 'C/C++ perspective run mode' and rebuild the index doesnt fix the problem, moving back to run mode either.

1. When creating a project (example hello_world), after the build the indexer is OK and knows about all includes
2. Switching to a Debug configuration and all the includes become "unresolved", the debug works fine but there is no way to make the indexer works.
Even if I switch back to "launch/build" configuration the indexer stay lost.
Sometime I manage to make it works again by restarting Eclipse and build the projet again but as soon as I launch the debug configuration I lost the indexer.

here is what I tried :
- launche and install Eclipse/ESP-idf 4.1/Tools with administrator privileges
- the #33 but my computer dont stop recalculating the indexer and not everything is found, that doesnt seem to be a good permanent solution 
-  select "Enable project specific settings" in the Project>C/C++ general>indexer

Thanks for your help

TobiasBoeschMiele, 2020-11-05:
Can confirm this. In my eclipse all esp-idf includes are also unresolved and I cannot get them back resolved using restart. (Symbols sometimes get back resolved after closing and starting eclipse - restart is not enough)

I use:

Espressif IDF Plugins for Eclipse	1.2.3.202010091505

Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-09 (4.17.0)
Build id: 20200910-1200
OS: Windows 10, v.10.0, x86_64 / win32

axbusch, 2020-11-05:
Lol... Was not doing anything with esp32 for months. Looking back in here again... Same errors as months ago.

One Bug fixed, 2 new it seems. You really should improve your quality managment. Sorry to say

kolipakakondal, 2021-04-26:
Unresolved inclusion issues in the debug mode issue got resolved with the 2.0.0 release, please update to the latest.

https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.1.0

closed 2021-04-26 by kolipakakondal