espressif/idf-eclipse-plugin #150: New ESP_IDF GDB OpenOCD Debugging run configuration results in java.lang.NullPointerException

craigpeacock, 2020-10-31:
**Describe the bug**
java.lang.NullPointerException exception when opening new ESP_IDF GDB OpenOCD Debugging Run Configuration

**To Reproduce**
Steps to reproduce the behavior:
1. Select Run Menu -> Debug Configurations
2. Right click on 'ESP-IDF GDB OpenOCD Debugging' and click New Configuration
3. See error "java.lang.NullPointerException"

Doing the same for 'GDB OpenOCD Debugging' results with no error.

**Screenshots**
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
[Eclipse Enviroment Information.txt](https://github.com/espressif/idf-eclipse-plugin/files/5469022/Eclipse.Enviroment.Information.txt)

ESP-IDF Tools Information:
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.9.0 (recommended, installed)

IDF Eclipse Plugin Version: 1.2.3.202010091505

kolipakakondal, 2020-11-03:
Hi @craigpeacock 

Have you selected any project in the project explorer before creating a new configuration? This issue happens when you're trying to create a new configuration without any project from the project explorer or creating a new configuration on the closed project.

Let me know more details about your scenario to reproduce this issue.

kolipakakondal, 2020-11-03:
https://github.com/espressif/idf-eclipse-plugin/pull/152

craigpeacock, 2020-11-07:
Kondal, 

Yes, you are right. The project was shown in the project explorer, but not actually opened.

kolipakakondal, 2020-11-09:
Fix will be available as part of v1.2.4 release

closed 2020-11-09 by kolipakakondal