espressif/idf-eclipse-plugin #151: we have to install esp idf inside a esp folder then it will work otherwise it wont work

swaroopbu, 2020-11-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

kolipakakondal, 2020-11-03:
Hi @swaroopbu 

Could you provide a reproducible scenario and details to explain your issue title? 

> we have to install esp idf inside a esp folder then it will work otherwise it wont work

kolipakakondal, 2021-02-16:
The problem description is not sufficient to troubleshoot the problem, hence closing the issue.

closed 2021-02-16 by kolipakakondal