espressif/idf-eclipse-plugin #154: xTaskCreate not resolvable (__attribute__ not known?)

TobiasBoeschMiele, 2020-11-04:
Issue found using the [ble_spp_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_server) example.

In file `ble_server_demo.c`:
````
    cmd_cmd_queue = xQueueCreate(10, sizeof(uint32_t));
    xTaskCreate(spp_cmd_task, "spp_cmd_task", 2048, NULL, 10, NULL);
````
`xQueueCreate` can be resolved and `xTaskCreate` can not.

Preview: This should not (and cannot) be solved by https://github.com/espressif/idf-eclipse-plugin/issues/33#issuecomment-558574707 (Linking folder components directory to eclipse project) because the xQueueCreate works and is not added to the project using linked ressources.

Possible reason: Keyword/function `__attribute__` not recognized. Here's how i got there:

In file `freertos/task.h` there is a macro for the xTaskCreate function:
```
#if( configSUPPORT_DYNAMIC_ALLOCATION == 1 )

	static inline IRAM_ATTR BaseType_t xTaskCreate(
			TaskFunction_t pvTaskCode,
			const char * const pcName,
			const uint32_t usStackDepth,
			void * const pvParameters,
			UBaseType_t uxPriority,
			TaskHandle_t * const pvCreatedTask)
	{
		return xTaskCreatePinnedToCore( pvTaskCode, pcName, usStackDepth, pvParameters, uxPriority, pvCreatedTask, tskNO_AFFINITY );
	}

#endif
```

which uses another macro: `IRAM_ATTR `
which defines sections:
```
#define IRAM_ATTR _SECTION_ATTR_IMPL(".iram1", __COUNTER__)
```

using the `__attribute__` keyword:
`#define _SECTION_ATTR_IMPL(SECTION, COUNTER) __attribute__((section(SECTION "." _COUNTER_STRINGIFY(COUNTER))))`

I guess the keyword `__attribute__` is not known be either idf-eclipse-plugin or eclipse, because replacing the `IRAM_ATTR` with just `__attribute__("testtext")` keeps the syntax error active.

Why is macro `__attribute__` not recognized and which software should do it? (Eclipse or this plugin)

TobiasBoeschMiele, 2020-11-12:
Closed and started eclipse today with the debug configuration selected. After that I was suddenyl able to navigate to `xTaskCreate` and even further.
The `__attribute__` keyword was shown as "built in". (Toolchain or language symbol - colored violet).
Immediately after switching back to the build configuration (not debug) most symbols like `xTaskCreate` were not resolvable again.

Closing and starting eclipse with debug configuration selected did not help this time.

kolipakakondal, 2020-11-12:
Hi @TobiasBoeschMiele 

Thanks for the update. We are working on these unresolved and header issues especially when we add esp-idf components to the project directory. We will update you on this soon.

Thanks

kolipakakondal, 2022-02-14:
Hello,

We have implemented source code navigation functionality(pressing F3) and fixed unresolved header issues with the plugin as part of the v2.4.0 release.

Please get the latest changes using the update site. Please check more release page for more details.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

We also have the v2.4.1 patch release, please make sure you get the v2.4.1 and update

Thank you

closed 2022-02-14 by kolipakakondal