espressif/idf-eclipse-plugin #165: > BTW I just noticed that you are trying to flash ELF file. That's wrong, you need to flash `build/myapp.bin`.

pwmb, 2020-12-14:
> BTW I just noticed that you are trying to flash ELF file. That's wrong, you need to flash `build/myapp.bin`.
> So, please, try the following command `openocd -l openocd_log.txt -d3 -f ./esp-huzzah-32.cfg -c "program_esp build/myapp.bin 0x10000 verify reset exit`

Using the `bin` file instead of the `elf`file makes flashing and verifying work. Thanks for looking at the details!
The [eclipse plugin](https://github.com/espressif/idf-eclipse-plugin) automatically selects the elf file for flashing. That should be changed.

Cannot close this issue - but I would do it. :-)

_Originally posted by @TobiasBoeschMiele in https://github.com/espressif/openocd-esp32/issues/128#issuecomment-744424732_

pwmb, 2020-12-14:
@kolipakakondal FYI

TobiasBoeschMiele, 2020-12-14:
Thanks @pwmb 
Maybe this is not an issue as long as there is no flashing over JTAG or in other words through the debugging configuration.
This screen made me try to flash the `elf` file with openocd command

TobiasBoeschMiele, 2020-12-14:
Thanks @pwmb 
Maybe this is not an issue in this plugin as long as there is no flashing over JTAG implemented in this plugin.
This screen chooses the `elf` file to debug by default:
[image]

I used the elf file for flashing using the openocd command because of this default setting in the debug configuration of the plugin.
For debugging this might be correct. If one wants to flash a binary though JTAG, [now  we know it should be the `bin`file](https://github.com/espressif/openocd-esp32/issues/128#issuecomment-744424732).

When flashing is considered in this eclipse plugin by default the team should make sure they flash the `bin` file by default, then debug, then debug the `elf` file.

If this all is correct, then this issue can be closed I think.

kolipakakondal, 2020-12-15:
> When flashing is considered in this eclipse plugin by default the team should make sure they flash the `bin` file by default, then debug, then debug the `elf` file.
> 

You're right. That's what is happening now. Eclipse has a two-step process - flashing and debugging. For flashing we use .bin and for debugging .elf file is used.

However, we can implement this in a single step going forward when we start on debugging it will automatically take care of flashing as well.

TobiasBoeschMiele, 2020-12-15:
>However, we can implement this in a single step going forward when we start on debugging it will automatically take care of flashing as well.

That would be nice while distinguishing between JTAG and serial flashing.

>You're right. That's what is happening now. Eclipse has a two-step process - flashing and debugging. For flashing we use .bin and for debugging .elf file is used.

Then this issue can be closed I think.

TobiasBoeschMiele, 2020-12-15:
>However, we can implement this in a single step going forward when we start on debugging it will automatically take care of flashing as well.

Is there a way to get this done in the current version already using specific openocd commands?

igrr, 2021-01-20:
@TobiasBoeschMiele uploading the application via JTAG is possible using `program_esp` OpenOCD command. Please see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#upload-application-for-debugging. At the moment uploading over JTAG is slower than over UART, as the JTAG loader doesn't implement binary compression yet.

closed 2021-02-16 by kolipakakondal