espressif/idf-eclipse-plugin #168: "Failed to create temporary file" during compile/flash process. Windows path length.

alezancomelit, 2020-12-18:
**Describe the error**
When build my project (with hammer icon) no error happens.
After that, if I click on "Run" icon to flash it, this happen:
- restart build process (strange!) with in console first line: 

> Warn about uninitialized values.

- return this error: 

> [713/1165] Building C object esp-idf/NxpNfcRdLib/CMakeFiles/__idf_NxpNfcRdLib.dir/C_/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/common/lib/NxpNfcRdLib/comps/phhalHw/src/Rc663/phhalHw_Rc663_Int.c.obj
FAILED: esp-idf/NxpNfcRdLib/CMakeFiles/__idf_NxpNfcRdLib.dir/C_/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/common/lib/NxpNfcRdLib/comps/phhalHw/src/Rc663/phhalHw_Rc663_Int.c.obj 
ccache C:\esp\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMIFARE_CLASSIC -DNXPBUILD_CUSTOMER_HEADER_INCLUDED -DPHDRIVER_CLEMESP32PN5180_BOARD -DPH_OSAL_FREERTOS -Iconfig -IC:/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/common/lib/NxpNfcRdLib/intfs -IC:/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/common/lib/NxpNfcRdLib/types -I../components/NxpNfcRdLibAL -I../main -IC:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include .....................
ccache: error: Failed to create temporary file for esp-idf/NxpNfcRdLib/CMakeFiles/__idf_NxpNfcRdLib.dir/C_/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/common/lib/NxpNfcRdLib/comps/phhalHw/src/Rc663/phhalHw_Rc663_Int.c.obj: No such file or directory

Same error obtained using via cmd line:
> idf.py build

In this case I resolve changing build directory:
> idf.py -B c:\temp_build build

I know that in Windows there is a limit into file path length.
I use Windows 10 64bits, I removed the limit into Register and Local group policy editor, but obtain same error!

If I move my project into "c:\repo", I build and flash correctly with both idf.py and Eclipse.
But It's not possible because my project it's shared with my workgroup and need to be compiled by different people without path limit.

I have 2 questions:
1. Can I remove the path limit?
2. If not possible, how I force use a different build directory with Eclipse build, flash and debug operations, like idf.py cmd line?

**Environment Information:** 
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/5716692/eclipse_config.txt)

**ESP-IDF Tools Information:**
Executing C:\Python39\python.exe C:\esp\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r2-8.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r2-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20191114 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

**IDF Eclipse Plugin Version:** 
1.2.3.202010091505

**IDF Version:** 
release 4.1

alezancomelit, 2020-12-18:
Just a comment about my description.

If I move my project into "c:\esp\":
- on "Run" I see in any case the rebuild
- but at time to compile the phhalHw_Rc663_Int.c file, no error happen and in console I see:
> [1026/1165] Building C object esp-idf/NxpNfcRdLib/CMakeFiles/__idf_NxpNfcRdLib.dir/C_/esp/active_reader_v3/common/lib/NxpNfcRdLib/comps/phhalHw/src/Rc663/phhalHw_Rc663_Int.c.obj

kolipakakondal, 2020-12-22:
> I have 2 questions:
> 
> 1. Can I remove the path limit?
> 2. If not possible, how I force use a different build directory with Eclipse build, flash and debug operations, like idf.py cmd line?
> 
> **E**

It's not possible to set a different build directory from eclipse while compiling a project. However, we will explore what it means to be and the possibility of it.

@igrr @projectgus Any suggestions on the path limit?

In general, Windows has 260 characters path limit and CMake seems to accept up to 250 characters as mentioned here https://gitlab.kitware.com/cmake/cmake/-/issues/19675

projectgus, 2020-12-22:
Hi @alezancomelit,

Unfortunately when the limit is coming from Windows then there's not a lot we can do from ESP-IDF side. In this case it looks like the error is coming inside ccache.exe when it tries to access this file.

Just to check, what is the full path to the build directory when it fails to build? The partial path `esp-idf/NxpNfcRdLib/CMakeFiles/_idf_NxpNfcRdLib.dir/C/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/common/lib/NxpNfcRdLib/comps/phhalHw/src/Rc663/phhalHw_Rc663_Int.c.obj` is 175 characters long, so I assume the absolute path is going to be around ~75 characters longer. However I wanted to be certain we're not looking at the wrong thing.

Is "opting in" to Windows 10 long paths an option for you and your colleagues?
https://docs.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation#enable-long-paths-in-windows-10-version-1607-and-later

Angus

alezancomelit, 2020-12-22:
> > I have 2 questions:
> > 
> > 1. Can I remove the path limit?
> > 2. If not possible, how I force use a different build directory with Eclipse build, flash and debug operations, like idf.py cmd line?
> 
> It's not possible to set a different build directory from eclipse while compiling a project. However, we will explore what it means to be and the possibility of it.
> 
> @igrr @projectgus Any suggestions on the path limit?
> 
> In general, Windows has 260 characters path limit and CMake seems to accept up to 250 characters as mentioned here https://gitlab.kitware.com/cmake/cmake/-/issues/19675

Thanks @kolipakakondal for your reply.

Some weeks ago, I found that into project configuration, _Build Settings_ -> Additional CMake Arguments_, if I add _-B c:\esp_build_, when press "hammer" icon, built correctly in directory c:\esp_build.
[image]
But if after click on green icon to Run, to flash into device, restart build process into /build directory under project directory.

Now I try to reproduce same test but, during build pressing "hammer" I obtain this:

Building in: C:\ALESSIO\PROGETTI\Vglect3\Software\active_reader_v3\application\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\esp\esp-idf\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DIDF_TARGET=esp32 -B c:\esp_build C:\ALESSIO\PROGETTI\Vglect3\Software\active_reader_v3\application
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.28.0.windows.1") 
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: alessio_dev@d3bf5b6443d2e4c7645
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\esp\esp-idf\requirements.txt are satisfied.
Loading defaults file C:/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/application/sdkconfig.defaults...
-- Found PythonInterp: C:/esp/.espressif/python_env/idf4.1_py3.9_env/Scripts/python.exe (found version "3.9") 
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- Adding linker script C:/esp_build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: NxpNfcRdLib NxpNfcRdLibAL app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter udb_library ulp unity utils vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/application/components/NxpNfcRdLib C:/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/application/components/NxpNfcRdLibAL C:/esp/esp-idf/components/app_trace C:/esp/esp-idf/components/app_update C:/esp/esp-idf/components/asio C:/esp/esp-idf/components/bootloader C:/esp/esp-idf/components/bootloader_support C:/esp/esp-idf/components/bt C:/esp/esp-idf/components/cbor C:/esp/esp-idf/components/coap C:/esp/esp-idf/components/console C:/esp/esp-idf/components/cxx C:/esp/esp-idf/components/driver C:/esp/esp-idf/components/efuse C:/esp/esp-idf/components/esp-tls C:/esp/esp-idf/components/esp32 C:/esp/esp-idf/components/esp_adc_cal C:/esp/esp-idf/components/esp_common C:/esp/esp-idf/components/esp_eth C:/esp/esp-idf/components/esp_event C:/esp/esp-idf/components/esp_gdbstub C:/esp/esp-idf/components/esp_http_client C:/esp/esp-idf/components/esp_http_server C:/esp/esp-idf/component [... cut ...] glect3/Software/active_reader_v3/application/components/utils C:/esp/esp-idf/components/vfs C:/esp/esp-idf/components/wear_levelling C:/esp/esp-idf/components/wifi_provisioning C:/esp/esp-idf/components/wpa_supplicant C:/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/esp_build
cmake --build . -- -v
Error: could not load cache
Build complete (1 errors, 0 warnings): C:\ALESSIO\PROGETTI\Vglect3\Software\active_reader_v3\application\build

alezancomelit, 2020-12-22:
Thanks @projectgus for your reply.

> Just to check, what is the full path to the build directory when it fails to build? The partial path [...]

The partial path is into project /build directory.
In my case `c:/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/application/build/esp-idf/NxpNfcRdLib/CMakeFiles/_idf_NxpNfcRdLib.dir/C/ALESSIO/PROGETTI/Vglect3/Software/active_reader_v3/common/lib/NxpNfcRdLib/comps/phhalHw/src/Rc663/phhalHw_Rc663_Int.c.obj`, that is 247 characters long.
I can try to reduce the depth:
- moving up my project
- reduce library name from `NxpNfcRdLib` to `nfclib` for example

The problem it's limited to some files into `NxpNfcRdLib` that it's a third-party library and I can not change internal folders structure.

> Is "opting in" to Windows 10 long paths an option for you and your colleagues?
> https://docs.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation#enable-long-paths-in-windows-10-version-1607-and-later

On Windows 10, I already make the fix into system register, as indicated in your link. But nothing change.

Note: Just for test, I try to build and flash using Visual Studio Code IDF plug-in.
Respect to use `idf.py` cmd line or Eclipse plug-in, with VSC compile and flash correctly!!
But for debugging VSC it's very limited......

closed 2021-02-16 by kolipakakondal