espressif/idf-eclipse-plugin #169: After installed esp-inf plugin, no ESP-IDF Tools Manager menu item under help

bonmotwang, 2020-12-20:
I installed eclipse cpp 2020-06 today.
Then installed espressif esp-idf eclipse plug in.
Buf after the installation and restart the computer, I cannot find ESP-IDF Tools Manager under help menu.
My Java version is 15.

tommy3001, 2020-12-20:
Same issue here.
Java version: 
openjdk version "1.8.0_252"
OpenJDK Runtime Environment (build 1.8.0_252-8u252-b09-1~19.10-b09)
OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)
Ubuntu 19.10

I can't proceed with the installation.

kolipakakondal, 2020-12-22:
> I installed eclipse cpp 2020-06 today.
> Then installed espressif esp-idf eclipse plug in.
> Buf after the installation and restart the computer, I cannot find ESP-IDF Tools Manager under help menu.
> My Java version is 15.

Could you please share the error log as mentioned here https://github.com/espressif/idf-eclipse-plugin#error-log

kolipakakondal, 2020-12-22:
Hi @tommy3001 @bonmotwang - @We have seen few users reported issues with the Java 15 version, could you please switch to Java 11 which is a more stable version

kolipakakondal, 2020-12-22:
> Same issue here.
> Java version:
> openjdk version "1.8.0_252"
> OpenJDK Runtime Environment (build 1.8.0_252-8u252-b09-1~19.10-b09)
> OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)
> Ubuntu 19.10
> 
> I can't proceed with the installation.

Java 1.8 is not supported by the plugin. The minimum required version to work with the plugin is Java 11 and above https://github.com/espressif/idf-eclipse-plugin#installing-prerequisites

real-iot, 2021-01-05:
same issue here using "jdk1.8.0_181" on windows x64.
checking out eclipse 4.18 with jre 15.0.1 instead and commit [bbccbb4]

icodk, 2021-02-03:
same issue here using "jdk-11" on windows x64.
Eclipse 2020-12
```
C:\WINDOWS\system32>java -version
openjdk version "11" 2018-09-25
OpenJDK Runtime Environment 18.9 (build 11+28)
OpenJDK 64-Bit Server VM 18.9 (build 11+28, mixed mode)
```
no ESP-IDF Tools Manager menu item under help after installing from
 ` https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/`

kolipakakondal, 2021-02-03:
> no ESP-IDF Tools Manager menu item under help after installing from

Could you share the error log - https://github.com/espressif/idf-eclipse-plugin#error-log

icodk, 2021-02-03:
Error log is long
here is en export of the Message section in the error log
Let me know if you need more
[eclipse_error_log_message.log](https://github.com/espressif/idf-eclipse-plugin/files/5917314/eclipse_error_log_message.log)

kolipakakondal, 2021-02-03:
> [eclipse_error_log_message.log](https://github.com/espressif/idf-eclipse-plugin/files/5917314/eclipse_error_log_message.log)

As per the log, your eclipse still running on Java .8.0_221 which is not supported by the plugin

```
eclipse.buildId=4.8.0.I20180611-0500
java.version=1.8.0_221
java.vendor=Oracle Corporation
```

icodk, 2021-02-03:
If I go to Help->About Eclipse IDE, then click on Copy Build id information To Clipboard
and then paste it here:
This is what I get:
```
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-12 (4.18.0)
Build id: 20201210-1552
OS: Windows 10, v.10.0, x86_64 / win32
Java version: 11
```

kolipakakondal, 2021-02-03:
> If I go to Help->About Eclipse IDE, then right click on the dialog form and select: Copy Build id information To Clipboard,
> and then paste it here:
> This is what I get:
> 
> ```
> Eclipse IDE for C/C++ Developers (includes Incubating components)
> Version: 2020-12 (4.18.0)
> Build id: 20201210-1552
> OS: Windows 10, v.10.0, x86_64 / win32
> Java version: 11
> ```

Sorry my bad, I didn't check the log file till the end, it seems to be you've tried Java 1.8 , Java 15 and Java11. I could see Java 11 in the end.

It seems to be plugin installation didn't go well because some of the dependency plugins are not satisfied. I would suggest you to go for a new installation on a new copy of Eclipse CDT 2020-12. 
https://www.eclipse.org/downloads/packages/release/2020-12/r/eclipse-ide-cc-developers

```
!SUBENTRY 1 org.eclipse.equinox.p2.director 4 10053 2021-02-01 14:48:41.580
!MESSAGE Cannot complete the install because one or more required items could not be found.
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 0 2021-02-01 14:48:41.580
!MESSAGE Software being installed: Espressif IDF Plugins for Eclipse 1.2.4.202101041022 (com.espressif.idf.feature.feature.group 1.2.4.202101041022)
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 0 2021-02-01 14:48:41.580
!MESSAGE Missing requirement: ESP-IDF Serial Monitor 1.0.0.202006021054 (com.espressif.idf.terminal.connector.serial 1.0.0.202006021054) requires 'osgi.bundle; org.eclipse.tm.terminal.control 4.5.0' but it could not be found
!SUBENTRY 2 org.eclipse.equinox.p2.director 4 1 2021-02-01 14:48:41.580
!MESSAGE Cannot satisfy dependency:
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-02-01 14:48:41.580
!MESSAGE From: Espressif IDF Plugins for Eclipse 1.2.4.202101041022 (com.espressif.idf.feature.feature.group 1.2.4.202101041022)
!SUBENTRY 3 org.eclipse.equinox.p2.director 4 0 2021-02-01 14:48:41.580
!MESSAGE To: org.eclipse.equinox.p2.iu; com.espressif.idf.terminal.connector.serial [1.0.0.202006021054,1.0.0.202006021054]

```

icodk, 2021-02-03:
Is java version 11 is the best for this plugin ?
Is eclipse version 2020-12 is the best for this plugin ?
What about idf it's self ? which is the best idf version ?
Are there other requirements that depends on specific version that I have to care about ?
Thanks

icodk, 2021-02-03:
Working !
Reinstalled Eclipse and  the plugin and now I get the ESP-IDF Tools Manager menu under help. I managed to install the tools and compile the template project. However, Eclipse installer (downloaded from your link) installed java 15.0.1   instead of 11 which I selected from the dropdown list. Now the Help-> About eclipse IDE  shows 
```
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-12 (4.18.0)
Build id: 20201210-1552
OS: Windows 10, v.10.0, x86_64 / win32
Java version: 15.0.1
```
So it probably works with 15 anyway ?
Also under Windows 10 it let you select the python version (probably auto detected) and not the python executable as in the document/readme
FYI The template generate the following warning:
```
1/9] ccache C:\Dev\esp32-idf\idf-tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/newlib/platform_include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/freertos/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/heap/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/log/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/lwip/include/apps -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/lwip/include/apps/sntp -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/lwip/lwip/src/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/lwip/port/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/lwip/port/esp32/include/arch -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/lwip/port/esp32/tcp_isn -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/soc/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/soc/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_rom/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_common/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/xtensa/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/xtensa/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/driver/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/driver/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_ringbuf/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/efuse/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/efuse/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/vfs/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_wifi/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_wifi/esp32/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_event/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_netif/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_eth/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/tcpip_adapter/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/app_trace/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/app_update/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/spi_flash/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/bootloader_support/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/mbedtls/port/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/mbedtls/mbedtls/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/nvs_flash/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/pthread/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/espcoredump/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/perfmon/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/wpa_supplicant/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/wpa_supplicant/port/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/wpa_supplicant/include/esp_supplicant -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/asio/asio/asio/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/asio/port/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/cbor/port/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/coap/port/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/coap/port/include/coap -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/coap/libcoap/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/coap/libcoap/include/coap2 -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/console -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/nghttp/port/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/nghttp/nghttp2/lib/includes -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp-tls -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_adc_cal/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_gdbstub/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/tcp_transport/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_http_client/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_http_server/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_https_ota/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/protobuf-c/protobuf-c -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/protocomm/include/common -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/protocomm/include/security -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/protocomm/include/transports -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/mdns/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_local_ctrl/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/sdmmc/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_serial_slave_link/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_websocket_client/include -IC:/Dev/esp32-idf/esp-idf-v4.1.1/components/expat/expat/ [... cut ...]
C:/Dev/esp32-idf/esp-idf-v4.1.1/components/esp_event/include/esp_event_legacy.h:231:11: note: declared here
 esp_err_t esp_event_loop_init(system_event_cb_t cb, void *ctx) __attribute__ ((deprecated));
 ```
Thanks a lot for your help. I appreciate your efforts to help me and  others.  Hope you will manage to improve this plugin in the future. As inspiration, I think you should look at STM32CubeIDE which is also an Eclipse based IDE. It is has an excellent installation and let the user concentrate on the application and less on orchestrating a development environment

kolipakakondal, 2021-02-04:
Thanks for the feedback @icodk. You can expect a better user experience and functionality going forward.

kolipakakondal, 2021-02-04:
Closing this issue, as this is to do either with the Eclipse Installation or Java version

closed 2021-02-04 by kolipakakondal