espressif/idf-eclipse-plugin #170: Different application binaries if compile with Eclipse plugin and idf.py cmd line

alezancomelit, 2020-12-21:
**Describe **
On my Windows 10 PC:
- I installed ESP-IDF via Windows installer
- I installed Eclipse IDF Plugin and point to use esp-idf already installed

If I build my project via "idf.py build" via ESP-IDF Command Prompt (cmd.exe) and via Eclipse "hammer" button, I obtain 2 different binaries of my application project.

If compare it, I see 2 difference:
1. 32 bytes from 0x000000B0 to 0x000000CF
2. last 33 bytes of binaries

About first 32 bytes, I found it's the "app_elf_sha256" of Application description structure.
But last 33 bytes? I not found any information about this.

**Environment Information:** 
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/5723205/eclipse_config.txt)

**ESP-IDF Tools Information:**
Executing C:\Python39\python.exe C:\esp\esp-idf\tools\idf_tools.py list

xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
esp-2020r2-8.2.0 (recommended, installed)
xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
esp-2020r2-8.2.0 (recommended, installed)
esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
2.28.51-esp-20191205 (recommended, installed)
esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
2.28.51-esp-20191205 (recommended, installed)
cmake: CMake build system
3.13.4 (recommended, installed)
openocd-esp32: OpenOCD for ESP32
v0.10.0-esp32-20191114 (recommended, installed)
ninja: Ninja build system
1.9.0 (recommended, installed)
idf-exe: IDF wrapper tool for Windows
1.0.1 (recommended, installed)
ccache: Ccache (compiler cache)
3.7 (recommended, installed)

**IDF Eclipse Plugin Version:** 
1.2.3.202010091505

**IDF Version:**
release 4.1

kolipakakondal, 2020-12-22:
@igrr Could you help us with this?

igrr, 2021-01-21:
I think the limitation here is that IDF doesn't provide a way at the moment to achieve reproducible builds of the .elf file. See https://github.com/espressif/esp-idf/issues/5873#issuecomment-694014340 for more context.
Because the ELF file is different, the SHA256 hash sum embedded into the binary image is also different. This affects the SH256 signature of the binary image itself, and the 1-byte checksum, which are appended at the end of the image (33 bytes).
I'll close the issue since this is not something that can be changed by IDF Eclipse plugin. If you'd like to receive notifications about the reproducible builds topic, please follow the link to that issue and click "Subscribe" button on the right.

closed 2021-01-21 by igrr