espressif/idf-eclipse-plugin #177: Eclipse 2020-12 Support ?

franz-ms-muc, 2020-12-29:
Hello, 

is it planned to support Eclipse 2020-12 ? 

if i install eclipse-embedcpp-2020-12-R-win32-x86\_64.zip it comes with new major release of the Embedded CDT plug-ins (v6.x);

the IDF Plugin still depends on the older 5.x Release CDT. 

so, i guess, need a full roundup for the Plugin to smootly run with 2020-12 right ? 

Thanks,

Franz

closed 2021-02-16 by kolipakakondal

franz-ms-muc, 2021-02-16:
> Hi @franz-ms-muc<https://github.com/franz-ms-muc>
>
> We have released v1.2.4 to support eclipse 4.18. However, this will by default recommend the old embedded plugin during the update. But you can always update to the latest and it should work smoothly.
>
> Let me know how it goes.
>
>
> [https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]
>
> Meisterschulen am Ostbahnhof
> Mühldorfstr. 6
> 81671 München
>
> Tel.: 089 416002 - 0
> Fax: 089 416002 - 111
>
> Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

Is it planned to support the Plugins that come with 2020-12?

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

kolipakakondal, 2021-02-16:
Hi @franz-ms-muc 

Our stable version 1.2.4 already supports eclipse 2020-12, and we also made 2.0.0-beta which also supports eclipse 2020-12 along with other fixes and improvements.

Please check the below release notes
1.2.4 https://github.com/espressif/idf-eclipse-plugin/releases/tag/v1.2.4

2.0.0-beta https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.0.0-beta