espressif/idf-eclipse-plugin #180: Installing tools stucked at 20%

sfeutrier, 2021-01-06:
**When Installing ESP-IDF Tools, installation is stalling**
I have check java, python and git version and it is what you recommend. All are in the PATH
I do not understand why even the simplest process are never working...

**To Reproduce**
1. Just follow the readme ^^

**ESP-IDF Tools Information:**
```
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2020r3-8.4.0 (recommended)
* riscv-none-embed-gcc: Toolchain for RISC-V (optional)
  - riscv-none-embed-gcc-8.2.0 (recommended)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended)
* cmake: CMake build system
  - 3.16.4 (recommended)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended)
* ninja: Ninja build system
  - 1.10.0 (recommended)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended)

```

**IDF Eclipse Plugin Version:** 
`ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD	ESP-IDF Eclipse Plugin	1.2.4.202101041022	com.espressif.idf.feature`

kolipakakondal, 2021-01-06:
Hi @sfeutrier 

Could you help us with the below logs to troubleshoot further
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-report-a-deadlock-or-eclipse-hang
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-access-the-error-log

sfeutrier, 2021-01-06:
Hi @kolipakakondal 

Here is the error log file
[log_eclipse.log](https://github.com/espressif/idf-eclipse-plugin/files/5774935/log_eclipse.log)

kolipakakondal, 2021-01-06:
Hi @sfeutrier 

Looking at the log I don't see any error particular to the idf plugin. However, there some issues reported on the other eclipse plugins because of the Java 15 compatibility.

What do you see on the console window while installing tools? Can you share the screenshot?

Also you've mentioned the process stuck at 20% - this might not be stuck as it shows it's actually downloading the required tools, please wait and see what happens

sfeutrier, 2021-01-06:
@kolipakakondal,

In fact it worked ! Thank you for your help.
Yesterday I was (this time really) stucked at 20% for 1 hour and thus, I thought it was the same this time.

To people having the issue : check again the **Installing Prerequisites** I had the last version of eclipse which does not support esp-idf yet and I did not have all my variables in the PATH (wheck with the control pannel).
To end this, when specifying the path of git, git.exe under Windows is in the _cmd_ folder and not in the **bin** folder.

closed 2021-01-06 by sfeutrier