espressif/idf-eclipse-plugin #187: Failed to build example project - cmake error

brainiac47, 2021-01-18:
After intalling Eclipse CDT 2020-12 and esp-idf, tried to build example 'ble_spp_client_demo' and got an error:

Building in: D:\dev\PROJECTS\spp-client\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=D:\dev\esp-idf-release-v4.1\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DIDF_TARGET=esp32 D:\dev\PROJECTS\spp-client
-- Found Git: D:/dev/Git/cmd/git.exe (found version "2.30.0.windows.2") 
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/brain/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/brain/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/brain/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/brain/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/brain/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from D:\dev\esp-idf-release-v4.1\requirements.txt are satisfied.
-- Project sdkconfig file D:/dev/PROJECTS/spp-client/sdkconfig
Loading defaults file D:/dev/PROJECTS/spp-client/sdkconfig.defaults...
CMake Error at D:/dev/esp-idf-release-v4.1/tools/cmake/component.cmake:302 (message):
  Include directory
  'D:/dev/esp-idf-release-v4.1/components/mbedtls/mbedtls/include' is not a
  directory.
Call Stack (most recent call first):
  D:/dev/esp-idf-release-v4.1/tools/cmake/component.cmake:478 (__component_add_include_dirs)
  D:/dev/esp-idf-release-v4.1/components/mbedtls/CMakeLists.txt:3 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "D:/dev/PROJECTS/spp-client/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': Не удается найти указанный файл.

Build complete (0 errors, 0 warnings): D:\dev\PROJECTS\spp-client\build

ESP-IDF Eclipse Plugin Version: 1.2.4.202101041022

[environment_info.txt](https://github.com/espressif/idf-eclipse-plugin/files/5828147/environment_info.txt)

kolipakakondal, 2021-01-18:
```
CMake Error at D:/dev/esp-idf-release-v4.1/tools/cmake/component.cmake:302 (message):
Include directory
'D:/dev/esp-idf-release-v4.1/components/mbedtls/mbedtls/include' is not a
directory.
```

This could a because of a submodules update. You can run this and see if it helps

`git submodule update --init --recursive`

kolipakakondal, 2021-02-16:
Closing this issue for now, please report us back if you see the same issue with our recent release

closed 2021-02-16 by kolipakakondal