espressif/idf-eclipse-plugin #189: Customize sdkconfig file for esp-idf

JosueSanchezEZM, 2021-01-18:
Hello, everyone I would like your help to understand how to customize sdkconfig file in order to hize component/libraries configuration options in order to show only those that I can add. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

kolipakakondal, 2021-01-18:
Hi @JosueSanchezEZM 

The sdkconfig UI editor is based on the `/build/config/kconfig_menus.json` file. This interacts with` idf.py confserver` to update the values.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#build-system-metadata
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#json-configuration-server

JosueSanchezEZM, 2021-01-18:
Thanks @kolipakakondal I will check your recommendation.

kolipakakondal, 2021-02-04:
Hi @projectgus  Any thoughts on this?

projectgus, 2021-02-04:
Hi @JosueSanchezEZM and @kolipakakondal,

I don't think this is something we support. The idea of the Kconfig system is to allow configuring of the entire ESP-IDF project, including all its components.

Is using the filter text field in Eclipse to narrow the selection to only those components enough? Or do you need it to be entirely not shown?

@JosueSanchezEZM can you explain a bit about your use case and why you need this? It seems like you might be looking to "white label" ESP-IDF into a configuration system for a particular product's firmware?

JosueSanchezEZM, 2021-02-04:
Thanks @projectgus  for your response. What I am looking for is to develop a particular product' firmware as you say. For this, I would like to "redesign" de sdkconfig file to only show my configuration options. Within these configurations, I will use espressif components variables to enable or disable them from my configuration options and to avoid using filter text field to search the specific component configuration option.

Reference Image (Edited from the first comment in this issue):
[image]

projectgus, 2021-02-04:
Hi @JosueSanchezEZM,

I'm afraid that we don't support this use case.

I think probably the best way to achieve what you want would be to edit the code for the configuration viewer in the Eclipse plugin and hard-code it so that it only shows the "EZM Configuration" menu and submenu, then build and distribute a new plugin.

Sorry we can't be of more help.

kolipakakondal, 2021-02-16:
Please let us know if you need any further help on this. Closing this issue for now.

closed 2021-02-16 by kolipakakondal