espressif/idf-eclipse-plugin #192: JTAG debugging: dynamic printf breakpoints or printf output in JTAG console

TobiasBoeschMiele, 2021-01-19:
Hello everyone,

I found that redirecting the serial console output to the host using JTAG only should be possible in general: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#logging-to-host
I haven't tried that method yet.
That works with a python script that reads out the logging data when called. Seems that this happens offline.

Is it possible to have live logging while in the debug session? If a printf statement is hit it is immediately printed in the openocd/JTAG console?

Related:
I found that dynamic printf breakpoints can be set using the plugin but they don't work. They don't print something when they were hit. This might be a separate issue but before creating the relation to this topic should be discovered. Perhaps they can be solved with the same solution.

igrr, 2021-01-20:
Hi @TobiasBoeschMiele, it should be possible with a bit of glue on the application side, please take a look at https://esp32.com/viewtopic.php?t=13997. This approach requires sending a `esp32 apptrace start con:` command to OpenOCD, which you can add to the list of GDB initialization commands (`mon esp32 apptrace start con:`).

TobiasBoeschMiele, 2021-01-21:
> Regarding dynamic printf breakpoints (dprintf), please try `set dprintf-style gdb` command before using dprintf.

Where exactly can I add this in eclipse? I guess in the debug configuration under gdb. Is this the way to go?
[image]

This setting does not seem to add dprintf functionality.

TobiasBoeschMiele, 2021-01-21:
> which you can add to the list of GDB initialization commands

how to that in eclipse? `.gdbinit` file or per debug configuration. (I would prefer via debug configuration)

TobiasBoeschMiele, 2021-01-21:
> > which you can add to the list of GDB initialization commands
> 
> How to that in eclipse? `.gdbinit` file or per debug configuration. (I would prefer via debug configuration)

I found this in your forum link (https://esp32.com/viewtopic.php?t=13997):
>You can run OpenOCD with the following extra argument:
>-c "init; esp32 apptrace start con:"

I'll try that.

TobiasBoeschMiele, 2021-01-21:
Regarding logging to JTAG:

The OpenOCD command was added to the debug configuration: `-c "init; esp32 apptrace start con:"`
App tracing was enabled in the menuconfig.
App tracing code was added to the application:
```
    char buf[] = "Hello World!";
    volatile esp_err_t res = esp_apptrace_write(ESP_APPTRACE_DEST_TRAX, buf, strlen(buf), 1000);
```

When this output is reached there is no output in the debug console in eclipse. Return value of the `esp_apptrace_write` is `0` 
`ESP_OK          0       /*!< esp_err_t value indicating success (no error) */`.

`esp_apptrace_host_is_connected(ESP_APPTRACE_DEST_TRAX)` shows there is no host connection.

Where to start finding the cause for the missing logging output?

igrr, 2021-01-21:
> Where exactly can I add this in eclipse? I guess in the debug configuration under gdb. Is this the way to go?

`set dprintf-style gdb` should go into the "commands" edit box. You can add other commands you would otherwise add to the .gdbinit file into that box.

Regarding logging to JTAG, couple of things to check:

* do you see any error messages in the OpenOCD output?
* do you also have a call to esp_apptrace_flush after esp_apptrace_write, similar to what is done in`apptrace_writefn` in the example posted on the forum?
* can you try moving `esp32 apptrace start con:` command from the OpenOCD command line to the .gdbinit script (or the "Commands" edit box in Eclipse debug configuration? To pass commands to OpenOCD you need to add `mon`, so the full command would look like `mon esp32 apptrace start con:`.

TobiasBoeschMiele, 2021-01-21:
### dynamic printf
> `set dprintf-style gdb` should go into the "commands" edit box. You can add other commands you would otherwise add to the .gdbinit file into that box.

Added the command as a new line to the eclipse debug configuration command box. It looks like this now:
```
set mem inaccessible-by-default off
set dprintf-style gdb
```
That doesn't make dprintf breakpoints work. Does it work on your machine?

### Logging to JTAG

> * do you also have a call to esp_apptrace_flush after esp_apptrace_write, similar to what is done in`apptrace_writefn` in the example posted on the forum?

**No... now I've added it and there is the output. So Logging to JTAG is solved I think.** Many thanks. I tought I also add this funciton for testing earlier, but I saw no reason under [Application Specific Tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#app-trace-application-specific-tracing) why flushing is needed in this case. Maybe one can change the docs if flushing is needed in general after `esp_apptrace_write` or explain here why it is needed in this case.

> * can you try moving `esp32 apptrace start con:` command from the OpenOCD command line to the .gdbinit script (or the "Commands" edit box in Eclipse debug configuration? To pass commands to OpenOCD you need to add `mon`, so the full command would look like `mon esp32 apptrace start con:`.

That makes the debug session to not fully start anymore. eclipse is still configuring gdb, but doesn't return within 30 seconds or so. (Far too long)
I like that approach. If you know how to add this command in the gdb command box correctly, please let me know.