espressif/idf-eclipse-plugin #194: Can't find include file in  IDF-plugin (Version: 1.2.4.202101041022)

icodk, 2021-02-05:
An header file  inside a Component can't be found by the build system.
(it could before I upgraded the plugin from this version https://www.youtube.com/watch?v=DqnZy4d03_s)
My project has a Component folder with different components and their header files.
In main.c (which is in the main folder) there is an `#include "network.h"` statement. 
If I right click  and select Open Declaration the editor can find the file. However when I build the project, I get 
`fatal error: network.h: No such file or directory	main.c	/myProject/main	line 54	C/C++ Problem`
(and this all worked in the older plugin)

Also, clicking on the Build hammer does not shows any error (or anything else) in the console tab, the error is shown in the Problem tab if it is open
Clicking on  Run (green icon with a white triangle) shown the error in the console.

**Environment Information:** 
[InstallationDetails_Configuration.txt](https://github.com/espressif/idf-eclipse-plugin/files/5931647/InstallationDetails_Configuration.txt)

**ESP-IDF Tools Information:**
Clicking on Help > `Espressif Tools Manager > List Installed Tools`
nothing happens. Tried several times

**IDF Eclipse Plugin Version:** 
`  Espressif IDF Plugins for Eclipse	1.2.4.202101041022	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD
`
Eclipse Version:
```
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-12 (4.18.0)
Build id: 20201210-1552
OS: Windows 10, v.10.0, x86_64 / win32
Java version: 15.0.1
```
So there are 3 issues here:
1. include file could not be found
2.  No console output when selecting the  build hammer
3. Help->ESP-IDF Tools Manager -> List Installed Tools does not show anything

r2d2r01, 2021-03-08:
hi, can u try in a shemm (cmd) **python.exe %IDF_PATH%\tools\idf_tools.py list**

C:\Users\root>**python.exe %IDF_PATH%\tools\idf_tools.py list**
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

C:\Users\root>python -V
Python 3.9.2

kolipakakondal, 2022-02-14:
Hello,

We have fixed unresolved headers and indexer problems with the v2.4.0 release, please get the latest changes using the update site and see if that helps.

https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

We also have the v2.4.1 patch release.

Thank you

closed 2022-02-14 by kolipakakondal