espressif/idf-eclipse-plugin #195: Java version confusion

icodk, 2021-02-06:
**Describe the bug**
As mentioned in  https://github.com/espressif/idf-eclipse-plugin/issues/169, there is an issue with java version for the current plugin version ( at  this time 1.2.4.202101041022) 
There are two "java versions" that might affect the functionality of the plugin (am I missing something here?)
1. The java version  of the eclipse installation
Eclipse 2020-06 shows (Help->About Eclipse->Copy Build information Id To Clipboard) 
```
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-06 (4.16.0)
Build id: 20200615-1200
OS: Windows 10, v.10.0, x86_64 / win32
Java version: 11.0.10
```

While Eclipse  2020-12:
```
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-12 (4.18.0)
Build id: 20201210-1552
OS: Windows 10, v.10.0, x86_64 / win32
Java version: 15.0.1
```
However  Eclipse 2020-12 sometimes shows also 11.0.10 and I am not sure what makes the different.

2. The "default" java version on the computer (JAVA_HOME path)
Is this also effect the plugin functionality ?

At the moment I have only java 11 on the machine and two Eclipse, 2020-06 and 2020-12 but both shows java 11
and almost everything works as it should. I can compile the examples from the IDF but Clicking on` Help->ESP-IDF Tools Manager->List Installed Tools` 
is not showing anything( and I have only java 11 as far as I can see)

IDF Installation issue:
in the IDF Tools Installation dialog I have to enter the  ESP-IDF directory. However, the installation uses the IDF_PATH variable.
So if I install a new IDF version in a different directory I must change the path in IDF_PATH in order to succeed the tools installation. It might be a better solution that the idf path will be taken from the path of the selected idf installation.

kolipakakondal, 2021-02-11:
Hi @icodk 

The way the eclipse picks up the JRE is using the system's PATH, other way what do you get to see if run "java -version" from the terminal window. To eclipse versions showing different Java versions is not possible unless eclipse.ini is configured with a different Java version using "-vm" argument

>  Help->ESP-IDF Tools Manager->List Installed Tools
> is not showing anything( and I have only java 11 as far as I can see)

Could you share an error log for this, you can refer https://github.com/espressif/idf-eclipse-plugin#error-log

> IDF Installation issue:
> in the IDF Tools Installation dialog I have to enter the ESP-IDF directory. However, the installation uses the IDF_PATH variable.
> So if I install a new IDF version in a [...]

Will verify and revert back to you on this. As far as I remembered, the IDF Tools installation dialog is launched with the newly installed esp-idf directory if you have installed esp-idf also from eclipse

kolipakakondal, 2021-04-26:
Updating of IDF_PATH and IDF_PYTHON_ENV_PATH is fixed with the latest release, please update to the latest
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.1.0

closed 2021-06-04 by kolipakakondal