espressif/idf-eclipse-plugin #198: Some Python dependencies must be installed

mt-tronic, 2021-02-10:
Hello team,

I have follow all steps on this page : https://github.com/espressif/idf-eclipse-plugin
Created new project based on hello_world example
When I try to build I have the errors below  : 
I have also run install.bat

my os : Windows 10 pro
Python 3.9.1
git version 2.30.1.windows.1
java 15.0.2 2021-01-19
Java(TM) SE Runtime Environment (build 15.0.2+7-27)
Java HotSpot(TM) 64-Bit Server VM (build 15.0.2+7-27, mixed mode, sharing)

CDT Build Console

Building in: C:\Users\Mickael\eclipse-workspace\test\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\Mickael\esp\esp-idf\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DIDF_TARGET=esp32 C:\Users\Mickael\eclipse-workspace\test
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "C:\Users\Mickael\esp\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\Mickael\.espressif\python_env\idf4.2_py3.9_env
    Python interpreter used: C:\Users\Mickael\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
CMake Error at C:/Users/Mickael/esp/esp-idf/tools/cmake/build.cmake:271 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/Mickael/esp/esp-idf/tools/cmake/build.cmake:397 (__build_check_python)
  C:/Users/Mickael/esp/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Mickael/eclipse-workspace/test/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': Le fichier spécifié est introuvable.

Build complete (0 errors, 0 warnings): C:\Users\Mickael\eclipse-workspace\test\build

kolipakakondal, 2021-02-11:
@igrr @projectgus Any thoughts on why are we getting this issue, I heard from a couple of other users as well. Any diagnostic information we can ask for?

projectgus, 2021-02-11:
@mt-tronic 

> I have also run install.bat

Could you post the output from running install.bat, please? Maybe this process is picking up a different Python executable, so the packages are already installed but for a different version than Python 3.9?

mt-tronic, 2021-02-11:
Hello @projectgus 

In attachement the output from install.bat.
As you can see teir is a big ERROR block.

[install_output.log](https://github.com/espressif/idf-eclipse-plugin/files/5963981/install_output.log)

projectgus, 2021-02-11:
Hi @mt-tronic,

Thanks for the update. This error is most likely why the later step failed, Python has failed to install dependencies.

As far as I know, installing this package shouldn't require any additional dependencies in Windows. I'm going to check with some colleagues about possible root causes for this.

mt-tronic, 2021-02-13:
Hi @projectgus 

By removing Python 3.9 and installing 3.8 it's work fine.

By adding PIP_EXTRA_INDEX_URL -> not work
By installing https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe -> not work

Many thanks

projectgus, 2021-02-14:
Thanks for the update, @mt-tronic . Will check on those other fixes with my colleague. In any case, I'm glad you got it working.

georgik, 2021-02-15:
@mt-tronic Thank you for reporting the issue. 

The solution with installing Python 3.8 works, because this version of Python has gevent and some other wheels ready at pypi.
The solution with installing Python 3.9 does not work, because some wheels are not built by maintainers and requires Visual Studio C++ installed to be compiled.

We pre-built wheels for Python 3.9 and published here https://dl.espressif.com/pypi as mentioned above. 

I'm curious why solutions with PIP_EXTRA_INDEX_URL and https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe  does not work. @mt-tronic My I ask you for additional diagnostic? Please, run the installer and post the log from installer. You can find it in ```C:\Users\Username\AppData\Local\Temp``` with filename ```Setup Log DATE.txt``.

Thank you very much.

miLORD1337, 2021-03-07:
@georgik FWIW, your suggested workaround using `PIP_EXTRA_INDEX_URL` did work for me. 

Thanks!

daverathbone, 2021-04-26:
From command line idf.py -p COM7 flash works. (Some times...)
From Eclipse  builds then .... (I have python in path. (text below) 
From Visual Studio Code .. Includes paths can't be found despite JSON paths being correct in VSC.

Why cant you code in C or C++ ? for a company that make smart chips your "helper" code is dumb! trying to fudge Eclipse and Visual Studio to call linker and complier with broken "smart" console scripts. 
OS dependent C/C++ code for Windows and Linux  is no big deal! ST,TI, Infineon, AVR and Microchip just works in these environments.   Mad that you need freertos for a timer and UART  in "Blink".  -Less is more.

I'm having to use Notepad++ with my own scripts for linker and compiler (Crazy!)
For commercial applications, as I can not spend each day fixing Espressif Hobby coder team.

BUG.
The following Python requirements are not satisfied:

ecdsa>=0.16.0
To install the missing packages, please run "C:\Users\Dave\esp\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\Dave\.espressif\python_env\idf4.1_py3.7_env
    Python interpreter used: C:\Users\Dave\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe
ESP-IDF v4.2-dirty

georgik, 2021-04-26:
@daverathbone Thank you for reporting the problem.

The ecosystem of tools around ESP-IDF is built on Python. The Python is used in tooling since the early days of ESP-IDF. Many open-source Python tools are contributed either by Espressif or by the community. We're aware of the problems with Python on Windows. Windows is one of the most challenging platforms in this context.

To resolve several problems with Windows we've updated the Tool installer which now contains Embedded Python and Embedded Git which avoid collision with other Pythons or Anacondas. The new installer contains also improved integration with Eclipse. 

Please use the following installer 2.7+ to update your environment:
https://dl.espressif.com/dl/esp-idf/

Regarding runtime for tools, we're testing the approach with Rust which allows building self-contained binaries which do not require installation of runtime like Python.

Regarding Visual Studio Code. Please try to update your environment with the installer. If the problem with Visual Studio Code is not resolved, please open a new issue in the project https://github.com/espressif/vscode-esp-idf-extension

JKSAR, 2021-07-24:
I'm having the same problem with Linux:

Ubuntu 20.04.2
Eclipse Version: 2020-12 (4.18.0)
openjdk 11.0.11 2021-04-20
esp-idf-v4.0.3
python 3.8.10
git version 2.25.1

Already set environment variable PIP_EXTRA_INDEX_URL with value https://dl.espressif.com/pypi

Any idea?

Thanks!

JKSAR, 2021-07-24:
> I'm having the same problem with Linux:
> 
> Ubuntu 20.04.2
> Eclipse Version: 2020-12 (4.18.0)
> openjdk 11.0.11 2021-04-20
> esp-idf-v4.0.3
> python 3.8.10
> git version 2.25.1
> 
> Already set environment variable PIP_EXTRA_INDEX_URL with value https://dl.espressif.com/pypi
> 
> Any idea?
> 
> Thanks!

I managed to solve all the problems I was having as follows:

1. I installed Linux (Ubuntu) from scratch

2. I carefully performed ALL the steps described in the following link:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-2-get-esp- idf

**OBS1:** in this step I had not installed neither the JDK, nor Eclipse nor the Espressif plugin. I recommend that the installation of the JDK, Eclipse and especially the Espressif plugin be performed AFTER performing all the steps in item 2 (above)

Once you've finished all the steps, you'll be able to generate the 'hello_world' example, connect to the device (via Putty), compile, upload to ESP32 and run, all without having installed Eclipse yet

**OBS2.:** in this step you will have to install the ESP-IDF package, so in the future when you install Eclipse and the plugin, you should not install this package again

3. If you were successful in the previous step, follow the instrucitons to intall Espressif Plugin (https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md)
Install JDK (recommend openjdk-11-jdk) Eclipse (recommend 2020-12) and finally, install the Espressif plugin. 

When configuring the ESP-IDF package, choose the option _"Check Use an existing ESP-IDF directory from the file system"_ and then _"Choose an existing ESP-IDF directory from the file system"_. Select the installation folder of the ESP-IDF package you performed in step 2 (in my case the folder is /home/<usr>/esp/esp-idf)

After everything is installed, try creating an Espressif IDF Project in Eclipse as a copy of the "Hello World" example.

Best Regards,

Julio Cesar