espressif/idf-eclipse-plugin #199: still 'Build not configured correctly' bug in v2.0.0-beta

steph0xr, 2021-02-12:
**Describe the bug**
In some cases, where more than one project is open and closed, 'Build not configured correctly' message still happens, even if "Launch target" and "Launch configuration" are configured correctly. Only an Eclipse restart fix the problem.

**To Reproduce**
Steps to reproduce the behavior:
1. open eclipse
2. create a new esp-idf project called project1, and build it.
2. right click on project1 and click on 'close project'.
3. create a new esp-idf project called project2, and build it.
4. right click on project2 and click on 'close project'.
5. Now right click on project1 and click on 'open project', to reopen the first created project.
6. modify "Launch Configuration" to choose the project1 configuration.
7. build project1 again: now is not possible to build because 'Build not configured correctly' message happens.
8. From this point is not possible to build neither project1 or project2 again. The only solution to build project1 for example, is to open it and set up right "launch configuration", then close Eclipse and reopen it with the configurations previously set.

**Expected behavior**
I expect that closing a project and opening another project in my workspace it's possible to build it without close and re-open eclipse.

**Environment Information:** 
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-12 (4.18.0)
Build id: 20201210-1552

**ESP-IDF Tools Information:**
Executing python /hdd1/esp/esp-idf-v4.2/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)

**IDF Eclipse Plugin Version:** 
Version: 2.0.0.202102021949

kolipakakondal, 2021-02-18:
Hi @steph0xr 

I couldn't reproduce the issue with the exact steps which you've mentioned in the ticket, did you anything different?

kolipakakondal, 2021-02-18:
@sigmaaa Could you confirm if you're able to reproduce this issue with 2.0.0-beta?

sigmaaa, 2021-02-18:
@kolipakakondal I'm tried to reproduce it on macOS and Windows 10. This problem happens only on macOS for me.

steph0xr, 2021-02-18:
@kolipakakondal I've reproduced the bug on Ubuntu. I can also try on windows10.

ranbros, 2021-02-18:
I have the same issues with eclipse 2020-09 on windows10 and ubuntu.
How I can set up right "launch configuration"?

steph0xr, 2021-02-19:
@ranbros the last 2.0.0 version must be run with eclipse 2020-12 or at least updated CDT 10.1.0. Seems that on Windows10 it works, but not in linux and macOS.

ranbros, 2021-02-19:
When I create a new project it works. I can build and flash.
When I close the project and re-open it gives me "Build not configure correctly".
I have tryed also last 2.0.0 and eclipse ctd 2020-12 but is still the same problem.
Is there some setup on lauch configuration to set for solve issues?
I have tryed eclipse 2020-12 2020-9 on windows 10 and ubuntu. Plugin beta and stable always with same issues.
On Esp-idf all works fine.

kolipakakondal, 2021-03-05:
This is fixed with the 2.0.0 release, please get the latest build to verify

closed 2021-03-05 by kolipakakondal