espressif/idf-eclipse-plugin #202: FTDI installation instructions for macOS Big Sur with ESP-DEVKIT-V4

ammaree, 2021-02-19:
I am trying to follow the instructions for installing the FTDI drivers (2.4.4) but the instructions fail at the point where it comes to install the driver, complaining that it needs to be "approved" in System Preferences.

Having done some homework it seems that Big Sur has Apple drivers preinstalled and that both ports on the FT2232 have USB Serial drivers loaded.

From what I understand the JTAG ports needs to be freed (USB Serial driver removed) up so it can used by OpenOCD with the basic FTDI driver.

I assume the instructions were prepared for an older version of macOS, hence the failure. Can updated instructions please be made available for Big Sur?

kolipakakondal, 2021-03-05:
@projectgus @projectgus Could you please check if anything has to be updated.

ammaree, 2021-03-07:
After the last BigSur update the device is now reflected differently again. As in the attached screenshot each of the 2 serial ports are now reflected as 4 separate and differently addressed devices, 8 devices in total.
The normal UART port on my ESP-DEVKIT-V4 is represented by -00003014B and -14301 entries.
[image]

/dev/cu.usbserial-00003014B starts and works as per normal:
/dev/tty.usbserial-00003014B starts but warns about hanging if gdb launched, revert to /dev/cu version:
[image]

/dev/tty.usbserial-14301 and /dev/cu.usbserial-14301 both start, issues same warning but then crashes with following message:
[image]

So, out of all the options presented, how do I get the FTDI port work with OpenOCD?

kolipakakondal, 2021-03-10:
Hi @ammaree Thanks for the update.

Here is the link for OpenOCD debugging https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/OpenOCD%20Debugging.md

ammaree, 2021-03-23:
Hi @kolipakakondal @projectgus 

I updated macOS to the latest version 11.2.3, I also checked that the FTDI driver installed is the unmodified version with the original FT2232H_A and FT2232H_B entries.
[image]

I then did a clean install, all the latest tools on Eclipse 2021-03, followed with the instruction linked above to complete the installation and configuration.

However, when I launch the debug configuration it still fails consistently with the following output.
[image]

[image]

Both FTDI posts a & B are shown in the device listing.
[image]

The device can be flashed & monitored using UART port addressed with "-p /dev/cu.usbserial-1442301"

What is currently not clear includes:
a) Must the FTDI driver config still me modified or not?
b) if it is to be modified, can updated instructions please be provided for the latest version of macOS?
c) If not, based on the debug output provided here, what is wrong with my config?

Please, I have wasted a huge amount of time having to debug using printf and cannot continue like this.

igrr, 2021-03-25:
From the log, it seems that Eclipse is trying to do "load <filename>.elf". @kolipakakondal we need to remove the checkbox "Load executable" in https://github.com/espressif/idf-eclipse-plugin/raw/master/docs/images/OpenOCDDebug_7.png.

@ammaree Please try removing that checkbox.

kolipakakondal, 2021-04-05:
By default, the "Load executable" option is disabled with the eclipse plugin. It seems to be screenshot is not updated. I just updated the screenshot as well https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/images/OpenOCDDebug_7.png

kolipakakondal, 2021-06-04:
Closing this issue as the document is updated.

closed 2021-06-04 by kolipakakondal