espressif/idf-eclipse-plugin #203: Eclipse IDF "Build not configured correctly"

SunnyRuss, 2021-02-22:
I am attempting to create a project in Eclipse.  The project creates without an error.

When I try to build it, I get "Build not configured correctly".

Steps:
1. MD \workspace
2. MD \workspace\project
3. CD \workspace\project
4. XCOPY /E <git-clone-dir)\examples\protocols\http_server\restful_server\*
5. I can use IDF.PY fullclean build to generate a .BIN file (so I assume that IDF.PY is working correctly)
6. CD \workspace
7. RMDIR /Q /S project
8. MD project
9. XCOPY /E <git-clone-dir>\examples\protocols\http_server\restful_server\*
10. Start Eclipse
11. Select C:\workspace as the workspace directory
12. Import Espressif project (name it project and select C:\workspace\project as its directory)(do not copy as it is already copied)
13. Seems to import without error
14. When I try to do anything it says there is no tool chain

What do I need to change?

IDF 4.2

sigmaaa, 2021-02-26:
Hi @SunnyRuss. 

Can you provide a list of installed tools, please? 
To do this, in eclipse, click on Help -> ESP-IDF Tools Manager -> List installed tools.

Also, check if environment variables and configuration tools are set. Thank you
https://github.com/espressif/idf-eclipse-plugin#configuring-environment-variables

SunnyRuss, 2021-02-26:
I have uninstalled and reinstalled everything I could think of, I even did this multiple times.  As before IDF.PY BUILD works as one would expect (all of the IDF.PY commands work as expected).  I installed the plugin into eclipse and imported a project I created (didn't copy from examples) and eclipse works as one would expect.  However, when I copy the RESTFUL_SERVER project from examples and import that project into eclipse I get this:

No esp launch target found. Please create/select the correct 'Launch Target'

when I right-click on the project and select "Build Project".  However, if I click on the "Hammer" (build) icon, the project builds.  If I right-click on the the project and select "Clean Project", it "cleans" the project as one would expect.  Most of the items in the right-click drop-down menu fail or are grayed out (unselectable).

I copied the "working" project to another directory and imported it again as a different name, it fails just like the RESTFUL_SERVER project I copied.

To answer your question, here the installed tools:

Executing C:\Python39\python.exe C:\Sunsight\esp\esp-idf-v4.2\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

sigmaaa, 2021-02-26:
@SunnyRuss thank you for the detailed answer. I'm able to reproduce this bug only on an older plugin's version. I think this problem is fixed in version 2.0.0-beta.
Can you try it and say if it works for you, please? https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.0.0-beta

SunnyRuss, 2021-02-28:
You are correct, using 2.0.0-beta of the plugin fixes the problem I reported.

Thanks
-Russ

closed 2021-02-28 by SunnyRuss