espressif/idf-eclipse-plugin #206: Eclipse setup instructions incorrect - unable to build project

jonathan84clark, 2021-02-24:
**Describe the bug**
Going through the process of setting up an environment with eclipse I get build failures, particularly the following 
'No Toolchain found for Target Local'

**To Reproduce**
Steps to reproduce the behavior:
1. Go through the steps in https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf-tools document
2. In the "Configuring launch target" section there is no new widget, furthermore there is no 'ESP Target' dialog or options
3. Go to Project -> Build all
4. Build fails

I attempted (unsuccessfully) to correct the issue using the following forums
https://www.esp32.com/viewtopic.php?t=15117
https://github.com/espressif/idf-eclipse-plugin/issues/119

**Expected behavior**
I expect the build to succeed on both an empty project as well as a hello_world template

**Environment Information:** 
[eclipseConfig.txt](https://github.com/espressif/idf-eclipse-plugin/files/6039262/eclipseConfig.txt)

**ESP-IDF Tools Information:**
Executing C:\Users\Clark\AppData\Local\Programs\Python\Python39\python.exe C:\Users\Clark\Documents\Esp32Stuff\esp-idf-v4.2\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	1.2.4.202101041022	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

Any help on this would be great.

Thanks.

Jonathan L Clark
Software Engineer
Special Services Group LLC

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

kolipakakondal, 2021-02-26:
Hi @jonathan84clark 

Are you able to see the Eclipse CDT toolbar, something like below?
[image]

If the toolbar itself is not available, there could be some issue with the Eclipse CDT installation. You can get a fresh copy of the Eclipse CDT package from here https://www.eclipse.org/downloads/packages/release/2020-12/r/eclipse-ide-cc-developers

kolipakakondal, 2021-06-04:
I believe this got fixed after updating to the Eclipse CPP as mentioned above. Hence closing the issue

closed 2021-06-04 by kolipakakondal