espressif/idf-eclipse-plugin #208: ninja failed with exit code 1

rezguiichrak, 2021-02-25:
hello,
I Have ninja build error how can I fix this issue
I have installed:
eclipse 20.09
Python 3.8
java:default jdk-11
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/ 
ESP-IDF :master version installed from eclipse 

------------------------------------------------------------------------console----------------------------------------------------------------------------

[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cd /home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py && /home/admina/.espressif/tools/cmake/3.16.4/bin/cmake -D IDF_PATH="/home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf" -D SERIAL_TOOL="python /home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/home/admina/eclipse-workspace/hello-world/build" -P /home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 app-template.bin
esptool.py v3.1-dev
Serial port /dev/ttyUSB0

Traceback (most recent call last):
  File "/home/admina/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py", line 4290, in <module>
    _main()
  File "/home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py", line 4283, in _main
    main()
  File "/home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py", line 3864, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py", line 118, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "/home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py", line 304, in __init__
    self._port = serial.serial_for_url(port)
  File "/home/admina/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/home/admina/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
CMake Error at run_serial_tool.cmake:50 (message):
  python
  /home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash 
cd /home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py && /home/admina/.espressif/tools/cmake/3.16.4/bin/cmake -D IDF_PATH="/home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf" -D SERIAL_TOOL="python /home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/home/admina/eclipse-workspace/hello-world/build" -P /home/admina/eclipse-cpp-2020-09-R-linux-gtk-x86_64/eclipse/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
Executing action: flash
Running ninja in directory /home/admina/eclipse-workspace/hello-world/build
Executing "ninja flash"...
ninja failed with exit code 1

kolipakakondal, 2021-02-26:
Hi @rezguiichrak 

Could you this out https://github.com/espressif/esp-idf/issues/818

rezguiichrak, 2021-02-26:
hi @kolipakakondal  
I solve this issue by adding my user to dialout group
using this command ::
$ sudo adduser username dialout 
then I restart my laptop

kolipakakondal, 2021-02-26:
Hi @rezguiichrak  I'm glad it was helpful

closed 2021-02-26 by kolipakakondal

italijancic, 2021-08-23:
> sudo adduser username dialout
 
How can i do this in MacOS ?

Hubidrei, 2022-02-21:
How can i solve this on Win10?
I use Eclipse 4.19.0 with IDF4.4