espressif/idf-eclipse-plugin #209: Tools installation fail in latest version of Eclipse plugin in Windows 10

60north, 2021-02-26:
Tools installation fail, due to the incorrect slash in paths

**To Reproduce**
Install Eclipse.
Point plugin to use the existing installation of IDF ( v 4.0)
Try to install tools.
[image]

Error:
Error: [Errno 22] Invalid argument: ' C:\\Users\\fedor\\Documents\\Projects\\esp-idf\\tools/tools.json'
etc
[image]
[image]

<https://cl1p.net/espidfproblem >

kolipakakondal, 2021-03-16:
HI @60north 

Could you check the existence of this file or location `C:\Users\fedor\Documents\Projects\esp-idf\tools/tools.json`

alizeyni, 2021-03-16:
Bitcoin

60north, 2021-03-16:
> HI @60north
> 
> Could you check the existence of this file or location `C:\Users\fedor\Documents\Projects\esp-idf\tools/tools.json`

You do not notice anything wrong in your question?
Of course the file is there, but the slashes in the path are not correct.

There is a bug in script somewhere that is handling slashes incorrectly and developers do not care.
The plugin is broken.
Hello, Espressif? are you there?

kolipakakondal, 2021-03-16:
Hi @projectgus Could you please check this?

igrr, 2021-03-16:
Mixed forward/backward slashes in paths are actually handled correctly by Python on Windows, so `C:\\Users\\fedor\\Documents\\Projects\\esp-idf\\tools/tools.json` is not a problem by itself.

However in the error message, after `OSError: [Errno 22] Invalid argument: ` note that there is a space between opening quote and `C:`. Is it possible that a space got added somehow before IDF_PATH value?

georgik, 2021-03-16:
@60north Thank you for detailed report. Please, check the extra white character mentioned by @igrr .

Meanwhile, let me suggest an alternative solution. I've seen that you're using Python 3.9 from Windows Store which might cause some problems. We prepared new installer which contains specific version of Embedded Python and pre-configured Eclipse with IDF plugin which is deployed to ~\.espressif\tools directory.
You can find 2.6 beta installers here:
- online version - 2.1 MB - https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.6.exe
- offline version - 1.2 GB (IDF 4.1.1, 4.2) - https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.6.exe

kolipakakondal, 2021-08-18:
Marking this issue as inactive and closing. However, if you have any other issues please open a new issue after verifying with our new builds

Please check our recent version here https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0

closed 2021-08-18 by kolipakakondal