espressif/idf-eclipse-plugin #215: CMD.EXE does not support UNC paths - How to change to PowerShell intead ?

kifantidis, 2021-03-05:
Hello there !

I'm posting some specifications for my desktop and SW:
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-06 (4.16.0)
Build id: 20200615-1200
OS: Windows 10, v.10.0, x86_64 / win32
Java version: 11.0.8
ESP-IDF Eclipse Plugin 1.2.3.202010091505
NAS drive is mapped into my pc and working with all my other programs. 

I have stored all my work files in a NAS storage in order to have automated back-ups and file versioning features. But for a quite time now I can not do this for my esp32 projects because when I'm trying to compile my project I get this error message:
> CMD.EXE was started with the above path as the current directory.
> UNC paths are not supported.  Defaulting to Windows directory.

Is there a way to change the kerner into using PowerShell instead ? After some searching in the web I've found that PowerShell does not have this limitation and I would like to give it a try.

Regards, Kostas.

kolipakakondal, 2021-03-05:
Hi @igrr @projectgus Any pointers on this?

igrr, 2021-03-05:
@kifantidis Are you able to build ESP-IDF projects from Powershell prompt, when using UNC paths? Instructions for Powershell are pretty similar to cmd.exe, except `.ps1` scripts are used instead of `.bat`: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#id2.

If that works for you from the command line, we can see about supporting this use case in Eclipse.

That said, the build process reads and writes quite a lot of files, so I expect building from a NAS share to be much much slower than even building from an HDD. For version control, you may consider using git, and push the code periodically to a Git repository on your NAS; just as a suggestion.

kifantidis, 2021-03-05:
Thank you for the fast response to my request. 
> That said, the build process reads and writes quite a lot of files, so I expect building from a NAS share to be much much slower than even building from an HDD

I'm using company's NAS (a synology one) and already designing PCBs + 3D models without delay. A colleague of mine is compiling for ST MCUs without problem. He mentioned that it is a bit slower but not that much as an HDD. 
I'm going to try the PowerShell prompt as soon as I get home I'll let you know about the results. 

Best regards, Kostas.

r2d2r01, 2021-03-08:
Map your UNC on a drive to see ..

kifantidis, 2021-03-09:
@r2d2r01 I've have already mapped my NAS on a drive. It does not work.

r2d2r01, 2021-03-09:
Agree with igrr in case of performance (depending local hdd type ...)
Equal to u but in my config (2x syno 3611 6gbram/tiering 2xsdd the rest in hdd 900mb) mounted in ha over Gbeth
All my eclipses projects are in a workspace through raw lun/iscsi connector & classic shares on somes volumes.
No problem on w10/eclipse 2019/esp-idf 3.3/make/msys2 that's work fine and speed
Precision, only local security , no windows security (ad,..)

kifantidis, 2021-03-11:
Good morning !!

My setup is w10/ eclipse 2020-06/ eps-idf 4.1.1/ esp-idf pluggin for eclipse. And the tools provided from there are running with windows command line and not with msys2. The problem in my setup is that windows command line does not support UNC paths from mapped drives on the other hand msys2 support UNC paths as PowerShell does too as I've read in many threads.

Because I'm inexperienced with these stuff I can not find how to change the paths and the settings of the tool-set in order to use other terminal programms for compilation etc.

I hope I find some time today in order to check igrr's suggestion. 

Regards, Kostas.

lsd-techno, 2022-08-17:
Any updates on that?
I'm trying to start `idf.py menuconfig` on a mapped drive bring me into same trouble.
`UNC paths are not supported.  Defaulting to Windows directory.`

Adk90, 2022-09-14:
Using ESP-IDF (Config Editor) in Visual Studio is causing the same problem.  Building and running works.

lucumon, 2025-05-08:
Any news on this?