espressif/idf-eclipse-plugin #218: Unable to open library source files

m-ymn, 2021-03-06:
**Only header files are opening inside eclipse**

After project is built i'm only seeing function declarations in header files instead of function definition in source files.
When i click on any library function,i should be able to open the function definition in source file inside ide.

I'm new to this, please advise if there is any workaround for it.

HrMitrev, 2021-11-25:
Hi
I highly anticipate if there will be any updates on this. For now linking the components folder is not really cutting it. Even if I thinker with the indexer settings and hide the irrelevant esp soc's folders the outcome is not great.

kolipakakondal, 2021-11-25:
Hi @HrMitrev 

Could you take this [dev-build](https://github.com/espressif/idf-eclipse-plugin/suites/4461171242/artifacts/118542971) and let us know how it goes. It will be available in the next release.

How to install this build? Please refer [here](https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile)

What is the expected behavior?
-  By default `esp_idf_components` folder will be created in the project after the build
- All esp components source files will be indexed so that you will be able to navigate to the source definitions by pressing `F3` or `Ctrl+click link` options
- No header resolving issues will be shown either in your project source code or linked project source code

Appreciate if you could provide feedback on this.

HrMitrev, 2021-11-25:
Hello again

What do you mean by "No header resolving issues will be shown either in your project source code or linked project source code"? Header resolving issues are important I think. Not sure what the impact of not showing them would be.

I installed your dev build and the folder is created but it's empty.
[image]
[image]

Also coud you try to add resource filters in the expected behaviour? I mean to exclude all folders of the soc's you're not building for. For example I'm building for esp32 and that's the resource filters I use:
[image]

kolipakakondal, 2021-11-25:
Sorry, I meant to say we have fixed header "Unresolved inclusion" issues with this.

Could you delete a build folder and rebuild the project? Ideally, it should link all project dependent esp components to `esp_idf_componenets` folder

Please also share the error log it would help us to analyze the issue. https://github.com/espressif/idf-eclipse-plugin#error-log

> Also coud you try to add resource filters in the expected behaviour? I mean to exclude all folders of the soc's you're not building for. For example I'm building for esp32 and that's the resource filters I use:

 regarding this, we link files based on the compile_commands.json file and this should be having includes only related to a specific board the one used to build a project.

resource filters are helpful if you are adding it manually

HrMitrev, 2021-12-01:
```

!ENTRY org.eclipse.egit.core 4 0 2021-12-01 14:23:42.934
!MESSAGE Failed finding RepositoryMapping
!STACK 1
org.eclipse.core.internal.resources.ResourceException(/weatherstation/esp_idf_components)[369]: java.lang.Exception: Resource '/weatherstation/esp_idf_components' is not local.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	at org.eclipse.core.internal.resources.Resource.checkLocal(Resource.java:342)
	at org.eclipse.core.internal.resources.Resource.checkAccessibleAndLocal(Resource.java:211)
	at org.eclipse.core.internal.resources.Resource.getSessionProperty(Resource.java:1182)
	at org.eclipse.egit.core.project.GitProjectData.getRepositoryMapping(GitProjectData.java:568)
	at org.eclipse.egit.core.project.RepositoryMapping.findMapping(RepositoryMapping.java:360)
	at org.eclipse.egit.core.project.RepositoryMapping.getMapping(RepositoryMapping.java:296)
	at org.eclipse.egit.core.internal.util.ResourceUtil.getRepository(ResourceUtil.java:708)
	at org.eclipse.egit.core.internal.info.GitItemStateFactory.getIndexDiffDataOrNull(GitItemStateFactory.java:106)
	at org.eclipse.egit.ui.internal.decorators.GitLightweightDecorator.decorateResource(GitLightweightDecorator.java:193)
	at org.eclipse.egit.ui.internal.decorators.GitLightweightDecorator.decorate(GitLightweightDecorator.java:171)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorDefinition.decorate(LightweightDecoratorDefinition.java:247)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorManager$LightweightRunnable.run(LightweightDecoratorManager.java:105)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorManager.decorate(LightweightDecoratorManager.java:360)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorManager.getDecorations(LightweightDecoratorManager.java:346)
	at org.eclipse.ui.internal.decorators.DecorationScheduler$1.ensureResultCached(DecorationScheduler.java:412)
	at org.eclipse.ui.internal.decorators.DecorationScheduler$1.run(DecorationScheduler.java:388)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
!SUBENTRY 1 org.eclipse.core.resources 4 369 2021-12-01 14:23:42.934
!MESSAGE Resource '/weatherstation/esp_idf_components' is not local.
!STACK 0
java.lang.Exception: Resource '/weatherstation/esp_idf_components' is not local.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	at org.eclipse.core.internal.resources.Resource.checkLocal(Resource.java:342)
	at org.eclipse.core.internal.resources.Resource.checkAccessibleAndLocal(Resource.java:211)
	at org.eclipse.core.internal.resources.Resource.getSessionProperty(Resource.java:1182)
	at org.eclipse.egit.core.project.GitProjectData.getRepositoryMapping(GitProjectData.java:568)
	at org.eclipse.egit.core.project.RepositoryMapping.findMapping(RepositoryMapping.java:360)
	at org.eclipse.egit.core.project.RepositoryMapping.getMapping(RepositoryMapping.java:296)
	at org.eclipse.egit.core.internal.util.ResourceUtil.getRepository(ResourceUtil.java:708)
	at org.eclipse.egit.core.internal.info.GitItemStateFactory.getIndexDiffDataOrNull(GitItemStateFactory.java:106)
	at org.eclipse.egit.ui.internal.decorators.GitLightweightDecorator.decorateResource(GitLightweightDecorator.java:193)
	at org.eclipse.egit.ui.internal.decorators.GitLightweightDecorator.decorate(GitLightweightDecorator.java:171)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorDefinition.decorate(LightweightDecoratorDefinition.java:247)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorManager$LightweightRunnable.run(LightweightDecoratorManager.java:105)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorManager.decorate(LightweightDecoratorManager.java:360)
	at org.eclipse.ui.internal.decorators.LightweightDecoratorManager.getDecorations(LightweightDecoratorManager.java:346)
	at org.eclipse.ui.internal.decorators.DecorationScheduler$1.ensureResultCached(DecorationScheduler.java:412)
	at org.eclipse.ui.internal.decorators.DecorationScheduler$1.run(DecorationScheduler.java:388)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

!ENTRY org.eclipse.egit.core 4 0 2021-12-01 14:23:42.935
!MESSAGE Failed finding RepositoryMapping
!STACK 1
org.eclipse.core.internal.resources.ResourceException(/weatherstation/esp_idf_components)[369]: java.lang.Exception: Resource '/weatherstation/esp_idf_components' is not local.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	at org.eclipse.core.internal.resources.Resource.checkLocal(Resource.java:342)
	at org.eclipse.core.internal.resources.Resource.checkAccessibleAndLocal(Resource.java:211)
	at org.eclipse.core.internal.resources.Resource.getSessionProperty(Resource.java:1182)
	at org.eclipse.egit.core.project.GitProjectData.getRepositoryMapping(GitProjectData.java:568)
	at org.eclipse.egit.core.project.RepositoryMapping.findMapping(RepositoryMapping.java:360)
	at org.eclipse.egit.core.project.RepositoryMapping.getMapping(RepositoryMapping.java:296)
[... cut ...]
!MESSAGE Indexed 'weatherstation' (41 sources, 293 headers) in 4.85 sec: 15,023 declarations; 23,878 references; 41 unresolved inclusions; 18 syntax errors; 120 unresolved names (0.31%)
```

Sorry for the delayed response. I am quite busy right now.

So I cleared the whole log, cleaned the project and deleted the build directory. After build and flash that's the whole log. I use my eclise for not only idf projects. Also I have imported the whole idf sdk as a separate project with content filters in order to have a make-shift source code parsing

luciantodor, 2021-12-03:
Hello, I was facing a similar issue like this and following the guidance of kolipakakondal, I've linked manually the sources into eclipse. Of course, filters are required to have only the relevant sources.

kolipakakondal, 2022-02-14:
Hello All,

With the release v2.4.0, we have linked sources of esp-idf components and now you should be able to navigate sources by pressing F3. This release also fixed a lot of unresolved header issues and indexer problems.

Please get the latest changes using the update site and see if that helps. https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

Thank you

closed 2022-02-14 by kolipakakondal