espressif/idf-eclipse-plugin #219: Eclipse can't resolve and navigate some nested dependencies

userbogd, 2021-03-07:
### Issue
Eclipse can't resolve and navigate some nested dependencies

### Steps to reproduce
1.Create new 'Espressif IDF Project' from template {IDF_PATH}\examples\protocols\http_server\file_serving
2.Build the project
3.After building, go to the file 'file_server.c', where remains unresolved types from <http_parser.h>
4.Try to navigate from 'file_server.c' to "esp_http_server.h", and further, to <http_parser.h>
5.Try to navigate to the others headers, included in 'esp_http_server.h'

### Expected behavior
'could not be resolved' errors in Eclipse Environment, but there is no errors when build with idf.py

### Environment Information
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-12 (4.18.0)
Build id: 20201210-1552

### IDF Eclipse Plugin Version
Version: 2.0.0.202103040836

### ESP-IDF Tools Information
Executing C:\Users\bogd\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\bogd\esp-idf-v4.2\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

### Additional information
-This behavior is detected on 2.x.x plugin version only. 
-In 1.2.4 all works fine.
-Some others dependencies are not resolved in this sample, seems due to similar reason

kolipakakondal, 2021-03-08:
Hi @userbogd Will verify this and update you.

DigKleppe, 2021-04-13:
Maybe the same subject, I miss the project C/C++ general -> path and symbols. So Eclipse cannot find them.
Dig

kolipakakondal, 2021-05-07:
Hi @patrykkrz Are you on the latest version 2.1.0, we have resolved header resolving issues with that. Could you please verify and let us know?

patrykkrz, 2021-05-07:
Yes, 2.1.0. But I have legacy project imported to Eclipse. Maybe problem is in my project. New created project looks good (I checked it after your comment). Issue is open, so I didn't know that problem was resolved.

patrykkrz, 2021-05-07:
I got it! In my project, I have esp-idf directory (as git submodule). After remove esp-idf from project directory it's ok. It looks like I need to reorganize this.

patrykkrz, 2021-05-12:
I would like to leave the esp-idf directory in my project. Propably, problem with symbols from "esp_log.h" is caused by the file "esp-idf\components\spi_flash\sim\stubs\log\include\esp_log.h".
Propably, indexer finds 2 files with the same symbols.
Any idea, how exclude from indexer file: esp-idf\components\spi_flash\sim\stubs\log\include\esp_log.h ?
I could remove it, but it doesn't seem like the correct solution to me.

kolipakakondal, 2022-02-14:
Hello,

Recently we have released v2.4.0/v2.4.1 and this is having some important fixes for indexer issues and unresolved header problems. This release also comes with the source code navigation functionality(Press F3)

Please get the latest changes using the update site and see if that helps.
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

Thank you

closed 2022-02-14 by kolipakakondal

1271, 2022-02-23:
Hello @kolipakakondal . Thanks for your work!

The problem remains, but at the same time, eclipse sometimes correctly displays dependencies.

At one start everything is OK, at the other it is displayed as "missed".

__Code from "blink" example:__

There is also a problem with constants. In this case, the build is successful. <details> <summary>image</summary>
[image]
</details>

broken constants  <details> <summary>image</summary>
[image]
</details>

And... it fixed itself  <details><summary>image</summary>
[image]
 </details>

Anyway, thanks for idf!

kolipakakondal, 2022-02-23:
Hi @userbogd 

Indexing triggers after the build and it might take a couple of seconds(based on system) to resolve headers and symbols and to refresh the project.

Let me know if you see the symbols/headers issue even after waiting for a min? Indexing and refresh status is shown on the right side of the bottom toolbar

userbogd, 2022-02-23:
Hi, @kolipakakondal 
I would like to add some observations about this subject. As i understand there are two different kind of indexing - breef and full. In case of full indexing everything perfect but it takes quite. In some cases, for instance, when swith to debug perspective and back, apply breef indexing, which for some reason leads to indexing errors. Manual rebuild indexis from project menu works not every time.

My solution is reopen and rebuild project on current perspective. Possible all abowe mentioned related to Windows platform only.

In any case the plugin is awersome!!! It allowed me to full migrate to esp32 platform. Thank you so much for you great work!