espressif/idf-eclipse-plugin #221: eclipse 202012R / esp-idf 4.2 unresolved include

r2d2r01, 2021-03-08:
Hi all,
Build in cmd(idf build) build)/ Eclipse build OK
But all includes are in unresolved inclusion else sdkconfig which is in current project path
Try many things (reindex, cmakelist, ...)
Thansk's for helping
Frederic

### Context
laptop (windows 10x64/core i7/32Gb/1Tbhdd)
eclipse-cpp-2020-12-R-win32-x86_64 (Eclipse IDE 2020-12 R Packages - Eclipse IDE for CC++ Developers)
esp-idf: 4.2 => D:\_DEV\esp-idf\esp-idf-v4.2
python 3.9.2=> D:\_DEV\Python39\python.exe 

**Expected behavior**
[eclipse_config.txt](https://github.com/espressif/idf-eclipse-plugin/files/6101674/eclipse_config.txt)

[image]
**Screenshots**

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Executing D:\_DEV\Python39\python.exe D:\_DEV\esp-idf\esp-idf-v4.2\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: D:\_DEV\esp-idf\esp-idf-v4.2
IDF_PYTHON_ENV_PATH: ${PYTHON}
PATH: I:/eclipse202012R/eclipse//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin/server;I:/eclipse202012R/eclipse//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin;D:\_DEV\Python39;D:\_DEV\Python39\Scripts\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;D:\Program Files\TortoiseGit\bin;D:\Program Files\Git\cmd;D:\_DEV\Java211\bin;I:\msys64\opt\xtensa-esp32-elf\bin;D:\_DEV\esp-idf\esp-idf-v4.2\components\esptool_py\esptool;D:\_DEV\Python39\Lib\;D:\_DEV\Python39\Tools\scripts\;D:\_DEV\esp-idf\esp-idf-v4.2;C:\Users\ffvc\.espressif\tools\cmake\3.16.4\bin;D:\_DEV\Python39\Scripts\;D:\_DEV\Python39\;C:\Users\ffvc\AppData\Local\Microsoft\WindowsApps;;I:\eclipse202012R\eclipse;${PATH}

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.18.0.v20201202-1800
Eclipse CDT Version: 10.1.0.202010062020
Traceback (most recent call last):
  File "D:\_DEV\esp-idf\esp-idf-v4.2\tools\idf.py", line 45, in <module>
    from idf_py_actions.tools import (executable_exists, idf_version, merge_action_lists, realpath)  # noqa: E402
  File "D:\_DEV\esp-idf\esp-idf-v4.2\tools\idf_py_actions\tools.py", line 1, in <module>
    import click
ModuleNotFoundError: No module named 'click'
Python set for IDF_PYTHON_ENV: Python 3.9.2
Python declared in PATH variable: Python 3.9.2

****IDF Eclipse Plugin Version:**** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
ESP-IDF Eclipse Plugin
Version: 2.0.0.202103040836
Copyright (c) 2019 ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD. All rights reserved.

r2d2r01, 2021-03-15:
no answers, switch on vsc for esp32, bye

igrr, 2021-03-15:
Hi Frederic, sorry that noone got back to you earlier. The team developing this project works in 2-week sprints. We will try to get this issue investigated and resolved in the next sprint, starting on March 23rd.

r2d2r01, 2021-03-15:
No pb Ivan, when u can, thank's for reply. I switch because i can't wait investigation, but i practice eclipse since some years, editor is not simple but very complete. I will help u when you will be back. See u.

kolipakakondal, 2021-03-16:
Hi @r2d2r01  While we investigate and fix this, could you delete the build folder and rebuild the project. I think that will fix unresolved includes issue for you.

r2d2r01, 2021-03-16:
Hi Kondal, i precise all these installations have been made on a new laptop without others conflictuous configurations/components to be sure it exists no interactions. I have install/uninstall many times with , each time with a idf.py clean/full clean (manual rm delete tree ) by using esp-idf-tools-setup-2.3 or 2.4 for v4.2 branch/last eclipse or version-1 & some plugin version. I recognize espressif  have been done a magnific work with plugins but .... Very thank's.

closed 2021-04-09 by r2d2r01