espressif/idf-eclipse-plugin #223: ESP-IDF Serial Monitor does show any communication

ajhuitsing, 2021-03-10:
**Describe the bug**
Opening ESP-IDF Serial Monitor fails with a 'unhandled event loop exception' , and the communication does not open. Somehow this has worked fine before on my machine, but now it always crashes.

**To Reproduce**
launch terminal -> select 'ESP-IDF Serial Monitor' -> press 'OK' to open

**Expected behavior**
The serial monitor should open and show messages arriving on the serial port, and pause during flash upload.
(note: this has worked perfectly before, but now it doesn't anymore)

**Environment Information:** 
[environment_information.txt](https://github.com/espressif/idf-eclipse-plugin/files/6112238/environment_information.txt)

**ESP-IDF Tools Information:**
[esp-idf_tools_information.txt](https://github.com/espressif/idf-eclipse-plugin/files/6112239/esp-idf_tools_information.txt)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	2.0.0.202103040836

here's the exception:
[IllegalArgumentException.txt](https://github.com/espressif/idf-eclipse-plugin/files/6112242/IllegalArgumentException.txt)

kolipakakondal, 2021-03-10:
Hi @ajhuitsing 

Please select a project before you launch a serial monitor and that should work.

However, will improve on the error message.

Thanks

ajhuitsing, 2021-03-10:
continued to work on my ESP project today... unfortunately now the monitor doesn't pause during download anymore, despite selecting the active project.
this results in 2 processes which have the serial port open (the upload process and the monitor process)
--> so sorry to report that it in fact *is not* solved yet for me.. :-(

ESP3235, 2021-03-14:
I have the same problem , serial monitoring worked good. but now the serial monitoring not open the com port. 
if for example COM3 opened with putty , serial monitor in eclipse show window with title COM3 with no reaction.I closed the putty and test again , eclipse show the same window. 
[image]

sometimes , open the port and reset the module.

My module is ESP32-Devcit  wroom32

esp-idf version is 4.2
windows 10 build 19042.867
Thanks

kolipakakondal, 2021-03-16:
HI @ESP3235 Could you please share the error log to troubleshoot further on this issue.
https://github.com/espressif/idf-eclipse-plugin#error-log

ESP3235, 2021-03-16:
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D D:\esp-idf-v4.2\components\esptool_py && C:\Users\___\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="D:/esp-idf-v4.2" -D ESPTOOLPY="python D:/esp-idf-v4.2/components/esptool_py/esptool/esptool.py --chip esp32" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="D:/_Projects/eclipse_workspace/a2dp_sink/build" -P D:/esp-idf-v4.2/components/esptool_py/run_esptool.cmake"
esptool.py --chip esp32 -p COM3 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 a2dp_sink.bin
esptool.py v3.0
Serial port COM3
Traceback (most recent call last):
  File "D:/esp-idf-v4.2/components/esptool_py/esptool/esptool.py", line 3969, in <module>
    _main()
  File "D:/esp-idf-v4.2/components/esptool_py/esptool/esptool.py", line 3962, in _main
    main()
  File "D:/esp-idf-v4.2/components/esptool_py/esptool/esptool.py", line 3551, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "D:/esp-idf-v4.2/components/esptool_py/esptool/esptool.py", line 271, in __init__
    self._port = serial.serial_for_url(port)
  File "C:\Users\___\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\Users\___\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

FAILED: CMakeFiles/flash 
cmd.exe /C "cd /D D:\esp-idf-v4.2\components\esptool_py && C:\Users\___\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="D:/esp-idf-v4.2" -D ESPTOOLPY="python D:/esp-idf-v4.2/components/esptool_py/esptool/esptool.py --chip esp32" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="D:/_Projects/eclipse_workspace/a2dp_sink/build" -P D:/esp-idf-v4.2/components/esptool_py/run_esptool.cmake"
ninja: build stopped: subcommand failed.
Executing action: flash
Running ninja in directory d:\_projects\eclipse_workspace\a2dp_sink\build
Executing "ninja flash"...
ninja failed with exit code 1

dobairoland, 2021-03-16:
> the monitor doesn't pause during download anymore

I think this is the issue. On Windows you cannot connect to the same port with two applications. The monitor has to be stopped before flashing is started. Otherwise the "access is denied" error will show up on the second attempt of connection.

ESP3235, 2021-03-16:
Exactly this is my problem, because no application used COM port. for example at the same time I can open COM3 in putty then disconnect it and try again, but again that error occurred.

franz-ms-muc, 2021-03-19:
Yes, @dobairoland is right, in before releases the Monitor did Pause automatically during Downloads. 
now it is no more. 
this did definitely WORK in Windows before.

JianWang18, 2021-07-14:
> Exactly this is my problem, because no application used COM port. for example at the same time I can open COM3 in putty then disconnect it and try again, but again that error occurred.

Have you solved this problem? I have the same problem as you.

JianWang18, 2021-07-14:
@kolipakakondal   
[image]

kolipakakondal, 2021-07-14:
> @kolipakakondal
> [image]

Hello, Could you share Java and Eclipse CDT version which is used? I would suggest using Java 11+ and Eclipse 2020-12(4.18) and above

Please check this https://github.com/espressif/idf-eclipse-plugin#installing-prerequisites

Also provide the environment and plugin information to troubleshoot further https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-provide-eclipse-envrionment-and-plugins-information

JianWang18, 2021-07-15:
> Hello, Could you share Java and Eclipse CDT version which is used? I would suggest using Java 11+ and Eclipse 2020-12(4.18) and above
>
> Please check this [...]

I follow this link to download and install.

* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.18.0.v20201202-1800
Eclipse CDT Version: 10.1.0.202010062020
IDF Eclipse Plugin Version: 2.1.2.202106231127
ESP-IDF v4.2.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.7.3
Python declared in PATH variable: Python 3.7.3

kifantidis, 2021-07-26:
Hello there. 

I'm anticipating similar problem. I'm opening the serial TM and it seems that it does not connect with my device. 
I'm posting screenshots below:

[image]

[image]

[image]

When I open with an other serial monitor program the connection everything is working as it should be.

[image]

My system's specs are: 
[Eclipse envrionment and plugins information.txt](https://github.com/espressif/idf-eclipse-plugin/files/6876263/Eclipse.envrionment.and.plugins.information.txt)

IDF Eclipse Plugins version: 2.1.2.202106231127

Best regards, Kostas.

JianWang18, 2021-07-26:
> Hello there.
> 
> I'm anticipating similar problem. I'm opening the serial TM and it seems that it does not connect with my device.
> I'm posting screenshots below:
> 
> [image]
> 
> [image]
> 
> [image]
> 
> When I open with an other serial monitor program the connection everything is working as it should be.
> 
> [image]
> 
> I don't know if this means anything but I've noticed that "ESP-IDF Serial Monitor Settings" are missing from Espessif's preferences.
> 
> [image]
> [image]
> 
> My system's specs are:
> [Eclipse envrionment and plugins information.txt](https://github.com/espressif/idf-eclipse-plugin/files/6876263/Eclipse.envrionment.and.plugins.information.txt)
> 
> IDF Eclipse Plugins version: 2.1.2.202106231127
> 
> Best regards, Kostas.

Yes,my problem is the same as yours.
You can see the error log.

franz-ms-muc, 2021-07-29:
For me the Terminal works.

Just the 80 Char limit is boring.

[03]<https://user-images.githubusercontent.com/32487118/126946905-ef3d7331-c679-4716-b98f-d5928efa83b5.png>

[04]<https://user-images.githubusercontent.com/32487118/126947042-e8d1e3cd-a8d4-456a-ad21-bc2ef08f4b98.png>

[06]<https://user-images.githubusercontent.com/32487118/126948281-307a3af7-b07a-4e18-8c33-8594c0d95cba.png>
[05]<https://user-images.githubusercontent.com/32487118/126948290-58f3413e-743c-4891-9179-426f14ca145e.png>

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

kifantidis, 2021-07-30:
Hello  franz-ms-muc. 

Can you please share your setup ? Is there any difference in Eclipse version or in OS ?? Also which esp-eclipse plugin version are you using ?

kolipakakondal, 2021-07-30:
> Just the 80 Char limit is boring

Hello, the 80 char limit issue got fixed on the master branch. It will be available with the next release. We have added preference settings to configure this https://github.com/espressif/idf-eclipse-plugin#esp-idf-serial-monitor-settings

kolipakakondal, 2021-07-30:
> > Hello there.
> > I'm anticipating similar problem. I'm opening the serial TM and it seems that it does not connect with my device.

Hello, Could you close the Terminal view and reopen it? Maybe the port got blocked by another process.

Please share an error log to verify further https://github.com/espressif/idf-eclipse-plugin#error-log

franz-ms-muc, 2021-08-28:
80 Char Problem solved with IDF Eclipse Plugin Version: 2.2.0.202108110455

kolipakakondal, 2022-02-14:
Hello,

We have released v2.4.0/2.4.1 recently, please get the latest changes using the update site and verify. https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0

Thank you

closed 2022-02-14 by kolipakakondal