espressif/idf-eclipse-plugin #225: eclipse plugin seems not to update the `IDF_PYTHON_ENV_PATH` variable after updating esp-idf to 4.4

TobiasBoeschMiele, 2021-03-10:
**Describe the bug**
* Updated to the latest esp-idf version (4.4 I think currently (not available as a release yet I think)
* ran the tools installer in the eclipse plugin and
* it created a new python environment called `idf4.4_py3.6_env`

The eclipse plugin did not update the `IDF_PYTHON_ENV_PATH` variable to that environment. That led to an error with a missing python dependency (`construct`).

Should that path variable be updated in such a scenario?

**Expected behavior**
Environment variable gets updated to the latest? python environment or the one that matches the esp-idf version?

**Environment Information:** 
eclipse 2020-12 (4.18.0)

**ESP-IDF Tools Information:**
Executing C:\espressif_ESP_IDF_Tools\python_env\idf4.4_py3.6_env\Scripts\python.exe C:\Versionsverwaltung\esp-idf\tools\idf_tools.py list
WARNING: tool idf-python version 3.9.1 is installed, but has reported version 3.6.5
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - 1.24.0.123_64eb9ff-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)
* idf-python: Embeddable Python distribution (optional)
  - 3.9.1 (recommended)
* idf-python-wheels: Python Wheels distribution bundled for the specific version of IDF and Python
  - idf4.3_py3.9_2021-01-07 (recommended, installed)

**IDF Eclipse Plugin Version:** 
Espressif IDF Plugins for Eclipse	2.0.0.202103040836	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

kolipakakondal, 2021-03-16:
Thanks for reporting @TobiasBoeschMiele - Will fix it in the next release. 

To manually modify this, please follow the instructions from here https://github.com/espressif/idf-eclipse-plugin#configuring-environment-variables

- Go to Eclipse preferences
- Go to `C/C++  > Build > Environment`
- Modify I`DF_PYTHON_ENV_PATH` by clicking on `Edit...` button

TobiasBoeschMiele, 2021-03-16:
> To manually modify this,

Already did - thanks.

kolipakakondal, 2021-04-19:
FIXED with 2.1.0 release

closed 2021-04-19 by kolipakakondal