espressif/idf-eclipse-plugin #226: ESP Compiler Lockup

donpowrie, 2021-03-10:
I'll start by stating I'm new to the ESP compiler.

I have installed, uninstalled and started over a about 6 times so far using the instructions found here for ESP under Eclipse on a Windoze 10 box:
https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url

Every time I load the Hello World demo project and try to build it gets to the same point and locks up Eclipse as shown in the image I'm hoping is attached to this post. The only way to be able to use Eclipse again is to reboot my PC.

This certainly seems like an issue with coordinating with Python but I'm at a loss for what to try next.  The paths are all set correctly.

Any/all comments welcome.

[image]

kolipakakondal, 2021-03-16:
Hi @donpowrie Sorry that it didn't work out of the box. To troubleshoot further could you add additional CMake arguments`--trace --debug-output` to the build and see what it shows.

Please find the screenshot below.
[image]

kolipakakondal, 2021-03-16:
Hi @projectgus @igrr Any thoughts on why the build process stuck on checking `Checking python dependencies...`

dobairoland, 2021-03-16:
I have never seen this behavior before. Some antivirus or Windows itself might block access. Isn't there any opened window in the background asking for permissions? Maybe those accesses are permanently blocked so the given tool won't ask for permission.

Do you have access to the ESP-IDF? You can check this by opening any ESP-IDF file from Eclipse. 

It could be useful to check if running manually `<python> $IDF_PATH/tools/check_python_dependencies.py` has the same issue. Can you please @kolipakakondal advise how to do that? I don't know which `<python>` is used in the case of Eclipse.

donpowrie, 2021-03-16:
Kondal,

I added

--trace --debug-output

to the Launch Configuration but the compiler still locks up.

Here is the Console output:

                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(15):  else()
   Called from: [7] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [6] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(16):  get_property(val
TARGET __idf_build_target PROPERTY ${property} )
   Called from: [7] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [6] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(18):  set(${var}
${val} PARENT_SCOPE )
   Called from: [7] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [6] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(43):  if(NOT property
IN_LIST build_properties )
   Called from: [6] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(321):  unset(_var )
   Called from: [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(390):
 idf_build_get_property(project_dir PROJECT_DIR )
   Called from: [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(12):
 cmake_parse_arguments(_ GENERATOR_EXPRESSION   ${ARGN} )
   Called from: [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(13):
 if(__GENERATOR_EXPRESSION )
   Called from: [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(15):  else()
   Called from: [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
[... 544 lines cut ...]
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(18):  set(${var}
${val} PARENT_SCOPE )
   Called from: [5] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(266):  message(STATUS
Checking Python dependencies... )
   Called from: [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(267):
 execute_process(COMMAND ${python}
${idf_path}/tools/check_python_dependencies.py RESULT_VARIABLE result )
   Called from: [4] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3] C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2] C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1] C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt

The compile process locks up here.

Regards,
Don Powrie

donpowrie, 2021-03-16:
Roland,

In my first post in the Espressif forum I show the main.c file in Eclipse.
Doesn't this show that eclipse has access to ESP-IDF files?  If not, please
advise the file you would like for me to try and open.

----------------

I checked the Python requirements with this:

C:\Users\don\Desktop\esp-idf\tools>python check_python_dependencies.py
Python requirements from C:\Users\don\Desktop\esp-idf\requirements.txt are
satisfied.

These are the contents of requirements.txt:

# This is a list of python packages needed for ESP-IDF. This file is used
with pip.
# Please see the Get Started section of the ESP-IDF Programming Guide for
further information.
#
setuptools
# The setuptools package is required to install source distributions and on
some systems is not installed by default.
# Please keep it as the first item of this list.
#
click>=5.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22

Regards,
Don Powrie

donpowrie, 2021-03-16:
Roland,

My active hello_world project is located in my workspace: c:\eclipse-WORKSPACE.

I located another .c file (mesh_main.c) and was able to open the file into eclipse from this location:
C:\Users\don\Desktop\esp-idf\examples\mesh\internal_communication\main

Please advise if this does not answer your earlier question:
Do you have access to the ESP-IDF? You can check this by opening any ESP-IDF file from Eclipse.

kolipakakondal, 2021-03-17:
Hi, @donpowrie Thanks for the logs. I believe the issue something at the system level. We can run the following tests to verify this.

**From Eclipse terminal**
1.  Go to Eclipse ESP-IDF terminal https://github.com/espressif/idf-eclipse-plugin#esp-idf-terminal
2. Run $IDF_PATH/tools/check_python_dependencies.py

```
bash-3.2$ $IDF_PATH/tools/check_python_dependencies.py
Python requirements from /Users/kondal/esp/esp-idf/requirements.txt are satisfied.
```

Also, verify the same command from the OS command prompt 
1.  Go to OS command prompt/shell
2. Run C:/Users/don/Desktop/esp-idf/tools/check_python_dependencies.py

kolipakakondal, 2021-03-17:
Please also find the 2.6 beta installers here:
online version - 2.1 MB - https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.6.exe
offline version - 1.2 GB (IDF 4.1.1, 4.2) - https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.6.exe

This comes with the Eclipse 2020-12 and idf eclipse plugin 2.0.0 pre-installed. Please install and see if that helps as well.

dobairoland, 2021-03-17:
> Do you have access to the ESP-IDF? You can check this by opening any ESP-IDF file from Eclipse.

Yes, you answered my question. It doesn't seem to be related to access rights.

donpowrie, 2021-03-17:
Kondal,

As a reminder, I am running Windows 10.

This first attempt appears to have failed.

C:\Users\don\Desktop\esp-idf>$IDF_PATH/tools/check_python_dependencies.py
'$IDF_PATH' is not recognized as an internal or external command,
operable program or batch file.

*** Perhaps this is the problem?
How do I assign "C:\Users\don\Desktop\esp-idf" to $IDF_PATH ??
Seems like this path is something that should have been setup by the
installer program.

--------------

I then tried this full path from the Eclipse Terminal:

C:\Users\don\Desktop\esp-idf\tools>python check_python_dependencies.py
Python requirements from C:\Users\don\Desktop\esp-idf\requirements.txt are
satisfied.

--------------

This is from the ESP-IDF CMD Prompt:
C:\Users\don\Desktop\esp-idf\tools>python check_python_dependencies.py
Python requirements from C:\Users\don\Desktop\esp-idf\requirements.txt are
satisfied.

--------------

*** When I try this from the regular Windows CMD prompt the window locks
up.  I was unable to CTRL Break or CTRL-C to end the process.  This is also
what has been happening when I try to compile in Eclipse.  The only way to
kill the process is to reboot the PC.

Don

kolipakakondal, 2021-03-18:
Hi, Could you run run the below command direclty without $IDF_PATH to verify. $IDF_PATH is configured only with in the Eclipse environment, not at the system level.

> C:/Users/don/Desktop/esp-idf/tools/check_python_dependencies.py

donpowrie, 2021-03-18:
I did this in my previous post yesterday.  I ran it 3 different ways.  See above.
If you want this command run a different (4th) way, other than what is shown above, then please advise specific instructions.

dobairoland, 2021-03-18:
> I did this in my previous post yesterday. I ran it 3 different ways. See above.

I missed those as well. The output is hidden in Github under "...". I guess you've replied with e-mail.

The output seems to be correct for me. That means only Eclipse has a problem running that script.

donpowrie, 2021-03-18:
I will now try the 2.6 beta installer.

donpowrie, 2021-03-18:
This also failed at the same point.

.....previous text deleted... 
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(267):  message(STATUS Checking Python dependencies... )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(268):  execute_process(COMMAND ${python} ${idf_path}/tools/check_python_dependencies.py RESULT_VARIABLE result )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
...locked up here...

This looks OK:
C:\Users\don\Desktop\esp-idf\tools>python check_python_dependencies.py
Python requirements from C:/Users/don/Desktop/esp-idf\requirements.txt are satisfied.

donpowrie, 2021-03-18:
It still locks up with version 2.6 of the beta installer.

I have attached the complete build log to this email.

                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(15):  else()
   Called from: [7]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [6]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(16):  get_property(val TARGET __idf_build_target PROPERTY ${property} )
   Called from: [7]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [6]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(18):  set(${var} ${val} PARENT_SCOPE )
   Called from: [7]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [6]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(43):  if(NOT property IN_LIST build_properties )
   Called from: [6]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [5]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(322):  unset(_var )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(391):  idf_build_get_property(project_dir PROJECT_DIR )
   Called from: [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(12):  cmake_parse_arguments(_ GENERATOR_EXPRESSION   ${ARGN} )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(13):  if(__GENERATOR_EXPRESSION )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(15):  else()
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(16):  get_property(val TARGET __idf_build_target PROPERTY ${property} )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(18):  set(${var} ${val} PARENT_SCOPE )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
[... 463 lines cut ...]
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
[... cut ...]
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt
C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake(268):  execute_process(COMMAND ${python} ${idf_path}/tools/check_python_dependencies.py RESULT_VARIABLE result )
   Called from: [4]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [3]	C:/Users/don/Desktop/esp-idf/tools/cmake/build.cmake
                [2]	C:/Users/don/Desktop/esp-idf/tools/cmake/project.cmake
                [1]	C:/eclipse-WORKSPACE/hello_world/CMakeLists.txt

kolipakakondal, 2021-03-19:
Hi @dobairoland Looks like the Java process is waiting/blocked by the`check_python_dependencies.py` call. The only reason I could think of for this is python buffer is stuck without emitting a message back. Probably, I could see set `PYTHONUNBUFFERED=TRUE` to allow log messages to be dumped immediately without being buffered.

kolipakakondal, 2021-03-19:
I have created a PR for this https://github.com/espressif/idf-eclipse-plugin/pull/232

kolipakakondal, 2021-03-19:
Hi @donpowrie - Could you please verify this build and see if that helps to resolve the issue which you're facing.

Here is the local build https://github.com/espressif/idf-eclipse-plugin/suites/2295093680/artifacts/48085616
Clicking on the above link will load the artifacts.zip file for this change

To update your eclipse with the above `artifacts.zip`, please follow the instructions from here https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

donpowrie, 2021-03-19:
... set PYTHONUNBUFFERED=TRUE  -- I do not know how to do this.  

I installed the artifacts and restarted eclipse.  When it came back up my project had no Launch Mode, Configuration or Target so I deleted the project and went to reload the HelloWorld example project but the File, New, Espressif IDF Project option was not present so I reinstalled the IDF plugin but 
"Download and Configure ESP-IDF" is not available from the Help menu. 

At this point I'm not sure what to do next.

kolipakakondal, 2021-04-26:
This problem is resolved with the release 2.1.0, please update to the latest plugin
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.1.0

closed 2021-04-26 by kolipakakondal