espressif/idf-eclipse-plugin #227: Target configuration select wrong com port

icodk, 2021-03-11:
Target configuration  select wrong com port on Windows
**To Reproduce**
1. Connect your board to the PC
2. Configure a new target and select the port assigned to your board from the list of ports:
    On my PC I have COM1 , COM3 and COM5 which is assigned to the esp32 board, so I select COM5 from the list.
3. Disconnect your board and connect another ESP board.
    On my PC, the new board is assigned COM6, However the previously configured target is magically  changed to COM3
     probably because it can't find COM6. 
     This is not optimal. It will be better if the assigned port will not be changed by the plugin. So I could configure two targets,
     one for COM5 and one for COM6 and manually change when I change board.
   Additional improvement:
1. A target delete button
    sometimes you want to cleanup the target list from unused targets
2. A baud rate configuration field 
    On my PC the baud rate is  460800. (what an evil number :-)  Where is it come from ?
     Even the Arduino IDE let you select different baud rate: 512000, 230400, 256000, 115200,921600
     What the idea of deciding for for the developer which baud rate to use ?
     DEMOCRATISE the Baud rate !! Thanks
     

**Environment Information:** 
[Eclipse_Install_cfg.txt](https://github.com/espressif/idf-eclipse-plugin/files/6122977/Eclipse_Install_cfg.txt)
Running on Windows 10 Pro
**ESP-IDF Tools Information:**
```
Executing C:\Dev\esp32-idf\idf-tools\.espressif\python_env\idf4.2_py3.6_env\Scripts\python.exe C:\Dev\esp32-idf\esp-idf-v4.2\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)
```
**IDF Eclipse Plugin Version:** 
`Version: 2.0.0.202103040836`

kolipakakondal, 2021-03-16:
Hi @icodk 

> A baud rate configuration field
> On my PC the baud rate is 460800. (what an evil number :-) Where is it come from ?
> Even the Arduino IDE let you select different baud rate: 512000, 230400, 256000, [...]

With the 2.0.0 release, we have integrated the Eclipse ESP-IDF Serial monitor with the `idf.py monitor` hence we have removed this option from the UI. But you can always change the idf.py baud rate from the sdkconfig editor. Please find the screenshot below.

[image]

> A target delete button
> sometimes you want to cleanup the target list from unused targets

There is no option for now. We will work on this. Thanks for the feedback.

kolipakakondal, 2021-03-16:
Hi @icodk 

Here is the way if you want to delete the launch target manually.

1. Go to the Eclipse workspace directory. For example:  in my case  /Users/<username>/myTesteclipseWorkspace
2. Navigate to `.metadata/.plugins/org.eclipse.core.runtime/.settings` folder in the workspace directory
3. Look for `org.eclipse.launchbar.core.prefs` file and open it in the editor
4. Search for the launch target name you want to delete and remove all those entries from the file
5. Save the file
6. Restart the eclipse

icodk, 2021-05-29:
@kolipakakondal 
You write above: `....With the 2.0.0 release, we have integrated the Eclipse ESP-IDF Serial monitor with the idf.py monitor hence we have removed this option from the UI`
However this is not answering my point which is about the flasher NOT the monitor.

Moving options from the UI to the  sdkconfig  for options  unrelated to code(in this case flasher and monitor)  is not a good idea because this will force a complete recompile of all the IDF components. even if nothing was changed in the code.

sigmaaa, 2021-08-17:
Autoselecting the wrong port is fixed. The option to delete the target from the target bar is also added. Please, check the [latest release 2.2.0.](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0) Feel free to re-open in case of some problmes. Thanks

closed 2021-08-17 by sigmaaa