espressif/idf-eclipse-plugin #228: ESP-IDF tools appear to load but error no toolchain occurs on build

orbitcoms, 2021-03-11:
This is my first post here.
I followed the video to install ESP-IDF plug-in for eclipse and get error "No Toolchain file found".
There is another post relating to this that did not solve my issue, and the post is now closed.
I am running Windows 10 Enterprise and installed Eclipse version 2020-12.
Went to "Help" -> "ESP-IDF Tools Manager" -> "Install Tools". Set path to my esp-idf folder, Git location and python location.

Below is terminal output showing the tools install,  it appears to me that there were no errors during tool installation.

Here is a list of tools installed, followed by Console output while installing the tools.

************  Installed Tools ******************
Executing C:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\Scripts\python.exe C:\esp\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)

************* Console output during tool install ********************
Installing tools...
This can take a while. Please be patient.
Executing C:\Users\david\AppData\Local\Programs\Python\Python39\python.exe C:\esp\esp-idf\tools\idf_tools.py install all
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)

Executing C:\Users\david\AppData\Local\Programs\Python\Python39\python.exe C:\esp\esp-idf\tools\idf_tools.py install-python-env
Processing c:\esp\esp-idf\tools\kconfig_new\esp-windows-curses
Requirement already satisfied: setuptools>=21 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from -r C:\esp\esp-idf\requirements.txt (line 4)) (53.0.0)
Requirement already satisfied: click>=5.0 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from -r C:\esp\esp-idf\requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from -r C:\esp\esp-idf\requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from -r C:\esp\esp-idf\requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from -r C:\esp\esp-idf\requirements.txt (line 11)) (3.4.6)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from -r C:\esp\esp-idf\requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from -r C:\esp\esp-idf\requirements.txt (line 13)) (0.27)
Requirement already satisfied: windows-curses in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from esp-windows-curses==0.1->-r C:\esp\esp-idf\requirements.txt (line 19)) (2.2.0)
Requirement already satisfied: cffi>=1.12 in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from cryptography>=2.1.4->-r C:\esp\esp-idf\requirements.txt (line 11)) (1.14.5)
Requirement already satisfied: pycparser in c:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\lib\site-packages (from cffi>=1.12->cryptography>=2.1.4->-r C:\esp\esp-idf\requirements.txt (line 11)) (2.20)
Building wheels for collected packages: esp-windows-curses
  Building wheel for esp-windows-curses (setup.py): started
  Building wheel for esp-windows-curses (setup.py): finished with status 'done'
  Created wheel for esp-windows-curses: filename=esp_windows_curses-0.1-py3-none-any.whl size=1170 sha256=6e5059fc40537648d0cec71f7e265b3111ec2cf3a328b16595c6df81f50db0f2
  Stored in directory: c:\users\david\appdata\local\pip\cache\wheels\5b\ac\71\26133b145a10c6c765f56efffebf6d9946bc473c834af7afae
Successfully built esp-windows-curses
Installing collected packages: esp-windows-curses
  Attempting uninstall: esp-windows-curses
    Found existing installation: esp-windows-curses 0.1
    Uninstalling esp-windows-curses-0.1:
      Successfully uninstalled esp-windows-curses-0.1
Successfully installed esp-windows-curses-0.1
Installing Python packages from C:\esp\esp-idf\requirements.txt

Executing C:\Users\david\AppData\Local\Programs\Python\Python39\python.exe C:\esp\esp-idf\tools\idf_tools.py export --format=key-value
OPENOCD_SCRIPTS=C:\esp\tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE=1
IDF_PYTHON_ENV_PATH=C:\esp\tools\.espressif\python_env\idf4.1_py3.9_env
PATH=C:\esp\tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\esp\tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\esp\tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\esp\tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\esp\tools\.espressif\tools\cmake\3.13.4\bin;C:\esp\tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\esp\tools\.espressif\tools\ninja\1.9.0\;C:\esp\tools\.espressif\tools\idf-exe\1.0.1\;C:\esp\tools\.espressif\tools\ccache\3.7\;C:\esp\tools\.espressif\python_env\idf4.1_py3.9_env\Scripts;C:\esp\esp-idf\tools;%PATH%
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.

kolipakakondal, 2021-03-16:
Hi @orbitcoms Please make sure you select esp launch target while building a project. This error appears when you try to build a project with the local target. 

https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#no-toolchain-found-for-target-esp32-error
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#no-esp-launch-target-found-please-createselect-the-correct-launch-target-error

Please follow the above instructions and see if that helps.

dizcza, 2021-09-07:
I have the same issue.
Eclipse 2021-06 (4.20.0)
Latest Eclipse IDF plugin.
I've git cloned esp-idf release v4.1.2 manually and set up the path to this folder. Then I ran the Install Tools as suggested in https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#no-toolchain-found-for-target-esp32-error. Here is the output
```
Installing tools...
This can take a while. Please be patient.
Executing /home/dizcza/miniconda3/bin/python /home/dizcza/tools/esp-idf-v4.1.2/tools/idf_tools.py install all
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20210401 (already installed)
Skipping ninja@1.9.0 (already installed)

Executing /home/dizcza/miniconda3/bin/python /home/dizcza/tools/esp-idf-v4.1.2/tools/idf_tools.py install-python-env
Python 3.9.1
pip 21.2.3 from /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages/pip (python 3.9)
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 4)) (57.4.0)
Requirement already satisfied: click>=5.0 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 8)) (8.0.1)
Requirement already satisfied: pyserial>=3.0 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 11)) (3.4.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 13)) (0.27)
Requirement already satisfied: cffi>=1.12 in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from cryptography>=2.1.4->-r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 11)) (1.14.6)
Requirement already satisfied: pycparser in /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from cffi>=1.12->cryptography>=2.1.4->-r /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt (line 11)) (2.20)
WARNING: You are using pip version 21.2.3; however, version 21.2.4 is available.
You should consider upgrading via the '/home/dizcza/.espressif/python_env/idf4.1_py3.9_env/bin/python -m pip install --upgrade pip' command.
Installing Python packages from /home/dizcza/tools/esp-idf-v4.1.2/requirements.txt

Executing /home/dizcza/miniconda3/bin/python /home/dizcza/tools/esp-idf-v4.1.2/tools/idf_tools.py export --format=key-value

Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.
```

I'm able to click Run and flash the project but I'm unable to click Build and built it. I get 
```
Building in: /home/dizcza/Projects/Embedded/eclipse-workspace/hello_world/build
No toolchain file found
```
This is annoying. The only reason I switched from the command line `idf.py` syntax to Eclipse is to be able to navigate through source files easily. But since I cannot build the project, I cannot resolve the conflicts Eclipse shows me.

kolipakakondal, 2021-09-07:
Hi @dizcza 

Have you selected the esp launch target from the toolbar before building a project? 

Could you check these https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target and https://github.com/espressif/idf-eclipse-plugin#compiling-the-project

dizcza, 2021-09-07:
> Have you selected the esp launch target from the toolbar before building a project?

Of course, I did.

As I said, I can run and flash the project by clicking the Run button. If I didn't select the esp launch target, I wouldn't be able to run and launch it.

dizcza, 2021-09-07:
Another thing.
I'm trying to build the `i2c_simple` example and to my surprise, it's not in the official latest release. Therefore, I've switched to esp-idf master branch. Again, everything works from the command line. But when I reconfigure Eclipse IDF plugin to Install Toolchain with the following settings

ESP-IDF directory: `/media/dizcza/media/esp-idf`
Git: `/usr/bin/git`
Python: `/home/dizcza/miniconda3/envs/esp32/bin/python3.9` (I've created & activated a separate environment in conda)

the error is shown in the Eclipse console:

```
Installing tools...
This can take a while. Please be patient.
Executing /home/dizcza/miniconda3/envs/esp32/bin/python3.9 /media/dizcza/media/esp-idf/tools/idf_tools.py install all
Traceback (most recent call last):
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1817, in <module>
    main(sys.argv[1:])
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1813, in main
    action_func(args)
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1408, in action_install
    tools_info = load_tools_info()
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 941, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 959, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 809, in from_json
    raise RuntimeError('supported_targets for tool %s is not a list of strings' % tool_name)
RuntimeError: supported_targets for tool xtensa-esp32-elf is not a list of strings

Executing /home/dizcza/miniconda3/envs/esp32/bin/python3.9 /media/dizcza/media/esp-idf/tools/idf_tools.py install-python-env
Python 3.9.1
pip 21.2.3 from /home/dizcza/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages/pip (python 3.9)
Traceback (most recent call last):
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1817, in <module>
    main(sys.argv[1:])
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1813, in main
    action_func(args)
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1529, in action_install_python_env
    wheels_dir = get_wheels_dir()
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1465, in get_wheels_dir
    tools_info = load_tools_info()
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 941, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 959, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 809, in from_json
    raise RuntimeError('supported_targets for tool %s is not a list of strings' % tool_name)
RuntimeError: supported_targets for tool xtensa-esp32-elf is not a list of strings

Executing /home/dizcza/miniconda3/envs/esp32/bin/python3.9 /media/dizcza/media/esp-idf/tools/idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1817, in <module>
    main(sys.argv[1:])
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1813, in main
    action_func(args)
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 1167, in action_export
    tools_info = load_tools_info()
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 941, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 959, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "/media/dizcza/media/esp-idf/tools/idf_tools.py", line 809, in from_json
    raise RuntimeError('supported_targets for tool %s is not a list of strings' % tool_name)
RuntimeError: supported_targets for tool xtensa-esp32-elf is not a list of strings
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.
```
though when I run `/home/dizcza/miniconda3/envs/esp32/bin/python3.9 /media/dizcza/media/esp-idf/tools/idf_tools.py install all` from the command line **there is no error**.

dizcza, 2021-09-07:
All right, I did a clean install, removed all Eclipse-related files from my hard drive and now the Build button works.
I'm using esp-idf master branch. Previously, I was using esp-idf v4.1.x.

kolipakakondal, 2022-04-04:
Hi All, We have improved a lot for the last few releases on the build configuration, launch target, indexer problems, and many more. 

Please update to the latest branch and open issues if you encounter any further problems.

Thanks

closed 2022-04-04 by kolipakakondal