espressif/idf-eclipse-plugin #230: Installation forces eclipse downgrade to 4.19 and breaks TM terminal

blizz, 2021-03-18:
I tried installing the plugin today and it wouldn't install right away but tried to find some solution. The solution it came up with was to downgrade eclipse but not install the plugin.  I though well I'll try it twice so I let it downgrade eclipse but not install the plugin and then tried installing a 2nd time and this time it did install.  The terminal button no longer does anything.  I can use CTRL ALT T to get a local terminal but no dropdown comes up before to allow selection of terminal type. So unable to get a terminal to the ESP32 uart from within eclipse.

This is the setup..
eclipse-cpp-2020-06-R-linux-gtk-x86_64.tar.gz
Ubuntu 20.04 on x64
openjdk 11.0.10 2021-01-19
OpenJDK Runtime Environment (build 11.0.10+9-Ubuntu-0ubuntu1.20.04)
OpenJDK 64-Bit Server VM (build 11.0.10+9-Ubuntu-0ubuntu1.20.04, mixed mode, sharing
Python 3

[EclipseConfig.txt](https://github.com/espressif/idf-eclipse-plugin/files/6160436/EclipseConfig.txt)

**ESP-IDF Tools Information:**
Executing /home/blizz/.espressif/python_env/idf4.2_py3.8_env/bin/python /home/blizz/u/eclipse-esp-idf/esp-idf-v4.2/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	2.0.0.202103040836	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

**Eclipse Error log:** 
[eclipse.log](https://github.com/espressif/idf-eclipse-plugin/files/6160452/eclipse.log)

kolipakakondal, 2021-03-18:
Hi @blizz 

Thanks for the detailed report and logs. Looks like your eclipse installation is corrputed. Below error from log shows you're trying to launch Ardunio terminal.
```

> java.lang.NoSuchMethodError: 'org.eclipse.cdt.serial.BaudRate org.eclipse.cdt.serial.BaudRate.getClosest(int)'
> 	at org.eclipse.tm.terminal.connector.cdtserial.connector.SerialSettings.getBaudRate(SerialSettings.java:116)
> 	at org.eclipse.cdt.arduino.ui.internal.terminal.ArduinoTerminalSettingsPage.loadSettings(ArduinoTerminalSettingsPage.java:239)
> 	at org.eclipse.cdt.arduino.ui.internal.terminal.ArduinoTerminalSettingsPage.createControl(ArduinoTerminalSettingsPage.java:193)
> 	at org.eclipse.cdt.arduino.ui.internal.terminal.ArduinoTerminalConfigPanel.setupPanel(ArduinoTerminalConfigPanel.java:46)
> 	at org.eclipse.tm.terminal.view.ui.controls.ConfigurationPanelControl.setupConfigurationPanels(ConfigurationPanelControl.java:236)
```
Could you get the fresh installation of Eclipse 2020-12 CDT package from here https://www.eclipse.org/downloads/packages/release/2020-12/r/eclipse-ide-cc-developers and update to latest plugin using the update site https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url

blizz, 2021-03-19:
My bad.  I tried it again today on a different computer and it worked.  I noticed the SHA512 checksum was different from yesterday and somehow I had downloaded the wrong version of eclispe then.
eclipse-cpp-2020-06-R-linux-gtk-x86_64.tar.gz
instead of 
eclipse-cpp-2020-12-R-linux-gtk-x86_64.tar.gz
which is the one that works.

Thanks for your quick response!

closed 2021-03-19 by blizz