espressif/idf-eclipse-plugin #239: Don't create a new debug configuration each time

szekelyisz, 2021-04-11:
**Describe the bug**

Every time I start a debugging session a new debug configuration is created in Eclipse.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a debugging session

**Expected behavior**

Use the debug configurations from before instead of creating a new one each time

**Screenshots**

[image]

**Environment Information:** 
[eclipse.txt](https://github.com/espressif/idf-eclipse-plugin/files/6292806/eclipse.txt)

**ESP-IDF Tools Information:**
```
Executing /home/szabi/.espressif/python_env/idf4.2_py3.8_env/bin/python /home/szabi/esp/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)
```

**IDF Eclipse Plugin Version:** 
2.0.0.202103040836

**Eclipse Error log:** 
[error.log](https://github.com/espressif/idf-eclipse-plugin/files/6292811/error.log)

alirana01, 2021-08-30:
@szekelyisz Sorry for late reply
Can you please try the latest release version of the idf plugin from the releases page this seems to be fixed and is not reproducible if you still face the issue please feel free to share the outcome.

Thanks and Regards
 
https://github.com/espressif/idf-eclipse-plugin/releases

alirana01, 2021-11-05:
@szekelyisz Issue is not reproducible in upcoming and current latest release

closed 2021-11-05 by alirana01