espressif/idf-eclipse-plugin #240: How to flash via JTAG

szekelyisz, 2021-04-11:
I want to use a ESP-PROG to flash my board via JTAG because the serial port is not available. But If I want to start a debug session this plugin tries to flash the firmware via serial which obviously fails.

How can I flash my firmware through JTAG? It works from the command line without a problem.

kolipakakondal, 2021-05-06:
This is not yet supported by the eclipse plugin, will take this up as a priority. Thank you for raising this issue.

sigmaaa, 2021-08-16:
Hi @szekelyisz,

The flashing via jtag is supported in the newly released version [ESP-IDF Eclipse Plugin v2.2.0 ](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0). Please check documentation related to this [here](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/JTAG%20Flashing.md). The flashing via jtag is also by default when you are debugging your project now. I'll close the issue, for now, feel free to reopen or create a new one if you will face some issues with it

closed 2021-08-16 by sigmaaa