espressif/idf-eclipse-plugin #241: ESP32 JTAG Debugging defaults can not work

M-Bab, 2021-04-13:
**Describe the bug**
When using the idf-eclipse-plugin default configuration for OpenOCD JTAG debugging it constructs the paths wrong. The openocd structure is: openocd/bin/openocd and openocd/share/openocd/scripts
When the default configuration is used or "Restore defaults" in the Debugger tab is clicked the constructed paths lead either to the wrong path for the openocd binary (${openocd_path}/${openocd_executable}) or the wrong path to the shared data (${openocd_path}/share/openocd/scripts). Which one is wrong depends on ${openocd_path}.

It should be either: ${openocd_path}/bin/${openocd_executable} and ${openocd_path}/share/openocd/scripts
or: ${openocd_path}/${openocd_executable} and ${openocd_path}/../share/openocd/scripts

The resulting debugging error is some cryptical like
```
Failed to execute MI command:
-target-select remote localhost:6666
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add a "ESP-IDF GDB OpenOCD Debugging" configuration with the defaults
2. Enter the Debugging tab and hit "Restore defaults" just to be sure
3. Compare the paths of bin and share with the actual paths. Debugging will result in an error.

**Expected behavior**
The default should have paths according to the esp-idf tool installation

**Screenshots**
[image]

**Environment Information:** 
Skipped - not necessary for this bug (I hope)

**ESP-IDF Tools Information:**
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.0 (recommended, installed)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	2.0.0.202103040836	com.espressif.idf.feature.feature.group	ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD

**Eclipse Error log:** 
Skipped - not necessary for this bug (I hope)

SimonPVS, 2021-04-26:
@kolipakakondal 
I also have this problem. Is this one also going to be fixed?

sigmaaa, 2021-08-17:
Hi @M-Bab, 

you are right, openocd_path should contain a bin folder in the path. You can check it in the Eclipse `Window -> Preferences -> Espressif -> Global OpenOCD Path`. These fields are prefilled after tools installation. I do have it like this: 

[image]

I recommend you try the [latest release 2.2.0](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.2.0) and install the latest ESP-IDF. A lot of problems were fixed and we added JTAG flashing, that will helps in the debugging.

sigmaaa, 2021-08-18:
Hi @SimonPVS,

have you fixed an issue? Looks like something is wrong with the JTAG connection. Maybe jumpers between FTDI and ESP32 are not connected or ESP32 is flashed with some program that has reconfigured one or more JTAG pins.
You can try to run openOCD manually with the `-d3 -l oocd.log` command, so it will generate a log.

SimonPVS, 2021-08-18:
Hi,

It is fixed!
Thanks

closed 2021-08-19 by sigmaaa

RameshGuptaExicom, 2021-09-06:
I am getting following message

Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Warn : Interface already configured, ignoring
Info : Configured 2 cores
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Error: Target not examined yet
Error executing event gdb-attach on target esp32:

Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected.

Kindly help to resolve this Issue

RameshGuptaExicom, 2021-09-06:
I am getting following messge on console after proper connection
pen On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Warn : Interface already configured, ignoring
Info : Configured 2 cores
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. CPU0: PC=0x400E2BBA (active)
Info : Target halted. CPU1: PC=0x400E2BBA 
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 4096 KB
Info : Using flash bank 'esp32.flash' size 4096 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : Using flash bank 'esp32.irom' size 80 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : Using flash bank 'esp32.drom' size 28 KB
undefined debug reason 8 - target needs reset
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Target halted, PC=0x400D385C, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400 
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400 
invalid command name "arm"
Error: Too large number of threads 1073400672!
Error: Too large number of threads 1073400672!
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400 
===== Xtensa registers
(0) pc (/32): 0x40000400
(1) ar0 (/32): 0x00000000
[... 147 lines cut ...]
(153) misc0 (/32): 0x00000000
(154) misc1 (/32): 0x00000000
(155) misc2 (/32): 0x00000000
(156) misc3 (/32): 0x00000000
(168) a0 (/32): 0x00000000
(169) a1 (/32): 0x00000000
(170) a2 (/32): 0x00000000
(171) a3 (/32): 0x00000000 (dirty)
(172) a4 (/32): 0x00000000
(173) a5 (/32): 0x00000000
(174) a6 (/32): 0x00000000
(175) a7 (/32): 0x00000000
(176) a8 (/32): 0x00000000
(177) a9 (/32): 0x00000000
(178) a10 (/32): 0x00000000
(179) a11 (/32): 0x00000000
(180) a12 (/32): 0x00000000
(181) a13 (/32): 0x00000000
(182) a14 (/32): 0x00000000
(183) a15 (/32): 0x00000000

Info : Target halted. CPU0: PC=0x400E2BBA (active)
Info : Target halted. CPU1: PC=0x400E2BBA 
Error: Too large number of threads 1073400672!
Info : dropped 'gdb' connection

sigmaaa, 2021-09-08:
Hi @RameshGuptaExicom,
make sure the drivers are installed correctly. The easiest way to install drivers is to download [offline installer ](https://dl.espressif.com/dl/esp-idf/?idf=4.4), choose "Custom installation" option, and choose the drivers option:
[image]

The other option to install drivers via zadiq tool https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html#windows

M-Bab, 2021-09-09:
I am using the most recent plugin and I cannot confirm that the paths are reasonable after "restore defaults". Furthermore, it currently works worse than ever before and my only chance currently is to run openocd manually. This is also caused by the fact, that I use openocd to also erase/flash the ESP32 - this can last a few dozens of seconds. As a result, gdb times out. It would be great if a delay field could be added to the option that delays the start of gdb vs. the start of openocd by X seconds. Or even better an automatic detection if openocd is actually ready for the gdb.

This is the line I use for the integrated programing option:

`openocd -s ${openocd_path}/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover-kit-1.8v.cfg -c "program_esp ${project_loc}/build/gui-embedded.bin 0x10000 verify"`

sigmaaa, 2021-09-09:
Hi @M-Bab,

what do you have in the `Window -> Preferences -> Espressif -> Global OpenOCD Path`? There should be a correct path, because after clicking on "restore defaults", we are getting value from there.

M-Bab, 2021-10-02:
That is correct. But the defaults combined still lead to either a wrong path to the openocd executable OR to the openocd shared folder.