espressif/idf-eclipse-plugin #250: Monitor forces 80-character wide printout.

liudr, 2021-04-20:
I recently updated my eclipse from 2020-12 to 2021-03 so I thought I should re-install idf-eclipse-plugin. After that, my monitor terminal will only print in 80-char long. I distinctly remember my print outs being as wide as the window but not confined to only about a quarter of the window, exactly 80-char wide. So please fix this! You can't have meaningful printouts forced into multiple lines by this nonsense.

kolipakakondal, 2021-04-20:
Hi @liudr 

Could you check "Maximum Character Width" option is enabled by chance in eclipse preferences? 

[image]

kolipakakondal, 2021-04-20:
https://stackoverflow.com/a/46455961/1812235

liudr, 2021-04-20:
Thanks but it doesn't matter what I have, fixed or not with a 320 char limit. The serial terminal, not console, always displays 80 characters.

liudr, 2021-04-21:
I installed eclipse on a different computer that doesn't have eclipse or IDF. Same outcome. The option @kolipakakondal posted doesn't affect the behavior of the serial terminal, regardless the settings of the max char width or fixed checkbox. I think this is a genuine bug.

Some observations, on my main computer if I start a project with one of the IDF examples, the dialog that should show markup rendeded on the right half shows up as not rendered. I didn't have this issue on the new install on this other computer or another computer I installed.

kolipakakondal, 2021-08-19:
This is fixed with release 2.2.0, we have provided an option to configure serial monitor character limit and number of lines.
https://github.com/espressif/idf-eclipse-plugin#esp-idf-serial-monitor-settings

Please get the latest changes using the update site.

closed 2021-08-19 by kolipakakondal