espressif/idf-eclipse-plugin #251: Latest plugin can't automatically close/open serial port terminal

liudr, 2021-04-20:
I updated my eclipse from 2020-12 to 2021-03 and reinstalled isf-eclipse-plugin afterwards to ensure I'm getting a working plugin. Right after that, I noticed that I can't just press run to have the firmware uploaded anymore if my terminal is open. In the past, the upload will turn off terminal, and once done it switches the terminal back on. Now is no longer the case. I must manually disconnect the terminal before each upload and connect back on after. It's a lot of extra steps that were automated and now is broken. Create a new project and try for yourself.

Windows 10 with IDF V4.3 beta-1

liudr, 2021-04-21:
I installed eclipse and idf on a computer that had neither of them just to see what happens. Same result. Run command fails to stop the serial port monitor. And I have to close serial port, compile-run, then reopen it. Lots of unnecessary steps if the plugin just works as it should in the past.

kolipakakondal, 2021-04-22:
Hi @projectgus @igrr I believe this is a standard behavior if we are using idf.py monitor. Can we do something in the idf.py flash/monitor to improve this or can I work on the Eclipse side to improve this functionality?

While flashing, look for the port if it's getting used, if yes close the port and then reopen again once the flashing is done.

SimonPVS, 2021-04-26:
I have the same problem. 
Is this bug going to be fixed?

kolipakakondal, 2021-04-26:
Hi @SimonPVS  We have the fix ready, it will be released with the next version, I will keep it posted here.

franz-ms-muc, 2021-04-27:
confirm: in 2.1.0 this Bug still exist.

liudr, 2021-05-13:
Any update regarding fixing the auto serial port open/close new release? The fixed 80-character width is also very annoying and unnecessary.

kolipakakondal, 2021-06-04:
we have fixed the serial monitor open/close issue with the v2.1.1 release. Please get the latest changes through update site https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

closed 2021-06-04 by kolipakakondal

franz-ms-muc, 2021-06-05:
This Link shows :

403 Forbidden
________________________________
nginx/1.14.0 (Ubuntu)

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

kolipakakondal, 2021-06-05:
Hi @franz-ms-muc Update site still works, please check from the eclipse.

403 forbidden - there is no web page content to show.

Thanks.

liudr, 2021-06-06:
How do you use this link? I'm asking the obvious because I obviously don't know how to utilize a forbidden link. Please, you did the hard work to make the update. How do you like to have that hard work appreciated by everyone who's going to use it?

kolipakakondal, 2021-06-07:
Hi @liudr 

Could you please check this link https://github.com/espressif/idf-eclipse-plugin#how-do-i-upgrade-my-existing-idf-eclipse-plugin

Thanks

liudr, 2021-06-07:
Thank you! Now this issue has both what's wrong and how to fix.
I can confirm that the solution fixes the serial monitor not auto open/close main issue. Should I raise the 80-char width limit as a separate issue? The older pluggin version didn't have this limit and frankly 80-width is for the 1980's or older.

mmadela, 2021-06-09:
after update the plugin to 2.1.1
I must close serial monitor manually now
it was not the case previously

fhgeoprospectors, 2021-06-30:
with ESP-IDF Eclipse Plugin v2.1.2 it works correct now. (please fix the other Issue 80 Char width)

mmadela, 2021-06-30:
even with 4.1.2 it does not work for me
please note, that there are two different terminal types with similar names:
**serial terminal**
and
**serial monitor**
and the subject of this bug suggest to talk about the first one

The first one is no longer terminated wiith 2.1.x plugins.
The second is closed, but it has some limitations like:
cannot send key chars to the target
requires the elf file (which is not always given)
and the fixed width of 80 chars (as mentioned above)

f-hoepfinger-hr-agrartechnik, 2021-06-30:
i see. 

ESP-IDF Serial Monitor, 
ESP-IDF Terminal (aka Command Line) 
and all else is Eclipse Standard. 

i think the Plugin can only fix things that come with the Plugin. 

the "Serial Terminal" is Standard Eclipse.