espressif/idf-eclipse-plugin #254: Build not configured correctly

shanku81, 2021-04-21:
Hi,

I installed Eclips in windows 10 and followed instructions as followed
https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md
But while building project it comes "Build not configured correctly"
[image]

kolipakakondal, 2021-04-21:
Hi @shanku81 

Could you share the complete screenshot of the eclipse toolbar, looking at the error this could happen when esp launch target selection is missing from the toolbar?

Please also share the plugin version and other details as described in the ticket description.

shanku81, 2021-04-21:
> SECURITY WARNING
> This mail is NOT from an Ottomate ID and could be a security threat. Please take due care before opening and/or responding.

Please find the attachment

kolipakakondal, 2021-04-22:
HI @shanku81  attachment is missing

shanku81, 2021-04-22:
Please find attachement
[image]
[image]

shanku81, 2021-04-22:
Hi,

Please find attachement.

Thanks & Regards
Shankara M

kolipakakondal, 2021-04-22:
Hi @shanku81  Thanks for the attachment. I don't see any issue with the launch target selection. Are you on the latest IDF Eclipse plugin? If you not, could get the latest plugin and see?

```
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
or
For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon
```

shanku81, 2021-04-22:
[image]
Pls find attachement

shanku81, 2021-04-22:
Please find attachement

sigmaaa, 2021-04-22:
Hi @shanku81, could you also share the product information log, please? It will possibly help us reproduce the problem

To do it go to the Help -> Product information.  
And then in the eclipse console will be printed information about installed tools, etc. 

The other thing that could be helpful is information about installed C/C++ toolchains:
Eclipse preferences... -> C/C++ -> Core build Toolchains

shanku81, 2021-04-22:
Hi ,

Please find product information log below,
IDF_PATH: C:\Users\PKJena\Desktop\esp-idf\esp-idf-v4.2
IDF_PYTHON_ENV_PATH: <NOT FOUND>
PATH: <NOT FOUND>

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
ESP-IDF version cannot be checked. IDF_PATH or IDF_PYTHON_ENV_PATH  are not set.
Python set for IDF_PYTHON_ENV: <NOT FOUND>
Python declared in PATH variable: Python 3.6.6

wheres this option we can get it i dont find anywhere this option
The other thing that could be helpful is information about installed C/C++ toolchains:
Eclipse preferences... -> C/C++ -> Core build Toolchains

Can you pls help?
Thanks& Regards
Shankara M

shanku81, 2021-04-22:
Please find product information log below,
IDF_PATH: C:\Users\PKJena\Desktop\esp-idf\esp-idf-v4.2
IDF_PYTHON_ENV_PATH: <NOT FOUND>
PATH: <NOT FOUND>

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
ESP-IDF version cannot be checked. IDF_PATH or IDF_PYTHON_ENV_PATH  are not set.
Python set for IDF_PYTHON_ENV: <NOT FOUND>
Python declared in PATH variable: Python 3.6.6

wheres this option we can get it i dont find anywhere this option
The other thing that could be helpful is information about installed C/C++ toolchains:
Eclipse preferences... -> C/C++ -> Core build Toolchains

sigmaaa, 2021-04-22:
@shanku81 thank you for the product log. Looks like IDF_PATH is not set, that could be the root of the problem.

To install esp-idf and set IDF_PATH variable click on the Help -> Download and Configure ESP_IDF. After configuring you should see this message: 
[image]

Click "Yes" to make sure that all esp-idf tools are installed

To check toolchains on windows you have to click on the Window -> Preferences -> C/C++ -> Core Build Toolchains.

Also, after configuring ESP_IDF check if the IDF_PATH variable is set. To check it: 
Window -> Preferences -> C/C++ -> Build -> Environment 
Or you can do it with Help -> product information option.

shanku81, 2021-04-22:
Hi,

I followed steps as you mentioned and still problem exists.

Attached Core Build Toolchains and Environment log.

Thanks & Regards
Shankara M

shanku81, 2021-04-22:
I followed steps as you mentioned and still problem exists.

Attached Core Build Toolchains and Environment log. 
[image]
[image]

sigmaaa, 2021-04-22:
For some reason, the necessary toolchains to build esp projects are missing. 
[image]

Are you sure, that tools installation is passed successfully? Try it again, please 
Help - ESP-IDF tools manager -> install tools

Then you can check again product log. The correct one should also contain information about installed tools. For example:

Executing C:\Users\Denys\.espressif\python_env\idf4.4_py3.7_env\Scripts\python.exe D:\esp-master\esp-idf\tools\idf_tools.py list
WARNING: tool idf-python version 3.8.7 is installed, but has reported version 3.7.0
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - 1.24.0.123_64eb9ff-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)
* idf-python: Embeddable Python distribution (optional)
  - 3.8.7 (recommended)
* idf-python-wheels: Python Wheels distribution bundled for the specific version of IDF and Python
  - 3.8-2020-01-21 (recommended, installed)

IDF_PATH: D:\esp-master\esp-idf
IDF_PYTHON_ENV_PATH: C:\Users\Denys\.espressif\python_env\idf4.4_py3.7_env
PATH: ...

Operating System: windows 10
Java Runtime Version: 11.0.9+7-LTS
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.0.0.202101052004
ESP-IDF v4.4-dev-930-ga6f33a9d3-dirty
Python set for IDF_PYTHON_ENV: Python 3.7.0
Python declared in PATH variable: Python 3.7.0

shanku81, 2021-04-22:
Hi,

I tried installing tools again and please below log,
Installing tools...
This can take a while. Please be patient.
Executing C:\Python\python.exe C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py install all
Traceback (most recent call last):
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 1685, in <module>
    main(sys.argv[1:])
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 1681, in main
    action_func(args)
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 1295, in action_install
    tools_info = load_tools_info()
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 935, in load_tools_info
    with open(tool_versions_file_name, 'r') as f:  # type: ignore
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\PKJena\\.platformio\\packages\\framework-espidf\\tools/tools.json'

Executing C:\Python\python.exe C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py install-python-env
fatal: not a git repository (or any of the parent directories): .git
WARNING: Git describe was unsuccessul: Command '['git', 'describe']' returned non-zero exit status 128.
WARNING: Is it not possible to determine the IDF version: [Errno 2] No such file or directory: 'C:\\Users\\PKJena\\.platformio\\packages\\framework-espidf\\components\\esp_common\\include\\esp_idf_version.h'
ERROR: IDF version cannot be determined

Executing C:\Python\python.exe C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 1685, in <module>
    main(sys.argv[1:])
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 1681, in main
    action_func(args)
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 1068, in action_export
    tools_info = load_tools_info()
  File "C:\Users\PKJena\Desktop\esp-idf\tools\idf_tools.py", line 935, in load_tools_info
    with open(tool_versions_file_name, 'r') as f:  # type: ignore
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\PKJena\\.platformio\\packages\\framework-espidf\\tools/tools.json'
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.

Thanks & Regards
Shankara M

[toolchains]<https://user-images.githubusercontent.com/24419842/115677940-46c62000-a351-11eb-85da-fd13d4b3d2bd.png>

sigmaaa, 2021-04-29:
Hi @shanku81 
I was trying to reproduce your problem, I tried same python version on fresh windows 10, but everything is ok. 

WARNING: Git describe was unsuccessful: Command '['git', 'describe']' returned non-zero exit status 128.

Looks s like some problem with git. Can you please check git path, which you are providing to the eclipse in the "Install Tools" dialog and check if the provided git.exe is executable from the command line?

shanku81, 2021-04-29:
Hi,

Thanks for support.

Problem was path provided in install tools and ENV variable was not correct.
and target we need to set it esp32-c3

Thanks & Regards
Shankara M

kolipakakondal, 2021-05-03:
Thanks for confirming @shanku81. Closing this issue as it was working for you.

closed 2021-05-03 by kolipakakondal