espressif/idf-eclipse-plugin #259: How to fix the mess after esp-idf-tools-setup-offline-2.6.exe  offline installer?

samsam4, 2021-05-03:
Installation passed smooth without any errors reported, but after Create a new project using ESP-IDF Templates (based on instructions in README.md) after Build returns:
Build not configured correctly

So I start looking in the settings that suppose to be set automatically from installer, but what I see from my installation is quite different from what you show in the README.md
Help>Product Information reports:
IDF_PATH: D:/ESP/esp-idf
IDF_PYTHON_ENV_PATH: <NOT FOUND>
PATH: <NOT FOUND>

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.18.0.v20201202-1800
Eclipse CDT Version: 10.1.0.202010062020
ESP-IDF version cannot be checked. IDF_PATH or IDF_PYTHON_ENV_PATH  are not set.
Python set for IDF_PYTHON_ENV: <NOT FOUND>
Python declared in PATH variable: Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
[image]
[image]

Help>ESP-IDF Tools Manager> List installed tools reports:
Some tools are not installed or IDF_PATH is not set. …
[image]

So how can make this installation work without to waste weeks of trying?
Thanks

samsam4, 2021-05-03:
Update1:

After set  
IDF_PYTHON_ENV_PATH  and PATH variables I’ve got some progress, though there are some bug in setting the PATH variable:

I run ESP-IDF 4.2 CMD to set extra path settings for ESP then copy the path var from there, replace \ with /  and  add  a PATH environment variable, but next when run Help>Product Information: I got some garbage for PATH in the report that stays in front of what I set for PATH:

PATH: D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin/server;D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\ActiveState Komodo Edit 11\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;D:\AndroidSDK\platform-tools\;C:\Program Files\Java\jdk-12.0.1\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Users\xx\AppData\Local\Microsoft\WindowsApps;;D:\ESP\esp-idf;and_from_here _start whatever_I_put_for_PATH…. 

So I go back delete the content of the PATH variable, then paste again and finally was fixed

Now Question1: what to do with those idf_python_env_path, idf_python_export_path, virtualenv_python
and  idf_version_str?

 
[image]

Question 2: Core build toolchains Do I need to add/edit anything:

 
[image]

Question3: How to fix those ‘fatal:’ at the end of the report from Help>Product Information I have and which dir to put in Windows or Eclipse PATH variable for Python (that was no found)?
….
error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule third-party/mruby
fatal: 'git status --porcelain=2' failed in submodule components/nghttp/nghttp2
WARNING: Git version unavailable, reading from source
ESP-IDF v4.2.0
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.

Thanks

kolipakakondal, 2021-05-04:
Hi @samsam4 

We are sorry that it didn't work out of the box. It seems to be there is some issue in running install tools and configuring environment variables.

To troubleshoot further, could you please provide the Eclipse error log https://github.com/espressif/idf-eclipse-plugin#error-log. You can export and attach a file.

samsam4, 2021-05-04:
Hello,

Attached is my error log as requested. Meanwhile could you, please, answer the 3 questions in my second positing after I partially resolved the issue as I any way prefer not to uninstall/install again the offline installer if it possible to fix whatever left manually?

Thanks

kolipakakondal, 2021-05-04:
Hello @samsam4 

Looks like you have forgotten to attach a log file. Could you check?

Also,can you run the install tools action from Eclipse: `Help > ESP-IDF Tools Manager > Install Tools`. This will take care of configuring CDT build environment variables. Somehow these were not configured correctly In your case.

Please send the console log if you see any error during the install tools.

samsam4, 2021-05-04:
No I didn't forge

t - I replied on the email i received, because it claimed that I could either reply directly to the email or online.
"
So obviously the problem is with  Git - only text part of the response is accepted, attachments are discarder :(
Now I am responding/attaching online, so shouldn't be a problem ..
Regards

[err3_5_21.log](https://github.com/espressif/idf-eclipse-plugin/files/6418863/err3_5_21.log)

georgik, 2021-05-04:
Thank you @samsam4 for reporting the problem. It seems that paths to Python and Git were not set properly for some reason. Would it be possible to attach content of file:
```D:\ESP\tools\idf-eclipse\2021-*\esp_idf.json```
This is the file where installer stores information about tools location and Eclipse plugins reads this data.

samsam4, 2021-05-04:
Hi, attached zipped else git doesn't accept it.
Regards
[esp_idf_json.zip](https://github.com/espressif/idf-eclipse-plugin/files/6419155/esp_idf_json.zip)

georgik, 2021-05-04:
@samsam4 Thank you for the log. 
Do you use Eclipse deployed by installer or your own version?
The Eclipse deployed by installer should have following version:
```
Version: 2020-12 (4.18.0)
Build id: 20201210-1552
```

This Eclipse is located in `C:\ESP\tools\idf-eclipse\2021-03`, the same directory where `esp_idf.json` is located. The installer should also deploy the link to this Eclipse on the desktop.

Eclipse plugin loads the configuration file from Eclipse root directory. If you're using custom installation of Eclipse, please make sure to copy this `esp_idf.json` file to Eclipse root. The error in the seconds screenshot looks like a failure of the plugin to load values from esp_idf.json. The indication are these "varaiables"  `idf_python_env_path, idf_python_export_path, virtualenv_python` which should not be there.

kolipakakondal, 2021-05-04:
Looks here is the problem from the error log. Let me look further on this.

```
!ENTRY com.espressif.idf.core 1 0 2021-04-30 10:26:34.653
!MESSAGE Executing D:/ESP/python_env/idf4.2_py3.8_env/Scripts/python.exe D:\ESP\esp-idf\tools\idf_tools.py export --format=key-value

!ENTRY com.espressif.idf.core 1 0 2021-04-30 10:26:35.055
!MESSAGE Traceback (most recent call last):
  File "D:\ESP\esp-idf\tools\idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "D:\ESP\esp-idf\tools\idf_tools.py", line 1488, in main
    action_func(args)
  File "D:\ESP\esp-idf\tools\idf_tools.py", line 1022, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "D:\ESP\esp-idf\tools\idf_tools.py", line 860, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', 'describe'],
  File "subprocess.py", line 411, in check_output
  File "subprocess.py", line 489, in run
  File "subprocess.py", line 854, in __init__
  File "subprocess.py", line 1307, in _execute_child
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

kolipakakondal, 2021-05-04:
As a workaround, you can always run `Help > ESP-IDF Tools Manager > Install Tools` and install the tools by providing the correct git, idf and python paths.

samsam4, 2021-05-04:
@georgik 
I use Eclipse deployed by installer (from the desktop shortcut).
Just check the version - same what you said.
I have and 2 other older Eclipse installation on the same computer, but links to them dont exist anywhere in the windows PATH or other system variables.
Regards

samsam4, 2021-05-04:
@kolipakakondal
PATH and IDF_PYTHON_ENV_PATH did not exist after installation.
I first added IDF_PYTHON_ENV_PATH as was pretty straightforward which dir to set and tried to see if PATH will be created and autofilled. There was some progress (different output)  from Help>Product Information, but PATH didnt show yet, so I tried to solve what to put manually there as the example you posted in readme.md doesn't  correspond to what I had after installation (as per my second posting - I only forget to mentioned there, that I add and paths tom my existing jdk V12 installation).
Now after I try to analyze further the problem and trying to find best solution, maybe would be better to use (instead of my old jdk) paths to openjdk that came with installation and unexpectedly appears in the PATH while I was trying back and forth to find solution:

PATH: D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin/server;D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin; ....

Can you elaborate only why installer set **//** (double slashes)  - is it bug, intentionally or doesn't matter?
Thanks

samsam4, 2021-05-05:
@kolipakakondal
Help > ESP-IDF Tools Manager > Install Tools   doesn't help at all :(

[image]

I made two different attempts: First delete PATH env var hoping that Install tools will create it. 
During the install Error Log window popup with some warning messages.
[err4_5_21-1.log](https://github.com/espressif/idf-eclipse-plugin/files/6424621/err4_5_21-1.log)
After install PATH var was not created.

So I made second attempt, create and set PATH variable as:
D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin/server;D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;D:/ESP/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin;D:/ESP/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin;D:/ESP/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin;D:/ESP/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin;D:/ESP/tools/cmake/3.16.4/bin;D:/ESP/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin;D:/ESP/tools/ninja/1.10.0/;D:/ESP/tools/idf-exe/1.0.1/;D:/ESP/tools/ccache/3.7/;D:/ESP/tools/dfu-util/0.9/dfu-util-0.9-win64;D:/ESP/esp-idf/tools;D:/ESP/python_env/idf4.2_py3.8_env/Scripts;D:/ESP/tools/idf-git/2.30.1/cmd/;C:/Program Files (x86)/Common Files/Oracle/Java/javapath;C:/ProgramData/Oracle/Java/javapath;C:/Program Files (x86)/ActiveState Komodo Edit 11/;C:/WINDOWS/system32;C:/WINDOWS;C:/WINDOWS/System32/Wbem;C:/WINDOWS/System32/WindowsPowerShell/v1.0/;C:/WINDOWS/System32/OpenSSH/;C:/Program Files/PuTTY/;D:/AndroidSDK/platform-tools/;C:/Program Files/Java/jdk-12.0.1/bin;C:/Program Files/Nordic Semiconductor/nrf-command-line-tools/bin/;C:/Users/xx/AppData/Local/Microsoft/WindowsApps;

Run again Help > ESP-IDF Tools Manager > Install Tools
This is the error log: 
[err4_5_21-2.log](https://github.com/espressif/idf-eclipse-plugin/files/6424633/err4_5_21-2.log)

After run Help>Product information (**pay attention what PATH variables is reported, although in settings still show whatever I set when created it; neither 'fatals' for git at the end were resolved**) :(  :

Executing D:\ESP\python_env\idf4.2_py3.8_env\Scripts\python.exe D:\ESP\esp-idf\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: D:/ESP/esp-idf
IDF_PYTHON_ENV_PATH: D:/ESP/python_env/idf4.2_py3.8_env
PATH: C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\ActiveState Komodo Edit 11\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;D:\AndroidSDK\platform-tools\;C:\Program Files\Java\jdk-12.0.1\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Users\xx\AppData\Local\Microsoft\WindowsApps;;D:\ESP\esp-idf;D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin/server;D:/ESP/tools/idf-eclipse/2021-03//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.1.v20201027-0507/jre/bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;D:/ESP/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin;D:/ESP/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin;D:/ESP/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin;D:/ESP/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin;D:/ESP/tools/cmake/3.16.4/bin;D:/ESP/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin;D:/ESP/tools/ninja/1.10.0/;D:/ESP/tools/idf-exe/1.0.1/;D:/ESP/tools/ccache/3.7/;D:/ESP/tools/dfu-util/0.9/dfu-util-0.9-win64;D:/ESP/esp-idf/tools;D:/ESP/python_env/idf4.2_py3.8_env/Scripts;D:/ESP/tools/idf-git/2.30.1/cmd/;C:/Program Files (x86)/Common Files/Oracle/Java/javapath;C:/ProgramData/Oracle/Java/javapath;C:/Program Files (x86)/ActiveState Komodo Edit 11/;C:/WINDOWS/system32;C:/WINDOWS;C:/WINDOWS/System32/Wbem;C:/WINDOWS/System32/WindowsPowerShell/v1.0/;C:/WINDOWS/System32/OpenSSH/;C:/Program Files/PuTTY/;D:/AndroidSDK/platform-tools/;C:/Program Files/Java/jdk-12.0.1/bin;C:/Program Files/Nordic Semiconductor/nrf-command-line-tools/bin/;C:/Users/xx/AppData/Local/Microsoft/WindowsApps

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.18.0.v20201202-1800
Eclipse CDT Version: 10.1.0.202010062020
error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule third-party/mruby
fatal: 'git status --porcelain=2' failed in submodule components/nghttp/nghttp2
WARNING: Git version unavailable, reading from source
ESP-IDF v4.2.0
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.

georgik, 2021-05-06:
@samsam4 Hm, that's interesting error.

Please try following workaround.
- add following paths to PATH variable:
```
D:/ESP/python_env/idf4.2_py3.8_env/
D:/ESP/tools/idf-git/2.30.1/cmd/
```
- start new Eclipse

If this approach does not work, please help us to diagnose more information using following steps:
- download latest online installer 2.8: https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.8.exe
- 1. screen - accept license
- 2. screen - let the System check finish, click Full log, there should be more diagnostic details, please attach the log in case of failure to this issue
- 3. screen - select Use an existing ESP-IDF directory and specify `D:\esp\esp-idf`
- 4. screen - change cache directory if necessary or keep it. According to previous logs it should be `D:\esp`, in my opinion it's better to use separate directory which does not overlap with ESP-IDF e.g. `D:\esp\.espressif`, but your configuration should work
- 5. screen - check also Eclipse, check PowerShell - it will deploy new version with new version of the plugin
- 6. screen - Install and finish installation

Expected result:
- icon on Desktop pointing to new Eclipse in `D:\esp\tools\idf-eclipse\2021-04` which should also contain file `esp_idf.json` which points to Python virtualenv and to existing Git
- icon on Desktop pointing to PowerShell with ESP-IDF

Test build:
- open PowerShell by the desktop link, type 
```
cd examples/get-started/blink
idf.py build
```

Expected result:
- build should run and finis correctly

Test Eclipse:
- open Eclipse by icon on Desktop
- check configuration and it should match with the configuration in .json file

samsam4, 2021-05-06:
@georgik
Thanks - will try ASAP. Meanwhile could you elaborate on this - I found two different python.exe in this offline installation:

D:\ESP\python_env\idf4.2_py3.8_env\Scripts\python.exe 522kb
D:\ESP\tools\idf-python\3.8.7\python.exe 100kb

What is the second for and do I need to declare it anywhere in environment variables etc?
Regards

samsam4, 2021-05-06:
@georgik 
With adding those 2 sys vars to Windows Path, everything (fatal, error) cleared:  :)
D:\ESP\python_env\idf4.2_py3.8_env\Scripts\
D:\ESP\tools\idf-git\2.30.1\cmd\

Help>Product Information:
.....
Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.18.0.v20201202-1800
Eclipse CDT Version: 10.1.0.202010062020
ESP-IDF v4.2
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python 3.8.7

Will try though and new installer 2.8 in a new dir later, so could have some straws to compare if also doesnt go as expected ;)

Thank you and best regards

georgik, 2021-05-07:
@samsam4 Thank you for the feedback.

@kolipakakondal Please, could you check whether Eclipse plugin is adding path to Git and Python from the JSON to system PATH when launching subprocesses?
Other option would be to invoke shell initializers `.espressif\idf_cmd_init.bat` or `.espressif\Initialize-Idf.ps1` in the ESP-IDF directories. These initializers contains code which is able to find Git and Python specific for ESP-IDF instance. You can find them as target of desktop links deployed by installer.

kolipakakondal, 2021-05-07:
https://github.com/espressif/idf-eclipse-plugin/pull/261

Hi @samsam4 I have prepared a fix for this issue by setting up the git path to the java run-time process. To verify this, I need your help.

Download artifacts from here https://github.com/espressif/idf-eclipse-plugin/suites/2671932894/artifacts/58873860
And follow the instructions from here https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile

Let me know how it goes.

kolipakakondal, 2021-05-07:
Hi @georgik - problem is with the idf.py export command, which is getting failed because of the git path issue.  Let's wait for the @samsam4 feedback with the local build.

samsam4, 2021-05-07:
@kolipakakondal  #261
I am a little confused: Are we talking about version 2.6 which I already have installed and **fixed** OR for the new version 2.8 that I didn't start installing yet?
If we talking for 2.6: does your patch uses different dirs for 
D:\ESP\python_env\idf???_env\Scripts
D:\ESP\tools\idf-git\?.?.?\cmd\
else how I will check/be sure if the new patch works or not if I dont first remove manually older vars from Windows System Environment?
Also I remember that in order to move initially from total failure I had to set IDF_PYTHON_ENV_PATH manually - do I need to delete this or any other var in Window>Preferences>C/C++Build>Environment before try to install your fix?
Thanks

kolipakakondal, 2021-05-07:
It's simple! Maybe these instructions could help you @samsam4 

- Download a fresh copy of Eclipse CDT 2021-03 https://github.com/espressif/idf-eclipse-plugin#Prerequisites 
- Download local build copy https://github.com/espressif/idf-eclipse-plugin/suites/2671932894/artifacts/58873860
- Install as per the instructions mentioned here https://github.com/espressif/idf-eclipse-plugin#installPluginsUsingLocalFile
- Run this from eclipse: Help > ESP-IDF Tools Manager > Install Tools
- Verify build environment paths
- Verify create/build project

kolipakakondal, 2021-05-12:
Hi @samsam4 

Did you have a chance to verify the above build?

samsam4, 2021-05-12:
@kolipakakondal 
Sorry, not yet - have some work with higher priority to finish first. 
BTW Is your last patch embedded  in last offline installer? Because I'm more interested in the offline installer rather than to install everything separately?
Regards

kolipakakondal, 2021-05-12:
Hi @samsam4 No. It will be with the next version of the plugin and offline installer

samsam4, 2021-05-14:
@kolipakakondal 
Hi, From links that you posted above for Eclipse CDT and follow next link  I see "[Eclipse IDE for C/C++ Developers](https://www.eclipse.org/downloads/packages/release/2020-12/r/eclipse-ide-cc-developers)". There is a link [Eclipse CDT](https://wiki.eclipse.org/CDT/User/NewIn101) on the right side, but nothing to download from there?
So which Eclipse version to download??
Thanks

naveedabbasi, 2021-05-15:
Kindly try to look at this issue, I am unable to import projects other than the example in eclipse.
https://github.com/espressif/idf-eclipse-plugin/issues/267
................................................................

I was facing the same issue, I uninstall again and then install again, run the example, it workd.

samsam4, 2021-05-17:
More bugs:  :(
CMD shortcut seems was working  initially Ok, seems after Help > ESP-IDF Tools Manager > Install Tools  or any of the other manual tweak on environment variables (posted above) doesn't work anymore :(
[image]

The finny part is that I've made couple tries changing "echo."  to "echo ." or "echo _"  in idf_cmd_init.bat, but the output is always the same (including and if I type echo. in the same cmd window):  
'echo.' is not recognized as an internal or external command, operable program or batch file.

But if  open standard Windows10 cmd window or run a .bat file with some "echo." lines through a shortcut to cmd.exe, everything pass normal (outputs empty lines), so the screwed Win 10 was discarded as possible root cause and suspicion goes back to ESP plugin/tools :(

georgik, 2021-05-18:
@samsam4 Thank you for reporting problem with CMD initializer. The behavior is really strange.

echo. should print an empty line to the screen. Here is information from Microsoft manual: https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/echo

The initializer is using common Windows CMD `C:\WINDOWS\system32\cmd.exe`. You can see whole command when you right click the icon and select Properties.

Please, could you provide which Antivirus is present on the Windows PC? In some cases Antivirus adds it's own wrapper around cmd or runs it in container.

Do you experience the same problem when using PowerShell launcher from Desktop.

samsam4, 2021-05-18:
@georgik 
PowerShell still working Ok as at the beginning after installation.
CMD - I already mentioned in my previous post - I wrote a simple test_echo1.bat  with couple echo *** and echo.  
[test_echo1.zip](https://github.com/espressif/idf-eclipse-plugin/files/6501049/test_echo1.zip)
`
@echo off 
echo Test1
echo.
echo Test2
echo.
echo.
echo Test3
echo.
`
 and start it same way as Esp's cmd:  C:\Windows\System32\cmd.exe /k D:\W21\ESP\problem\echo_\test_echo1.bat
And it works  as suppose to:
[image]

 while at the sane time  idf_cmd_init.bat started from the installer generated shortcut  produces the errors output I posted before
Regards

samsam4, 2021-05-19:
Update: Cause for "'echo.' is not recognized as an internal or external command, operable program or batch file."  identified:
Somewhere between my post1 and post2  while I was trying to set manually some environment variables in order to kickoff Eclipse,  some of the Eclipse or ESP-IDF scripts produced a file with name echo (no extension) and length 0. 
So as far as such file exist in any Windows folder, starting a command prompt in this folder for "echo." always will return  "'echo.' is not recognized as an internal or external command, operable program or batch file."
but surprisingly "echo blablabla"  returns blablabla
So I moved away from esp-idf  folder in other folder this file "echo" and ESP-IDF 4.2 CMD works normal as was working at the very beginning :)

georgik, 2021-05-19:
@samsam4 Wow. That's insidious trap. I would never guess that `echo` file can generate such behavior. I'll double check deployment process for potential source of empty `echo` file.

georgik, 2021-05-19:
@samsam4 Regarding 'echo.' problem: I was not able to find the source of the 'echo.' file in deployment process.
I've added check to CMD initializer in PR: https://github.com/espressif/idf-installer/pull/21  
The updated installer is available here: https://github.com/espressif/idf-installer/releases/tag/online-2.9-beta-04

samsam4, 2021-05-25:
@georgik  
I've tested 2.9beta04 online installer - seems work Ok - just minor "cosmetic" issues had during installation.
Error log shows some errors and warnings  [err_log1.log](https://github.com/espressif/idf-eclipse-plugin/files/6535929/err_log1.log) , but I was able to build the 'blink' example without errors (didn't try to flash it yet though) [err_log_blink2.log](https://github.com/espressif/idf-eclipse-plugin/files/6535933/err_log_blink2.log) .

The only serious issue I have so far not resolved - cant get/access 'make' command (in 2.6 installer too). As per ESP documentation 'make' may be removed in future, but it claims should be still available, but I cant get it neither in ESP-IDF cmd nor in power shell?  
Regards

georgik, 2021-05-25:
@samsam4 Thank you for posting those logs and testing the installation.

Regarding make I've checked old documentation and v3.3 is referencing `make` as build tool for ESP-IDF: https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started/index.html

Make was replaced by `idf.py`, `CMake` and `ninja`. Referenced from documentation starting v4.0.2: https://docs.espressif.com/projects/esp-idf/en/v4.0.2/get-started/index.html. There should not be a need for `make`.

Please, could you send a link to the documentation mentioning `make`?

kolipakakondal, 2021-05-25:
Hi @samsam4 

The errors you see in the error log related to eclipse loading and it does not cause any issue to idf eclipse plugins and functionality. Anyway, those are not real errors to be considered. Hence, ignore them.

IDF Eclipse plugin works only with the CMake and ESP-IDF 4.0 and higher.

Glad that the online installer approach worked for you.  I believe we can close this issue.

samsam4, 2021-05-26:
Yes, I think you can close this issue. After I see 2.9 offline installer if there are some issues I could start a new thread.
There is only one "cosmetic" issue that I found - missing button "Make New Folder"
[image]
If you select "Browse" on previous dialog for Select Destination Location. There is a workaround to type the name of the path including the new folder, but it is not quite intuitive and could be problem for some people.

@georgik there were few places were is mentioned:
> Since ESP-IDF V4.0, the default build system is based on CMake. This documentation is for the legacy build system based on GNU Make. Support for this build system may be removed in future major releases.

, but it was my mistake - I didnt  read all the chain of links that follow - it's different toolchain that uses the old MSYS2. I was trying to explore[ one library for SPI displays](https://github.com/jeremyjh/ESP32_TFT_library) that is using esp-idf 4.0 but still using 'make' for the creation and flashing spiffs images, but obviously will have to search  other solution :(

Thanks for the cooperation in a timely manner!  Best regards!

georgik, 2021-05-27:
@samsam4 Thank you for pointing out the Browse problem. I'll check whether it's possible to add it somehow. The installer is based on Inno Setup which has limited options of changing UI.

samsam4, 2021-05-27:
@georgik  This is a 'cosmetic' i.e. not so important issue, but because I saw on the other dialogs in the same installer that you already have such button, I thought is just missed on this particular one that I mentioned above.
Best regards

[image]

samsam4, 2021-05-30:
Unfortunately the old problem with 'fatal's and cant find python in the PATH etc. resurfaced without to make any changes in installation, even didn't open this installation of Eclipse for couple days. Only downloaded one ESP project from [here](https://github.com/nopnop2002/esp-idf-ili9340), imported it and when I saw some errors checked Help>Product Information, then saw similar errors like before, closed the new project and return to 'blink' project - same error output for Help>Product Information, ... restart Eclipse, restart Windows, always same output:
[errLog30-5-21.log](https://github.com/espressif/idf-eclipse-plugin/files/6566372/errLog30-5-21.log)

Executing D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe D:\ESP2_9b\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: D:/ESP2_9b/
IDF_PYTHON_ENV_PATH: D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env
PATH: D:\ESP2_9b_tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;D:\ESP2_9b_tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;D:\ESP2_9b_tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;D:\ESP2_9b_tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;D:\ESP2_9b_tools\.espressif\tools\cmake\3.16.4\bin;D:\ESP2_9b_tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;D:\ESP2_9b_tools\.espressif\tools\ninja\1.10.0\;D:\ESP2_9b_tools\.espressif\tools\idf-exe\1.0.1\;D:\ESP2_9b_tools\.espressif\tools\ccache\3.7\;D:\ESP2_9b_tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;D:\ESP2_9b_tools\.espressif\python_env\idf4.2_py3.8_env\Scripts;D:\ESP2_9b\tools;D:/ESP2_9b_tools/.espressif/tools/idf-eclipse/2021-04//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.2.v20210201-0955/jre/bin/server;D:/ESP2_9b_tools/.espressif/tools/idf-eclipse/2021-04//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_15.0.2.v20210201-0955/jre/bin;D:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\;D:\ESP2_9b_tools\.espressif\tools\idf-git\2.30.1\cmd\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\ActiveState Komodo Edit 11\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;D:\AndroidSDK\platform-tools\;C:\Program Files\Java\jdk-12.0.1\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Users\xx\AppData\Local\Microsoft\WindowsApps;;D:\ESP2_9b;

Operating System: windows 10
Java Runtime Version: 15.0.2+7-27
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
**error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule third-party/mruby
fatal: 'git status --porcelain=2' failed in submodule components/nghttp/nghttp2
WARNING: Git version unavailable, reading from source**
ESP-IDF v4.2.1
Python set for IDF_PYTHON_ENV: Python 3.8.7
**Python declared in PATH variable: Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.**

samsam4, 2021-06-01:
Will close this issue as the latest problem although observed and before with 2.6 is with the newest version of plugin/installer 2.9beta, so better to rise it as a new issue.
Thanks everybody who helped to clear most of the issues in this thread so far.

closed 2021-06-01 by samsam4

georgik, 2021-06-02:
@samsam4 Thank you for all the data. Let me just add an update about Make New Folder.
The page where Make New Folder is missing is the default page of Inno Setup for selecting destination which has by default disabled flag for creating the folder.

One solution would be to implement a completely new page to handle the selection like here: https://stackoverflow.com/questions/42145152/inno-setup-how-do-i-specify-multiple-directories-for-select-destination-locatio

Feel free to add PR to https://github.com/espressif/idf-installer

We're considering moving away from Inno Setup which has many limitations like missing Make New Folder.

samsam4, 2021-06-02:
@georgik  Thanks for the update - I'd suggest to forget for now for this problem as it is really petty for most of the people that will use this installer. Seems there far more important problems to polish meanwhile, so better fix them with higher priority ;)
Best regards

kolipakakondal, 2021-06-29:
> Update1:
>
> After set
> IDF_PYTHON_ENV_PATH and PATH variables I’ve got some progress, though there are some bug in setting the PATH variable:
>
> I run ESP-IDF 4.2 CMD to set extra path settings for ESP [...]

The issue related to environment variables got fixed with v2.1.1

samsam4, 2021-07-01:
@kolipakakondal  The issue related to environment variables got fixed with v2.1.1

Unfortunately the [tests I made with the newest as of today v2.1.2 plugin](https://github.com/espressif/idf-eclipse-plugin/issues/283#issuecomment-870611778)  shows neither problem with environment variables were fixed even there are more problems than before  as the projects cant even kick off because the plugin keeps asking again  and again for git and python locations :( :(

nathangoss, 2021-07-09:
I am also experiencing the same issues on a fresh install of the latest Eclipse and ESP-IDF on my Mac.

@samsam4  - have you managed to have any luck getting a project opened?

samsam4, 2021-07-09:
@nathangoss 
Yes, I have a workaround, but is for Windows - add git and python paths to Windows Environment variable Path. Not sure how can achieve similar functionality with Mac OS. 
Also - older plugin  v2.1.0 behaves much better regarding this problem [than the newest 2.1.2](https://github.com/espressif/idf-eclipse-plugin/issues/283#issuecomment-870611778) (haven't tried the one between 2.1.1) but the older plugins  cant work with the newest Eclipse! (with Eclipse 2021-03 plugin v2.1.0 works fine regarding this problem than newest plugin)!  

https://user-images.githubusercontent.com/30199901/120491459-4032bb80-c387-11eb-88cd-72fc6702542d.png