espressif/idf-eclipse-plugin #260: ESP-IDF Eclipse Plugin v2.1.0 on Eclipse 2021-03 cannot complete the install because of a conflicting dependency

rtyle, 2021-05-04:
**Describe the bug**
ESP-IDF Eclipse Plugin v2.1.0 on Eclipse 2021-03 cannot complete the install because of a conflicting dependency.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install of Fedora 34 with bundled Eclipse 2021-03 CDT
2. Fresh eclipse workspace
2. Help: Install New Software (Espressif IDF Plugins for Eclipse) from (Add) https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

**Expected behavior**
Success.

**Screenshots**
Install Details
Cannot complete the install because of a conflicting dependency.
  Software being installed: Espressif IDF Plugins for Eclipse 2.1.0.202104161244 (com.espressif.idf.feature.feature.group 2.1.0.202104161244)
  Software currently installed: Eclipse Platform 4.19.0.v20210304-1735 (org.eclipse.platform.ide 4.19.0.v20210304-1735)
  Only one of the following can be installed at once: 
    Simple Configurator 1.3.600.v20210304-1735 (org.eclipse.equinox.simpleconfigurator 1.3.600.v20210304-1735)
    Simple Configurator 1.3.600.v20200721-1308 (org.eclipse.equinox.simpleconfigurator 1.3.600.v20200721-1308)
  Cannot satisfy dependency:
    From: Espressif IDF Plugins for Eclipse 2.1.0.202104161244 (com.espressif.idf.feature.feature.group 2.1.0.202104161244)
    To: org.eclipse.equinox.p2.iu; org.eclipse.equinox.simpleconfigurator [1.3.600.v20200721-1308,1.3.600.v20200721-1308]
  Cannot satisfy dependency:
    From: Eclipse Core Runtime Infrastructure 1.2.1200.v20210304-1735 (org.eclipse.core.runtime.feature.feature.group 1.2.1200.v20210304-1735)
    To: org.eclipse.equinox.p2.iu; org.eclipse.equinox.core.feature.feature.group [1.13.100.v20210304-1735,1.13.100.v20210304-1735]
  Cannot satisfy dependency:
    From: Equinox Core Function 1.13.100.v20210304-1735 (org.eclipse.equinox.core.feature.feature.group 1.13.100.v20210304-1735)
    To: org.eclipse.equinox.p2.iu; org.eclipse.equinox.simpleconfigurator [1.3.600.v20210304-1735,1.3.600.v20210304-1735]
  Cannot satisfy dependency:
    From: Eclipse Platform 4.19.0.v20210304-1735 (org.eclipse.platform.ide 4.19.0.v20210304-1735)
    To: org.eclipse.equinox.p2.iu; org.eclipse.core.runtime.feature.feature.group [1.2.1200.v20210304-1735,1.2.1200.v20210304-1735]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

[configuration.txt](https://github.com/espressif/idf-eclipse-plugin/files/6423150/configuration.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools

Not available.

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"

Not available

For v1.1.0 and above: Eclipse > About Eclipse > Click on "espresif" icon

Not available

**Eclipse Error log:** 
Window > Show View > Error Log

Please attach the error log as described here [Untitled.log](https://github.com/espressif/idf-eclipse-plugin/files/6423126/Untitled.log)

kolipakakondal, 2021-05-06:
Hi @rtyle Sorry that it didn't work out of the box. Let us take a look at it.

@sigmaaa I was not having this issue on macOS and Windows with Eclipse 2021-03 with plugin 2.1.0, could you please try on Linux box?

sigmaaa, 2021-05-07:
Hi @rtyle, @kolipakakondal 
I don't have this issue on fresh Linux Fedora also. 
We will continue to investigate the cause of the error and let you know.

kolipakakondal, 2021-05-07:
Thanks @sigmaaa 

Hi @rtyle Could you try on the fresh installation of Eclipse CDT 2021-03 with our recent version 2.1.0, I am not sure if you have installed any other additional CDT or other plugins before installing the IDF Eclipse plugin.

Let us know how it goes.

rtyle, 2021-05-07:
I did:
> 1. Fresh install of Fedora 34 with bundled Eclipse 2021-03 CDT
> 2. Fresh eclipse workspace

I also tried getting get rid of old eclipse artifacts before installing
`rm -rf ~/.{eclipse,p2}`

I have not

rtyle, 2021-05-07:
oops

kolipakakondal, 2021-05-07:
[image]

Please find the screenshot

rtyle, 2021-05-07:
This

> https://github.com/espressif/idf-eclipse-plugin/releases

references your v2.1.0 download

> https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-v2.1.0.zip

which contains

> ./plugins/org.eclipse.equinox.simpleconfigurator_1.3.600.v20200721-1308.jar

On the contrary,

> https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-v2.0.0.zip

contains no simpleconfigurator.

The Fedora 34 eclipse RPM

> eclipse-platform-4.19-3.fc34.x86_64

installs 

> /usr/share/java/eclipse/equinox.simpleconfigurator.jar

which has a conflicting version

`
unzip -qc /usr/share/java/eclipse/equinox.simpleconfigurator.jar META-INF/MANIFEST.MF | grep Bundle-Version
Bundle-Version: 1.3.600.v20210304-1735
`

So, I am able to install v2.0.0 but not v2.1.0

rtyle, 2021-05-07:
> Hi @rtyle Could you try the following configuration option and see that helps while identify the root cause.
> 
> 1. Go to `Applications/Eclipse_202103.app/Contents/Eclipse/configuration/`

There is no such directory.
I am on Fedora 34

kolipakakondal, 2021-05-08:
> > Hi @rtyle Could you try the following configuration option and see that helps while identify the root cause.
> > 
> > 1. Go to `Applications/Eclipse_202103.app/Contents/Eclipse/configuration/`
> 
> There is no such directory.
> I am on Fedora 34

Hi, Look for the `configuration` folder in the `eclipse` directory. What I mentioned above is the macOS complete path, it might be different on linux.

rtyle, 2021-05-08:
I looked.
I could not find any "config.ini" file like this.
Please address the simpleconfigurator version conflict

sigmaaa, 2021-05-10:
Hi @rtyle,
now I'm able to reproduce problem that you have. You are probably installing eclipse from package manager:
https://fedora.pkgs.org/34/fedora-updates-testing-x86_64/eclipse-platform-4.19-3.fc34.x86_64.rpm.html
am I right? While we are fixing issue, I'm suggesting you to install eclipse from official website:
https://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/2021-03/R/eclipse-cpp-2021-03-R-linux-gtk-x86_64.tar.gz
To install it, extract eclipse-installer folder and execute eclipse-inst 
with this version I don't have problem on my fedora 34 box.

rtyle, 2021-05-11:
> To install it, extract eclipse-installer folder and execute eclipse-inst

It seems you keep going back to MacOS.
There is no installer for Linux.
`tar tzf eclipse-cpp-2021-03-R-linux-gtk-x86_64.tar.gz | grep eclipse-inst`

Anyway, I tried this before and it did not work.
That is, eclipse just crashed.
I updated everything on my system, tried again and this time ...
It works!

Of course, I should not have maintain Eclipse as that is what the Fedora distribution is for.
I hope that you can reconcile this issue with Fedora.
Thanks.

kolipakakondal, 2021-06-24:
Hello, ESP-IDF Eclipse Plugin v2.1.2 (Build: 2.1.2.202106231127) is released to support Eclipse 2021-06(4.20) version. For update please check this https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.1.2

closed 2021-06-24 by kolipakakondal