espressif/idf-eclipse-plugin #267: Esp idf issue  Importing an existing projects other than examples in eclipse

naveedabbasi, 2021-05-12:
I am facing an issue while Importing an existing project other than examples in eclipse  Esp idf 2.6 and also in eclipse  Esp idf 2.8.  both versions run examples  and build with no errors,
 but whenever I add new > header and C files in the existing example project it is unable to add and find header and C files in the project, I also put these files in the main folder too. 
after searching on forums :
I also tried to add these files CMakeLists.txt but the problem not solved.

2nd:  whenever I  existing project other than examples ,  all files [.C files and .H files ] are added but this time 
eclipse   will unable to add these fils in the project , 
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
 
is any way I can import my existing projects without facing the problems.  I only want to use eclipse right now.

projectgus, 2021-05-14:
Hi @naveedabbasi,

> I also tried to add these files CMakeLists.txt but the problem not solved.

Could you please give some specific details about the content of these files:

* What is the directory layout of your project? (i.e. all the files and the directories they are in)
* Which C source file in your project contains app_main() function?
* What is the contents of the CMakeLists.txt file that refers to that source file?

Thanks.

naveedabbasi, 2021-05-14:
1]

This is my CMakeLists.txt in the example folder where take_picture.c" main c file is.

////////////////////////////////////////////////////////////////////////////
idf_component_register(SRCS "take_picture.c"
                    INCLUDE_DIRS "")
  
  set(COMPONENT_SRCS
    driver/camera.c
    driver/sccb.c
    driver/sensor.c
    driver/xclk.c
    sensors/ov2640.c
    sensors/ov3660.c
    sensors/ov5640.c
    sensors/ov7725.c
    sensors/ov7670.c
    sensors/nt99141.c
    conversions/yuv.c
    conversions/to_jpg.cpp
    conversions/to_bmp.c
    conversions/jpge.cpp
    conversions/esp_jpg_decode.c
    )

  set(COMPONENT_ADD_INCLUDEDIRS
    driver/include
    conversions/include
    )

  set(COMPONENT_PRIV_INCLUDEDIRS
    driver/private_include
    sensors/private_include
    conversions/private_include
    )

  set(COMPONENT_REQUIRES driver)
  set(COMPONENT_PRIV_REQUIRES freertos nvs_flash)

  register_component()
 
//////////////////////////////////////////////////////////////////
[image]

naveedabbasi, 2021-05-14:
What is the directory layout of your project? (i.e. all the files in your project and the directories they are in)

all the files are in my project directories .

Which C source file in your project contains app_main() function?
 Main File

What is the full contents of the CMakeLists.txt file that refers to that source file?
I think yes if I am not wrong.

projectgus, 2021-05-14:
Hi @naveedabbasi ,

> all the files are in my project directories .

I think the layout shown isn't including the `examples/CMakeLists.txt`, so none of your source files are part of the project yet. This is why you see all the yellow lines as well.

To start with, suggest renaming the "examples" directory in your project to "main". Then move the directories "drivers", "sensors", "conversion" under the main directory. This will either build, or give you a different error message that you can use to debug further.

A (more complex) sample project layout that you can refer to is shown here:
https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-guides/build-system.html#example-project

Or refer to the layout of any of the examples bundled in ESP-IDF (you should be able to copy the example project directory out to your own directory and work on it from there.)

> Which C source file in your project contains app_main() function?
> Main File

I don't see any file called "Main File" in your post. You need to provide specific information if you want assistance, a vague description is not enough.

naveedabbasi, 2021-05-14:
Sure .  by  Main File ,,, I meant take_picture.c which is in  [examples] folder

naveedabbasi, 2021-05-14:
suggest renaming the "examples" directory in your project to "main". Then move the directories "drivers", "sensors", "conversion" under the main directory. 

I did and moved the directories "drivers", "sensors", "conversion" under the main directory. but got same error,

I think my directories "drivers", "sensors", "conversion are included in projects , Because I not getting error on 

#include "esp_camera.h"  .. which is included in above directory   "drivers..

I am getting this error

Building in: C:\Users\FUUAST\eclipse-workspace\esp32-camera\build
cmake --build . -- -v
[1/5] cmd.exe /C "cd /D C:\Users\FUUAST\eclipse-workspace\esp32-camera\build\bootloader && C:\Users\FUUAST\.espressif\tools\cmake\3.16.4\bin\cmake.exe --build ."
ninja: no work to do.
[2/3] cmd.exe /C "cd . && C:\Users\FUUAST\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\esp32-camera.elf.rsp  -o esp32-camera.elf  && cd ."
FAILED: esp32-camera.elf 
cmd.exe /C "cd . && C:\Users\FUUAST\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\esp32-camera.elf.rsp  -o esp32-camera.elf  && cd ."
c:/users/fuuast/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp32/libesp32.a(cpu_start.c.obj):(.literal.main_task+0x18): undefined reference to `app_main'
c:/users/fuuast/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp32/libesp32.a(cpu_start.c.obj): in function `main_task':
C:/Users/FUUAST/Desktop/esp-idf/components/esp32/cpu_start.c:592: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Build complete (3 errors, 0 warnings): C:\Users\FUUAST\eclipse-workspace\esp32-camera\build
Total time taken to build the project: 4,028 ms