espressif/idf-eclipse-plugin #268: Could Not Build or Run Sample Project

FrozenNick, 2021-05-13:
Hi, I am having issue with setting up my ESP-IDF plugin in eclipse

I managed to follow the steps from this [link](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md) up until **Compiling the Project**

When I hit "Build", it shows an error like this:

-- Checking Python dependencies...
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
CMake Error at C:/Users/WeiNick/esp/esp-idf-v4.2.1/tools/cmake/build.cmake:271 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/WeiNick/esp/esp-idf-v4.2.1/tools/cmake/build.cmake:397 (__build_check_python)
  C:/Users/WeiNick/esp/esp-idf-v4.2.1/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

Can anybody help? I am stuck with what I need to do now, I checked my python PATH in my windows environment variable by simply using CMD, and typed "python" to see if it runs or not, and it did.

**Eclipse Installation Details**
[EclipseInstallationDetails.txt](https://github.com/espressif/idf-eclipse-plugin/files/6470432/EclipseInstallationDetails.txt)

**ESP-IDF Tools Information:**
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
2.1.0.202104161244

naveedabbasi, 2021-05-13:
Uninstall and Again install IDF Eclipse Plugin, it will solve the examples projects, But I do not know who I can import project in C

FrozenNick, 2021-05-14:
> 
> 
> Uninstall and Again install IDF Eclipse Plugin, it will solve the examples projects, But I do not know who I can import project in C

I tried multiple times dy, it has the same result, I think something is wrong with the path environment, but I am not sure where to start looking.

I tried setting up the environment in my Ubuntu system yesterday evening, it works well, but I really want to work with my windows system

sigmaaa, 2021-05-17:
Hi @FrozenNick, 
Can you provide a product information log, please? To do it, go to Help > Product Information.

FrozenNick, 2021-05-17:
Hi @si
Executing C:\Users\WeiNick\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: C:\Users\WeiNick\esp\esp-idf-v4.2.1
IDF_PYTHON_ENV_PATH: C:\Users\WeiNick\.espressif\python_env\idf4.2_py3.9_env
PATH: C:\Users\WeiNick\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\WeiNick\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\WeiNick\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\WeiNick\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\WeiNick\.espressif\tools\cmake\3.16.4\bin;C:\Users\WeiNick\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\WeiNick\.espressif\tools\ninja\1.10.0\;C:\Users\WeiNick\.espressif\tools\idf-exe\1.0.1\;C:\Users\WeiNick\.espressif\tools\ccache\3.7\;C:\Users\WeiNick\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\WeiNick\.espressif\python_env\idf4.2_py3.9_env\Scripts;C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files\Java\jdk-15.0.1\bin;C:\Program Files\Microchip\xc8\v2.31\bin;C:\Program Files\Microchip\xc16\v1.60\bin;C:\Program Files\Microchip\xc32\v2.50\bin;C:\Program Files\Git\cmd;C:\Program Files\Inkscape\bin;C:\Program Files (x86)\Google\Cloud SDK\google-cloud-sdk\bin;C:\Users\WeiNick\AppData\Local\Programs\Python\Python3;C:\Users\WeiNick\AppData\Local\Microsoft\WindowsApps\python.exe;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\CMake\bin;C:\Program Files\dotnet\;C:\Users\WeiNick\AppData\Local\Programs\Python\Python39;C:\Users\WeiNick\AppData\Local\Programs\Python\Python39\Scripts;C:\Users\WeiNick\AppData\Local\Microsoft\WindowsApps;C:\Program Files\JetBrains\PyCharm Community Edition 2020.2.4\bin;C:\Users\WeiNick\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\WeiNick\AppData\Local\GitHubDesktop\bin;C:\Program Files\Git\bin

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
Traceback (most recent call last):
  File "C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools\idf.py", line 45, in <module>
    from idf_py_actions.tools import (executable_exists, idf_version, merge_action_lists, realpath)  # noqa: E402
  File "C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools\idf_py_actions\tools.py", line 1, in <module>
    import click
ModuleNotFoundError: No module named 'click'
Python set for IDF_PYTHON_ENV: Python 3.9.5
Python declared in PATH variable: Python 3.8.5

> 
> 
> Hi @FrozenNick,
> Can you provide a product information log, please? To do it, go to Help > Product Information.

FrozenNick, 2021-05-17:
Executing C:\Users\WeiNick\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: C:\Users\WeiNick\esp\esp-idf-v4.2.1
IDF_PYTHON_ENV_PATH: C:\Users\WeiNick\.espressif\python_env\idf4.2_py3.9_env
PATH: C:\Users\WeiNick\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\WeiNick\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\WeiNick\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\WeiNick\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\WeiNick\.espressif\tools\cmake\3.16.4\bin;C:\Users\WeiNick\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\WeiNick\.espressif\tools\ninja\1.10.0\;C:\Users\WeiNick\.espressif\tools\idf-exe\1.0.1\;C:\Users\WeiNick\.espressif\tools\ccache\3.7\;C:\Users\WeiNick\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\WeiNick\.espressif\python_env\idf4.2_py3.9_env\Scripts;C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files\Java\jdk-15.0.1\bin;C:\Program Files\Microchip\xc8\v2.31\bin;C:\Program Files\Microchip\xc16\v1.60\bin;C:\Program Files\Microchip\xc32\v2.50\bin;C:\Program Files\Git\cmd;C:\Program Files\Inkscape\bin;C:\Program Files (x86)\Google\Cloud SDK\google-cloud-sdk\bin;C:\Users\WeiNick\AppData\Local\Programs\Python\Python3;C:\Users\WeiNick\AppData\Local\Microsoft\WindowsApps\python.exe;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\CMake\bin;C:\Program Files\dotnet\;C:\Users\WeiNick\AppData\Local\Programs\Python\Python39;C:\Users\WeiNick\AppData\Local\Programs\Python\Python39\Scripts;C:\Users\WeiNick\AppData\Local\Microsoft\WindowsApps;C:\Program Files\JetBrains\PyCharm Community Edition 2020.2.4\bin;C:\Users\WeiNick\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\WeiNick\AppData\Local\GitHubDesktop\bin;C:\Program Files\Git\bin

Operating System: windows 10
Java Runtime Version: 15.0.1+9-18
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.0.202104140338
Traceback (most recent call last):
  File "C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools\idf.py", line 45, in <module>
    from idf_py_actions.tools import (executable_exists, idf_version, merge_action_lists, realpath)  # noqa: E402
  File "C:\Users\WeiNick\esp\esp-idf-v4.2.1\tools\idf_py_actions\tools.py", line 1, in <module>
    import click
ModuleNotFoundError: No module named 'click'
Python set for IDF_PYTHON_ENV: Python 3.9.5
Python declared in PATH variable: Python 3.8.5

FrozenNick, 2021-05-17:
> 
> 
> Hi @FrozenNick,
> Can you provide a product information log, please? To do it, go to Help > Product Information.

Hi @sigmaaa , attached the product information

kolipakakondal, 2021-05-18:
Hi @FrozenNick Seems to be some python PATH issue, while we investigate this could you check our offline installer, this will install the required tools including the required python and eclipse ide.

Please give it a try and see if that helps.

kolipakakondal, 2022-02-15:
Hello,

With our recent releases v2.4.0/v2.4.1, we have made a couple of fixes in the path configuration and onboarding workflows.

If you are a windows user, please do check our espressif-ide offline installer which comes with the all required tools to work with esp-idf development.

Check this
https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/EspressifIDE.md#espressif-ide-v240-installer-for-windows

Thanks

closed 2022-02-15 by kolipakakondal

volkanunal, 2022-10-15:
@kolipakakondal  The link is broken.